noble/noble #119: npm install failure

rwaldron, 2015-01-03:
I've aleady read and tried the suggestions here: https://github.com/sandeepmistry/node-xpc-connection/issues/2

System:

- OSX 10.10.1
- node v0.10.35
- xcode-select version 2339

```
$ npm install noble

> noble@0.3.6 install /Users/rwaldron/ble/node_modules/noble
> node install.js

noble install: platform is "darwin'
noble install: installing xpc-connection ...

> xpc-connection@0.0.4 install /Users/rwaldron/ble/node_modules/noble/node_modules/xpc-connection
> node-gyp rebuild

  CXX(target) Release/obj.target/binding/src/XpcConnection.o
In file included from ../src/XpcConnection.cpp:1:
In file included from /System/Library/Frameworks/Foundation.framework/Headers/Foundation.h:10:
In file included from /System/Library/Frameworks/Foundation.framework/Headers/NSArray.h:7:
In file included from /System/Library/Frameworks/Foundation.framework/Headers/NSRange.h:5:
/System/Library/Frameworks/Foundation.framework/Headers/NSValue.h:12:1: error:
      'objc_returns_inner_pointer' attribute only applies to methods
@property (readonly) const char *objCType NS_RETURNS_INNER_POINTER;
^                                         ~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../src/XpcConnection.cpp:1:
In file included from /System/Library/Frameworks/Foundation.framework/Headers/Foundation.h:15:
In file included from /System/Library/Frameworks/Foundation.framework/Headers/NSCharacterSet.h:8:
/System/Library/Frameworks/Foundation.framework/Headers/NSString.h:248:1: error:
      'objc_returns_inner_pointer' attribute only applies to methods
@property (readonly) __strong const char *UTF8String NS_RETURNS_INNER_POINTER;  /...
^                                                    ~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../src/XpcConnection.cpp:1:
In file included from /System/Library/Frameworks/Foundation.framework/Headers/Foundation.h:17:
/System/Library/Frameworks/Foundation.framework/Headers/NSData.h:76:1: error:
      'objc_returns_inner_pointer' attribute only applies to methods
@property (readonly) const void *bytes NS_RETURNS_INNER_POINTER;
^                                      ~~~~~~~~~~~~~~~~~~~~~~~~
/System/Library/Frameworks/Foundation.framework/Headers/NSData.h:161:1: error:
      'objc_returns_inner_pointer' attribute only applies to methods
@property (readonly) void *mutableBytes NS_RETURNS_INNER_POINTER;
^                                       ~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../src/XpcConnection.cpp:1:
In file included from /System/Library/Frameworks/Foundation.framework/Headers/Foundation.h:25:
/System/Library/Frameworks/Foundation.framework/Headers/NSDecimalNumber.h:101:1: error:
      'objc_returns_inner_pointer' attribute only applies to methods
@property (readonly) const char *objCType NS_RETURNS_INNER_POINTER;
^                                         ~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../src/XpcConnection.cpp:1:
In file included from /System/Library/Frameworks/Foundation.framework/Headers/Foundation.h:31:
In file included from /System/Library/Frameworks/Foundation.framework/Headers/NSFileManager.h:8:
/System/Library/Frameworks/Foundation.framework/Headers/NSPathUtilities.h:34:1: error:
      'objc_returns_inner_pointer' attribute only applies to methods
@property (readonly) __strong const char *fileSystemRepresentation NS_RETURNS_INNER_POINTER;
^                                                                  ~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../src/XpcConnection.cpp:1:
In file included from /System/Library/Frameworks/Foundation.framework/Headers/Foundation.h:41:
/System/Library/Frameworks/Foundation.framework/Headers/NSKeyValueObserving.h:183:1: error:
      'objc_returns_inner_pointer' attribute only applies to methods
@property void *observationInfo NS_RETURNS_INNER_POINTER;
^                               ~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../src/XpcConnection.cpp:1:
In file included from /System/Library/Frameworks/Foundation.framework/Headers/Foundation.h:46:
/System/Library/Frameworks/Foundation.framework/Headers/NSMethodSignature.h:22:1: error:
      'objc_returns_inner_pointer' attribute only applies to methods
@property (readonly) const char *methodReturnType NS_RETURNS_INNER_POINTER;
[... 401 lines cut ...]
158 error argv "node" "/usr/local/bin/npm" "install" "noble"
159 error node v0.10.35
160 error npm  v2.1.17
161 error code ELIFECYCLE
162 error noble@0.3.6 install: `node install.js`
162 error Exit status 1
163 error Failed at the noble@0.3.6 install script 'node install.js'.
163 error This is most likely a problem with the noble package,
163 error not with npm itself.
163 error Tell the author that this fails on your system:
163 error     node install.js
163 error You can get their info via:
163 error     npm owner ls noble
163 error There is likely additional logging output above.
164 verbose exit [ 1, true ]
165 verbose unbuild node_modules/noble
166 info preuninstall noble@0.3.6
167 info uninstall noble@0.3.6
168 verbose unbuild rmStuff noble@0.3.6 from /Users/rwaldron/ble/node_modules
169 info postuninstall noble@0.3.6
170 verbose gentlyRm vacuuming /Users/rwaldron/ble/node_modules/noble
171 silly gentlyRm purging /Users/rwaldron/ble/node_modules/noble
172 silly gentlyRm removing /Users/rwaldron/ble/node_modules
173 silly gentlyRm finished vacuuming up to /Users/rwaldron/ble
```

sandeepmistry, 2015-01-04:
@rwaldron looks like it's using the wrong header directory during the build. I'm also using OS X 10.10.1 and xcode-select version 2339.

Let's compare:

```
Sandeep-Mistrys-MacBook-Air:test smistry$ which gcc
/usr/bin/gcc
```
```
Sandeep-Mistrys-MacBook-Air:test smistry$ gcc -v
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/usr/include/c++/4.2.1
Apple LLVM version 6.0 (clang-600.0.56) (based on LLVM 3.5svn)
Target: x86_64-apple-darwin14.0.0
Thread model: posix
```

```
Sandeep-Mistrys-MacBook-Air:test smistry$ xcode-select -p
/Applications/Xcode.app/Contents/Developer
Sandeep-Mistrys-MacBook-Air:test smistry$ xcode-select -v
xcode-select version 2339
```

sandeepmistry, 2015-02-08:
@rwaldron ping ...

sandeepmistry, 2015-02-14:
Closing due to inactivity.

rwaldron, 2015-05-06:
Sorry, I don't know why I completely missed your response (from over a year ago). As I'm just seeing this now and writing you a frantic response, I will attempt to install and report back. Sorry about that!!

ykro, 2015-05-06:
I'm having the same issue. 
* OSX 10.10.3
* node v0.12.2
* xcode-select version 2339
```
$gcc -v
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 5.0 (clang-500.2.79) (based on LLVM 3.3svn)
Target: x86_64-apple-darwin14.3.0
Thread model: posix
```

sandeepmistry, 2015-05-06:
@rwaldron no worries, I'll re-open.

@ykro let's see if you can get some more verbose output for ```node-gyp rebuild``` when running ```npm install xpc-connection```. It would be great if we could see what compile flags are passed to ```gcc``` and compare with what's on my Mac.

Let's also compare ```g++ -v```:
```sh
 g++ --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/usr/include/c++/4.2.1
Apple LLVM version 6.1.0 (clang-602.0.49) (based on LLVM 3.6.0svn)
Target: x86_64-apple-darwin14.3.0
Thread model: posix
```

sandeepmistry, 2015-05-16:
@ykro @rwaldron have you had a chance to look into this?

rwaldron, 2015-05-27:
@sandeepmistry confirmed that this issue still exists. 

`g++ -v`:

```
$ g++ -v
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 5.0 (clang-500.2.79) (based on LLVM 3.3svn)
Target: x86_64-apple-darwin14.3.0
Thread model: posix
```

sandeepmistry, 2015-05-27:
@rwaldron are you at JSConf?

rwaldron, 2015-05-27:
@sandeepmistry I am. Once I compared the output from my `gcc -v` and yours, it occurred to me that I had a very out of date xcode, so I'm downloading that now.

ykro, 2015-05-27:
@sandeepmistry Everything worked after updating xcode

rwaldron, 2015-05-27:
No more failures! As soon as I have a chance to try an actual device, I will report back

sandeepmistry, 2015-05-27:
@rwaldron @ykro awesome!

reconbot, 2015-05-27:
I'm having issues installing too, and it looks like my xcode is up to date.

```
★ rolling-spider$ npm install noble
\
> xpc-connection@0.1.3 install /Users/wizard/src/rolling-spider/node_modules/noble/node_modules/xpc-connection
> node-gyp rebuild

  CXX(target) Release/obj.target/binding/src/XpcConnection.o
In file included from ../src/XpcConnection.cpp:7:
../src/XpcConnection.h:9:10: fatal error: 'xpc/xpc.h' file not found
#include <xpc/xpc.h>
         ^
1 error generated.
make: *** [Release/obj.target/binding/src/XpcConnection.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/Users/wizard/.nvm/versions/node/v0.12.2/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:269:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
gyp ERR! System Darwin 14.3.0
gyp ERR! command "node" "/Users/wizard/.nvm/versions/node/v0.12.2/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/wizard/src/rolling-spider/node_modules/noble/node_modules/xpc-connection
gyp ERR! node -v v0.12.2
gyp ERR! node-gyp -v v1.0.3
gyp ERR! not ok 
npm WARN optional dep failed, continuing xpc-connection@0.1.3

> noble@0.3.14 install /Users/wizard/src/rolling-spider/node_modules/noble
> node-gyp rebuild

  LINK(target) Release/hci-ble
ld: library not found for -lgcc_s.10.5
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [Release/hci-ble] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/Users/wizard/.nvm/versions/node/v0.12.2/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:269:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
gyp ERR! System Darwin 14.3.0
gyp ERR! command "node" "/Users/wizard/.nvm/versions/node/v0.12.2/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/wizard/src/rolling-spider/node_modules/noble
gyp ERR! node -v v0.12.2
gyp ERR! node-gyp -v v1.0.3
gyp ERR! not ok 
npm ERR! Darwin 14.3.0
npm ERR! argv "/Users/wizard/.nvm/versions/node/v0.12.2/bin/node" "/Users/wizard/.nvm/versions/node/v0.12.2/bin/npm" "install" "noble"
npm ERR! node v0.12.2
npm ERR! npm  v2.7.4
npm ERR! code ELIFECYCLE

npm ERR! noble@0.3.14 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the noble@0.3.14 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the noble package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls noble
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/wizard/src/rolling-spider/npm-debug.log
★ rolling-spider$ g++ -v
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator8.3.sdk/usr/include/c++/4.2.1
Apple LLVM version 6.1.0 (clang-602.0.53) (based on LLVM 3.6.0svn)
Target: x86_64-apple-darwin14.3.0
Thread model: posix
```

reconbot, 2015-05-27:
Scratch that, I was out of date, but not that out of date. However even after updating...

```
★ rolling-spider$ npm install noble
/
> xpc-connection@0.1.3 install /Users/wizard/src/rolling-spider/node_modules/noble/node_modules/xpc-connection
> node-gyp rebuild

  CXX(target) Release/obj.target/binding/src/XpcConnection.o
In file included from ../src/XpcConnection.cpp:7:
../src/XpcConnection.h:9:10: fatal error: 'xpc/xpc.h' file not found
#include <xpc/xpc.h>
         ^
1 error generated.
make: *** [Release/obj.target/binding/src/XpcConnection.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/Users/wizard/.nvm/versions/node/v0.12.2/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:269:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
gyp ERR! System Darwin 14.3.0
gyp ERR! command "node" "/Users/wizard/.nvm/versions/node/v0.12.2/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/wizard/src/rolling-spider/node_modules/noble/node_modules/xpc-connection
gyp ERR! node -v v0.12.2
gyp ERR! node-gyp -v v1.0.3
gyp ERR! not ok 
npm WARN optional dep failed, continuing xpc-connection@0.1.3

> noble@0.3.14 install /Users/wizard/src/rolling-spider/node_modules/noble
> node-gyp rebuild

  LINK(target) Release/hci-ble
ld: library not found for -lgcc_s.10.5
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [Release/hci-ble] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/Users/wizard/.nvm/versions/node/v0.12.2/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:269:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
gyp ERR! System Darwin 14.3.0
gyp ERR! command "node" "/Users/wizard/.nvm/versions/node/v0.12.2/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/wizard/src/rolling-spider/node_modules/noble
gyp ERR! node -v v0.12.2
gyp ERR! node-gyp -v v1.0.3
gyp ERR! not ok 
npm ERR! Darwin 14.3.0
npm ERR! argv "/Users/wizard/.nvm/versions/node/v0.12.2/bin/node" "/Users/wizard/.nvm/versions/node/v0.12.2/bin/npm" "install" "noble"
npm ERR! node v0.12.2
npm ERR! npm  v2.7.4
npm ERR! code ELIFECYCLE

npm ERR! noble@0.3.14 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the noble@0.3.14 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the noble package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls noble
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/wizard/src/rolling-spider/npm-debug.log
★ rolling-spider$ gcc -v
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator8.3.sdk/usr/include/c++/4.2.1
Apple LLVM version 6.1.0 (clang-602.0.53) (based on LLVM 3.6.0svn)
Target: x86_64-apple-darwin14.3.0
Thread model: posix
```

reconbot, 2015-05-27:
@sandeepmistry pointed out that my `gxx-include-dir` was wrong a reboot fixed it.

sandeepmistry, 2015-05-31:
Everyone seems happy, closing.

Lessions learned:
1) Use latest Xcode
2) Try rebooting if ```--with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator8.3.sdk/usr/include/c++/4.2.1```

closed 2015-05-31 by sandeepmistry