noble/noble #120: node-webkit 'Module did not self-register'

callum, 2015-01-08:
Re: https://github.com/sandeepmistry/noble/issues/113 https://github.com/sandeepmistry/node-xpc-connection/pull/6

Still not able to get it working with node-webkit 0.11.5 x64 on OS X and noble 0.3.7.

```
> node-webkit .
[27754:0108/123157:WARNING:dns_config_service_posix.cc(151)] dns_config has unhandled options!
[27755:0108/123157:ERROR:breakpad_mac.mm(238)] Breakpad initializaiton failed
[27756:0108/123157:ERROR:breakpad_mac.mm(238)] Breakpad initializaiton failed
[27754:0108/123157:INFO:CONSOLE(354)] "Uncaught Error: Module did not self-register.", source: module.js (354)
```

It could be an unrelated issue.

callum, 2015-01-08:
Working by rebuilding xpc-connection with nw-gyp. Is that expected?

sandeepmistry, 2015-01-09:
@francishart @mhle any ideas on this?

francishart, 2015-01-13:
@sandeepmistry we haven't experience this in our use of noble so I am not sure what would be causing it.

sandeepmistry, 2015-01-17:
@callum @francishart @mhle

Let's compare the steps you guys are using to setup/install nw.js (node-webkit) and noble.

francishart, 2015-01-21:
@mhle is the best one to talk to about this since he set it all up on our side. We do have a script that runs nw-gyp rebuild which does seem to match with what @callum is saying fixes it:
```
if [ "$OS" = "Darwin" ]; then
			nw-gyp rebuild --target="$NWVER"
			cd node_modules/xpc-connection
			nw-gyp rebuild --target="$NWVER"
			cd -
elif [ "$OS" = "Linux" ]; then
			nw-gyp rebuild --target="$NWVER"
			echo "Root permissions needed to set permissions on BLE-related binary"
			sudo chown root:root build/Release/*-ble
			sudo chmod 4775 build/Release/*-ble
```
Hope that helps!

sandeepmistry, 2015-01-24:
@francishart  @mhle thanks for the clarification, that lines up with what I've read about on nw.js and nw-gyp.

You might be able to create an ``nw-noble`` package which depends on noble and runs the above post install.

closed 2015-01-24 by sandeepmistry