noble/noble #121: Running without UID=0

ecksun, 2015-01-14:
I have long been annoyed that you require root to run anything with noble and I would like to be able go get away from it.

With that in mind I was able to figure out that if you give the binary/process the CAP_NET_RAW capability you do not need to run it as root.

For example:
$ ./hci-ble
adapterState unauthorized
$ sudo setcap CAP_NET_RAW=eip hci-ble  
$ ./hci-ble 
adapterState poweredOn

I don't think this is anything that can be moved into noble, but at the very least we can change the error-message with information about CAP_NET_RAW, perhaps mentioning what this is on a wiki page?

sandeepmistry, 2015-01-17:
@ecksun good suggestion!

Have you looked int creating a node.js utility script to run that during or after a build manually?

If you can't get that working, please submit a pull request for a change to the error message and instructions in the read me.

ecksun, 2015-01-17:
I haven't looked into it any further, no, but Im not sure how any node.js code would help. To change capabilities you need to be UID=0, I guess you would create a postgyp (Im unfamiliar how the npm build system interacts with compiling c-code) script which does this.

On the other hand, I would personally be very worried if my installation of noble all of a suden required me to sudo.

Basically, I dont have any good idea on how to make the capability change fit into noble nicely, or any node.js/npm package for that matter (which is why I created the issue, to see if anyone else had better ideas).

One of my problems is also that Im not building noble on the same machine as Im running it, which means I will have to make the setcap part of my deploy pipeline instead.

sandeepmistry, 2015-01-24:
@ecksun agreed.

I'm just thinking it would be nice to have a single line command users could run. Which would find ``hci-ble`` recursively in the current directory and run ``setcap`` on it.  For example, if someone is using ``node-sensortag`` which depends on noble, it would suggest running ``node`` as ``sudo`` or some other magic command that runs ``setcap``.

schnippe, 2015-02-02:
I run 

    find -executable -path '*module/noble/build/Release/hci-ble' -exec echo sudo setcap 'cap_net_raw+eip' '{}'

sandeepmistry, 2015-02-03:
@schnippe awesome suggestion!

I've trimmed it down to:

```sh
sudo setcap 'cap_net_raw+eip' `find -executable -path '*build/Release/hci-ble' | xargs`
```

Do you have any ideas on how to make the command "shorter"? If not, I can update the warning with it.

cc: @ecksun

schnippe, 2015-02-03:
@sandeepmistry thx!

setcap does not support multiple targets. 

    find -path '*noble*Release/hci-ble' -exec sudo setcap cap_net_raw+eip '{}' \;

is the shortest I can come up with.

It seems more informative  to write 
    No permisson: hci-ble is missing  cap_net_raw capabilties. Please see noble/README.md or run as root.

ecksun, 2015-02-03:
I aggree with @schnippe, having the details in the README allows us to be much more verbose about what the actual problem is and solutions for it.

sandeepmistry, 2015-02-04:
@ecksun @schnippe thanks for the feedback/suggestion, I've incorporated into https://github.com/sandeepmistry/noble/commit/1ea45c7cff19fa31da98833cbc28c630eb65b466

Let me know what you think, then I can publish to npm and also port to bleno.

ecksun, 2015-02-04:
Looks good, however the capability is actually only required for BLE scanning, not ordinary bluetooth scanning, if I remember correctly

schnippe, 2015-02-04:
@ecksun @sandeepmistry I agree. Please add something like:
To edit capabilities libcap my be needed. Install with apt: `sudo apt-get install libcap2-bin` or with yum: `su -c \'yum install libcap2-bin\'`

sandeepmistry, 2015-02-05:
@ecksun @schnippe are you good with 5a9df56f5f6931e1a7537265c06fc3b09e7a69e8? Any more feedback?

schnippe, 2015-02-05:
@sandeepmistry 
all good, thanks a lot!

ecksun, 2015-02-05:
@sandeepmistry Looks good, thank you :)

closed 2015-02-05 by ecksun

sandeepmistry, 2015-02-06:
@ecksun @schnippe no problem, thanks for the feedback.

Version 0.3.8 has been published to npm with the changes. I'll also port to bleno over the next few days.