noble/noble #122: no txPowerLevel on OS X Yosemite

pwldp, 2015-01-15:
Helo,
txPowerLevel for peripheral is undefined on OS X Yosemite on Mac Pro.
This is feature or error?

sandeepmistry, 2015-01-17:
@pwldp please run with debug on to see what CoreBluetooth is reporting for advertisement data, and provide the output:

```sh
DEBUG=yosemite-bindings node <your file>.js
```

noble can only report the data CoreBluetooth/blued provides in the discover event. Was this working for you on older versions of OS X?

pwldp, 2015-01-19:
I run script ```DEBUG=yosemite-bindings node txPowerLevel.js```:
```
var noble = require('noble');

noble.on('stateChange', function(state) {
  if (state === 'poweredOn') {
    noble.startScanning([], true);
  } else {
    noble.stopScanning();
  }
});

noble.on('discover', function(peripheral) {
  console.log('peripheral discovered (' + peripheral.uuid+ '):');
  console.log("txPowerLevel: ",peripheral.advertisement.txPowerLevel );
  console.log();
});
```

And got debug:

```
bash-3.2DEBUG=yosemite-bindings node txPowerLevel.js 
  yosemite-bindings sendCBMsg: 1, {
  "kCBMsgArgName": "node-1421657847444",
  "kCBMsgArgOptions": {
    "kCBInitOptionShowPowerAlert": 0
  },
  "kCBMsgArgType": 0
} +0ms
  yosemite-bindings xpcEvent: {
  "kCBMsgId": 6,
  "kCBMsgArgs": {
    "kCBMsgArgState": 5
  }
} +4ms
  yosemite-bindings state change poweredOn +0ms
  yosemite-bindings sendCBMsg: 29, {
  "kCBMsgArgOptions": {
    "kCBScanOptionAllowDuplicates": 1
  },
  "kCBMsgArgUUIDs": []
} +0ms
  yosemite-bindings xpcEvent: {
  "kCBMsgId": 37,
  "kCBMsgArgs": {
    "kCBMsgArgAdvertisementData": {
      "kCBAdvDataManufacturerData": [
        76,
        0,
        2,
        21,
        235,
        239,
        208,
        131,
        112,
        162,
        71,
        200,
        152,
        55,
        231,
        181,
        99,
        77,
        245,
        36,
        0,
        1,
        0,
        2,
        203,
        84
      ],
      "kCBAdvDataIsConnectable": 0,
      "kCBAdvDataAppleMfgData": {
        "kCBScanOptionAppleFilterPuckType": 2
      }
    },
    "kCBMsgArgRssi": -49,
    "kCBMsgArgDeviceUUID": [
      35,
      191,
      203,
      8,
      14,
      99,
      65,
      156,
      172,
      66,
      227,
      196,
      13,
      176,
      72,
      234
    ]
  }
} +6s
  yosemite-bindings peripheral 23bfcb080e63419cac42e3c40db048ea discovered +1ms
peripheral discovered (23bfcb080e63419cac42e3c40db048ea):
txPowerLevel:  undefined

  yosemite-bindings xpcEvent: {
  "kCBMsgId": 37,
  "kCBMsgArgs": {
    "kCBMsgArgAdvertisementData": {
      "kCBAdvDataManufacturerData": [
        76,
        0,
        2,
        21,
        235,
        239,
        208,
        131,
        112,
        162,
        71,
        200,
        152,
        55,
        231,
        181,
        99,
        77,
        245,
        36,
        0,
        1,
        0,
        2,
        203,
        84
      ],
      "kCBAdvDataIsConnectable": 0,
      "kCBAdvDataAppleMfgData": {
        "kCBScanOptionAppleFilterPuckType": 2
      }
    },
    "kCBMsgArgRssi": -54,
    "kCBMsgArgDeviceUUID": [
      35,
      191,
      203,
      8,
      14,
      99,
      65,
      156,
      172,
      66,
      227,
      196,
      13,
      176,
      72,
      234
    ]
  }
} +1s
  yosemite-bindings peripheral 23bfcb080e63419cac42e3c40db048ea discovered +0ms
peripheral discovered (23bfcb080e63419cac42e3c40db048ea):
txPowerLevel:  undefined
```

sandeepmistry, 2015-01-24:
@pwldp it doesn't look like your device is advertising a TX power level, so noble reports ``undefined`` - this is expected behaviour. Btw, what device are you using?

I would expect a ``kCBAdvDataTxPowerLevel`` key in ``kCBMsgArgAdvertisementData``.

closed 2015-01-24 by sandeepmistry