noble/noble #124: Having trouble starting on Raspberry Pi

seandoyle, 2015-01-26:
Hi -

I've been playing with noble reading heart rate monitors (a Polar H7 and a FitBit Charge HR) on an Ubuntu laptop (3.18.2-031802-generic) using a CSR USB dongle. It's working pretty nicely.  Now I'm trying to get this same library working on the Raspberry Pi and I think I have some version mismatches in the files - btmon shows zero activity when I try to connect to the heart rate monitors. 

In the Raspberry Pi - I'm running the 3.18.3+ kernel and bluez 5.27. I can see the my devices via hcitool lescan.  But pretty much everything in noble is unresponsive and since there isn't an error I don't know how to proceed.

If I start btmon I see:
 sudo btmon
Bluetooth monitor ver 5.27
= New Index: 00:1A:7D:DA:71:13 (BR/EDR,USB,hci0)                [hci0] 0.111256

but when I run your test.js there is no additional output. If I run sudo hcitool lescan there is a lot of activity in btmon. What is the best path to dig to get better diagnostic information? Thanks!

Sean

sandeepmistry, 2015-01-28:
@seandoyle let's start with running in debug mode, and see what the output is:

```sh
sudo DEBUG=hci-ble,l2cap-ble test.js
```

seandoyle, 2015-01-28:
Thanks! I've put the output below together with hciconfig and a hcitool lescan. 

$ sudo hciconfig -a
hci0:	Type: BR/EDR  Bus: USB
	BD Address: 00:1A:7D:DA:71:13  ACL MTU: 310:10  SCO MTU: 64:8
	UP RUNNING 
	RX bytes:1158 acl:0 sco:0 events:63 errors:0
	TX bytes:1046 acl:0 sco:0 commands:63 errors:0
	Features: 0xff 0xff 0x8f 0xfe 0xdb 0xff 0x5b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH HOLD SNIFF PARK 
	Link mode: SLAVE ACCEPT 
	Name: 'raspberrypi-0'
	Class: 0x000000
	Service Classes: Unspecified
	Device Class: Miscellaneous, 
	HCI Version: 4.0 (0x6)  Revision: 0x22bb
	LMP Version: 4.0 (0x6)  Subversion: 0x22bb
	Manufacturer: Cambridge Silicon Radio (10)

$ sudo DEBUG=hci-ble,l2cap-ble node test.js
  hci-ble hciBle = /home/pi/proxome/node/noble/lib/linux/../../build/Release/hci-ble +0ms
bindings initialized to [object Object]
noble
  hci-ble close = -1 +392ms

$ sudo hcitool lescan
LE Scan ...
D0:B1:6C:D0:BD:9A (unknown)
D0:B1:6C:D0:BD:9A Charge HR
00:22:D0:48:67:F7 (unknown)
00:22:D0:48:67:F7 Polar H7 4867F714

sandeepmistry, 2015-01-28:
@seandoyle it's a bit strange the spawned ```hci-ble``` process ends after 392ms, it's suppose to run as long as node is running.

Things to try next:

1) ```sudo killall hci-ble```, then try the example advertisement-discovery.js

2) ```sudo hciconfig hci0 down && sudo hciconfig hci0 up```, then retry the example advertisement-discovery.js

3) Run ```sudo /home/pi/proxome/node/noble/lib/linux/../../build/Release/hci-ble```, I want to make sure runs by itself

FYI, I'm running ```3.12.33+``` on my Pi, with BlueZ 4.99, I can upgrade it later in the week if needed ...

seandoyle, 2015-01-28:
OK - I see the problem and I think that the main issue is that I'm very new to node. There was no build directory. I had cloned the files from github but I couldn't run a 'npm install noble' in this directory - I get the message 'npm WARN install Refusing to install noble as a dependency of itself
'.
So - I did this in another directory and saw that it did the compilation - but when I run test.js in the files from github it's loading the local copy of noble & can't find the build. 

I think this is solved by just keeping the files that I run in node to be out of the directory created when I cloned the code. 

Sorry for bothering you about this - I'm new at node and didn't quite connect the dots.  I can get the heartrate just fine now. Thanks again!

sandeepmistry, 2015-01-28:
@seandoyle no worries

closed 2015-01-28 by sandeepmistry