noble/noble #125: Difficulties connecting to device on Yosemite

stewart, 2015-01-26:
Hi. Having some difficulty connecting to a device on OS X Yosemite. The device in question is an [Ollie](http://www.gosphero.com/ollie/).

Trying with latest Noble release, along with source. Checking down the tree shows the [correct method/message](https://github.com/sandeepmistry/noble/blob/master/lib/mac/yosemite.js#L188-L193) is being called/sent, but the callback is not triggered.

This prevents any work being down with the device.

I'm not sure if this is Ollie-specific or not, or has to do with recent changes to Noble. I was able to successfully use Noble in December with a WICEDSense, but unfortunately don't have one on me to test with right now.

Here's the script I'm using to test:

```javascript
var noble = require("noble");

var UUID = "4381ae299c884ea99c52065c51340de3",
    peripheral;

console.log("starting to scan")
noble.startScanning();

noble.on("discover", function(p) {
  if (p.uuid === UUID) {
    console.log("found pheripheral");
    peripheral = p;

    console.log("stopping scan");
    noble.stopScanning();
  }
});

noble.on("scanStop", function() {
  console.log("connecting to peripheral", peripheral.uuid);

  peripheral.connect(function(err) {
    console.log("connected");
  });
});
```

Standard result:

    $ node script.js
    starting to scan
    found pheripheral
    stopping scan
    connecting to peripheral 4381ae299c884ea99c52065c51340de3

With debug enabled:

    $ DEBUG=* node script.js
      yosemite-bindings sendCBMsg: 1, {
      "kCBMsgArgName": "node-1422315778549",
      "kCBMsgArgOptions": {
        "kCBInitOptionShowPowerAlert": 0
      },
      "kCBMsgArgType": 0
    } +0ms
    starting to scan
      yosemite-bindings sendCBMsg: 29, {
      "kCBMsgArgOptions": {},
      "kCBMsgArgUUIDs": []
    } +6ms
      noble scanStart +0ms
      yosemite-bindings xpcEvent: {
      "kCBMsgId": 6,
      "kCBMsgArgs": {
        "kCBMsgArgState": 5
      }
    } +1ms
      yosemite-bindings state change poweredOn +0ms
      noble stateChange poweredOn +1ms
      yosemite-bindings xpcEvent: {
      "kCBMsgId": 37,
      "kCBMsgArgs": {
        "kCBMsgArgAdvertisementData": {
          "kCBAdvDataIsConnectable": 1,
          "kCBAdvDataTxPowerLevel": -10
        },
        "kCBMsgArgRssi": -66,
        "kCBMsgArgDeviceUUID": [
          67,
          129,
          174,
          41,
          156,
          136,
          78,
          169,
          156,
          82,
          6,
          92,
          81,
          52,
          13,
          227
        ],
        "kCBMsgArgName": "2B-61D2"
      }
    } +1s
      yosemite-bindings peripheral 4381ae299c884ea99c52065c51340de3 discovered +1ms
    found pheripheral
    stopping scan
      yosemite-bindings sendCBMsg: 30, null +1ms
      noble scanStop +1s
    connecting to peripheral 4381ae299c884ea99c52065c51340de3
      yosemite-bindings sendCBMsg: 31, {
      "kCBMsgArgOptions": {
        "kCBConnectOptionNotifyOnDisconnection": 1
      },
      "kCBMsgArgDeviceUUID": [
        67,
        129,
        174,
        41,
        156,
        136,
        78,
        169,
        156,
        82,
        6,
        92,
        81,
        52,
        13,
        227
      ]
    } +1ms
      yosemite-bindings xpcEvent: {
      "kCBMsgId": 37,
      "kCBMsgArgs": {
        "kCBMsgArgAdvertisementData": {
          "kCBAdvDataIsConnectable": 1,
          "kCBAdvDataServiceUUIDs": [
            [
              34,
              187,
              116,
              111,
              43,
              160,
              117,
              84,
              45,
              111,
              114,
              101,
              104,
              112,
              83,
              39
            ]
          ],
          "kCBAdvDataLocalName": "2B-61D2"
        },
        "kCBMsgArgRssi": -66,
        "kCBMsgArgDeviceUUID": [
          67,
          129,
          174,
          41,
          156,
          136,
          78,
          169,
          156,
          82,
          6,
          92,
          81,
          52,
          13,
          227
        ],
        "kCBMsgArgName": "2B-61D2"
      }
    } +0ms
      yosemite-bindings peripheral 4381ae299c884ea99c52065c51340de3 discovered +0ms

Let me know if there's any more details I can provide, or any further tests I can do.

sandeepmistry, 2015-01-28:
@stewart looks like OS X is not reporting the expected connected event (kCBMsgId 38).

Things to try out next:

1) Try connecting to it via LightBlue for OS X or another CoreBluetooth OS X app

2) Try with older version of noble (0.3.6)

3) Capture data using Hardware IO Tools packet logger (see https://github.com/sandeepmistry/noble/issues/102#issuecomment-64121038 and https://github.com/sandeepmistry/noble/issues/102#issuecomment-64995057)

stewart, 2015-02-04:
Hi @sandeepmistry, sorry for the delay - starting to investigate this now.

I've tried using previous versions of Noble, with similar results.

More interestingly, LightBlue does something similar to Noble, in that it just automatically disconnects after a duration when attempting to load provided services. Poking at this with PacketLogger reveals the Ollie is triggering the disconnect, possibly after a timeout duration.

[image]

I've also taken a screenshot of what appeared to be the pertinent filtered packets from PacketLogger, and [exported the packets captured during a connection attempt][file]. Unsure if the file will be of any help.

[image]

[file]: http://share.stwrt.ca/snoop.pklg

Let me know if there's any more data I can provide, but I'm starting to suspect it's an underlying issue with the Ollie, Yosemite's BLE stack, or both.

sandeepmistry, 2015-02-04:
@stewart probably the Ollie, it terminates after responding to the "connection parameter request".

It's also weird that it returns "read not permitted" when OS X tries to read the device name (0x2a00).

Have you tried it using noble on Linux, or on Android/iOS?

(Closing the issue, as it's not specific to noble, but feel ask Ollie BLE related questions here or via email).

closed 2015-02-04 by sandeepmistry

deadprogram, 2015-02-15:
Hi, @sandeepmistry I think this is not just Ollie specific, since several people using the MiP seem to have the same problem: https://github.com/hybridgroup/cylon-mip/issues/1

sandeepmistry, 2015-02-22:
@deadprogram do the packet logger captures look the same when using LightBlue? I wonder if both devices use the same or similar BLE chips.

deadprogram, 2015-02-22:
I was wondering that exact same thing. Let me ping the issue reporters, and see if I can get that info for analysis.