noble/noble #127: Unable to read data from iPhone (Intel Edison/Raspberry Pi)

alexeichhorn, 2015-01-27:
Hi,

I want to get the battery level of my iPhone with Bluetooth LE with my Intel Edison using noble. I want to read the data of characteristic 2A19 in service 180F (preinstalled in iOS). Now I could connect to the peripheral (iPhone) and could detect the service (180F) and the characteristic. But when I want to read the data of this characteristic, I get no respond. No error too. 
```characteristic.read(function(error, data) {
console.log("data is here");
});```

I thought that I made a programming mistake, so I took the code from the Getting started site (https://github.com/sandeepmistry/noble/wiki/Getting-started) and I changed the two UUID's. There's the exact same problem again. The characteristic was found, but it couldn't read the data.

I made this on my Intel Edison (all software up to date) and on my Raspberry Pi. In LightBlue on my mac I could read the value, so it shouldn't be a problem on iOS-side. 

Full code:
```javascript
var noble = require('noble');

noble.startScanning([], false);

noble.on('discover', function(peripheral) {
  peripheral.connect(function(error) {
    console.log('connected to peripheral: ' + peripheral.uuid);
    peripheral.discoverServices(['180f'], function(error, services) {
      var deviceInformationService = services[0];
      console.log('discovered device information service');

      deviceInformationService.discoverCharacteristics(['2a19'], function(error, characteristics) {
        var manufacturerNameCharacteristic = characteristics[0];
        console.log('discovered manufacturer name characteristic');

        manufacturerNameCharacteristic.read(function(error, data) {
          // data is a buffer
          console.log('data is here');
        });
      });
    });
  });
});
```

Thank you for your help and for this library. 
P.S. Sorry for the bad english. I'm Swiss ;)

sandeepmistry, 2015-01-28:
@alexeichhorn is your Mac paired with your iPhone?

Just tried connecting my Nexus 7 (using nRF Master Control Panel app) to iOS (running LightBlue in peripheral mode), and was presented with a pairing prompt.

noble on Linux relies on BlueZ's bluetoothd for pairing. I'm haven't tried to setup pairing with my Intel Edison and iPhone, so don't have any instructions for this yet.

If you run with debug, and provide logs I can confirm this:
```sh
sudo DEBUG=l2cap-ble node <your files>.js
```

alexeichhorn, 2015-01-28:
The log is:
```
  l2cap-ble 7A:80:4B:2D:92:98: l2capBle = /home/root/node_modules/noble/lib/linux/../../build/Release/l2cap-ble +0ms
  l2cap-ble 7A:80:4B:2D:92:98: buffer = "bind success\n" +39ms
  l2cap-ble 7A:80:4B:2D:92:98: line = bind success +2ms
  l2cap-ble 7A:80:4B:2D:92:98: buffer = "connect success\n" +103ms
  l2cap-ble 7A:80:4B:2D:92:98: line = connect success +2ms
  l2cap-ble 7A:80:4B:2D:92:98: write: 020001 +6ms
connected to peripheral: 7a804b2d9298
  l2cap-ble 7A:80:4B:2D:92:98: buffer = "data 100100ffff0028" +394ms
  l2cap-ble 7A:80:4B:2D:92:98: buffer = "data 100100ffff0028\n" +0ms
  l2cap-ble 7A:80:4B:2D:92:98: line = data 100100ffff0028 +1ms
  l2cap-ble 7A:80:4B:2D:92:98: ignoring request/command ... +0ms
  l2cap-ble 7A:80:4B:2D:92:98: buffer = "data 039e00\n" +66ms
  l2cap-ble 7A:80:4B:2D:92:98: line = data 039e00 +1ms
  l2cap-ble 7A:80:4B:2D:92:98: new MTU is 158 +1ms
  l2cap-ble 7A:80:4B:2D:92:98: write: 100100ffff0028 +1ms
  l2cap-ble 7A:80:4B:2D:92:98: buffer = "data 110601000500001806" +402ms
  l2cap-ble 7A:80:4B:2D:92:98: buffer = "data 1106010005000018060009000118\n" +1ms
  l2cap-ble 7A:80:4B:2D:92:98: line = data 1106010005000018060009000118 +1ms
  l2cap-ble 7A:80:4B:2D:92:98: write: 100a00ffff0028 +2ms
  l2cap-ble 7A:80:4B:2D:92:98: buffer = "data 11140a000e006643ae10" +200ms
  l2cap-ble 7A:80:4B:2D:92:98: buffer = "data 11140a000e006643ae107948f8a59145b4bb781e61d00f001800d0002d121e4b0fa4994eceb531f40579\n" +1ms
  l2cap-ble 7A:80:4B:2D:92:98: line = data 11140a000e006643ae107948f8a59145b4bb781e61d00f001800d0002d121e4b0fa4994eceb531f40579 +0ms
  l2cap-ble 7A:80:4B:2D:92:98: write: 101900ffff0028 +3ms
  l2cap-ble 7A:80:4B:2D:92:98: buffer = "data 110619001c" +197ms
  l2cap-ble 7A:80:4B:2D:92:98: buffer = "data 110619001c000f181d0022000518\n" +1ms
  l2cap-ble 7A:80:4B:2D:92:98: line = data 110619001c000f181d0022000518 +1ms
  l2cap-ble 7A:80:4B:2D:92:98: write: 102300ffff0028 +0ms
  l2cap-ble 7A:80:4B:2D:92:98: buffer = "data 111423002e00dcf855ad02" +201ms
  l2cap-ble 7A:80:4B:2D:92:98: buffer = "data 111423002e00dcf855ad02c5f48e3a43360f2b50d389\n" +1ms
  l2cap-ble 7A:80:4B:2D:92:98: line = data 111423002e00dcf855ad02c5f48e3a43360f2b50d389 +0ms
  l2cap-ble 7A:80:4B:2D:92:98: write: 102f00ffff0028 +1ms
  l2cap-ble 7A:80:4B:2D:92:98: buffer = "data 01102f000a\n" +268ms
  l2cap-ble 7A:80:4B:2D:92:98: line = data 01102f000a +0ms
discovered device information service
  l2cap-ble 7A:80:4B:2D:92:98: write: 0819001c000328 +9ms
  l2cap-ble 7A:80:4B:2D:92:98: buffer = "data 09071a00121b00192a\n" +193ms
  l2cap-ble 7A:80:4B:2D:92:98: line = data 09071a00121b00192a +1ms
  l2cap-ble 7A:80:4B:2D:92:98: write: 081c001c000328 +1ms
  l2cap-ble 7A:80:4B:2D:92:98: buffer = "data 01081c000a\n" +201ms
  l2cap-ble 7A:80:4B:2D:92:98: line = data 01081c000a +0ms
discovered battery level characteristic
  l2cap-ble 7A:80:4B:2D:92:98: write: 0a1b00 +6ms
  l2cap-ble 7A:80:4B:2D:92:98: buffer = "data 010a1b0005\n" +197ms
  l2cap-ble 7A:80:4B:2D:92:98: line = data 010a1b0005 +0ms
  l2cap-ble 7A:80:4B:2D:92:98: buffer = "security = medium\n" +2ms
  l2cap-ble 7A:80:4B:2D:92:98: line = security = medium +0ms
  l2cap-ble 7A:80:4B:2D:92:98: write: 0a1b00 +1ms
```

about 20 seconds later: 
```
  l2cap-ble 7A:80:4B:2D:92:98: buffer = "disconnect\n" +28s
  l2cap-ble 7A:80:4B:2D:92:98: line = disconnect +1ms
  l2cap-ble 7A:80:4B:2D:92:98: close = 0 +18ms
```

I don't see any informations in these logs. I hope you'll see something ;) Thank you :)

sandeepmistry, 2015-01-29:
@alexeichhorn definitely looks like a pairing issue.

Did you get a chance to check if your Mac is paired with you iPhone?

alexeichhorn, 2015-01-29:
@sandeepmistry I could connect from my mac with LightBlue to my iPhone and read the battery level. But it disconnected the iPhone after about 20 seconds (like Intel Edison does). Here's the log:
```
- Connected to device UUID: 614B6F4B-8D03-41D7-9889-CCD0011B8304 Named: iPhone 6 von Alexander Eichho
- UUID 2A19 Read Value: 0x1D
- Disconnected from device UUID: 614B6F4B-8D03-41D7-9889-CCD0011B8304 Named: iPhone 6 von Alexander Eichho
```

sandeepmistry, 2015-01-30:
@alexeichhorn from the log snippet above:

1) noble sends handle 0x001b
```
  l2cap-ble 7A:80:4B:2D:92:98: write: 0a1b00 +6ms
```

2) iOS device responds with error ATT_ECODE_AUTHENTICATION
```
  l2cap-ble 7A:80:4B:2D:92:98: buffer = "data 010a1b0005\n" +197ms
  l2cap-ble 7A:80:4B:2D:92:98: line = data 010a1b0005 +0ms
```

3) noble upgrades security to medium
```
  l2cap-ble 7A:80:4B:2D:92:98: buffer = "security = medium\n" +2ms
  l2cap-ble 7A:80:4B:2D:92:98: line = security = medium +0ms
```

4) noble retries read handle 0x001b again
```
  l2cap-ble 7A:80:4B:2D:92:98: write: 0a1b00 +1ms
```

5) iOS device connects

1 - 4 is what I would expect from noble. The lower Linux implementation of noble is completely different from OS X. However, I'm not sure what security level OS X uses (maybe high instead of medium?), I had some instability with Linux and high security, so I left it at medium.

You could try experimenting with high security,  in [l2cap-ble.c](https://github.com/sandeepmistry/noble/blob/master/src/l2cap-ble.c) change both ```BT_SECURITY_MEDIUM ```'s to ```BT_SECURITY_HIGH ```and run ```node-gyp rebuild``` (you might have to globally npm install node-gyp).

alexeichhorn, 2015-01-30:
@sandeepmistry Unfortunately it doesn't work. Still the same error. But I've tried to read a characteristic which I created on LightBlue (on iOS). This worked perfectly. It not even had to upgrade the security. 
Probably only the built-in services of iOS have problems. With ANCS (node-ancs) I've the same problems.

I've found something in an Apple document: 
> iOS devices support Encryption Mode 3, Security Mode 4, and Service Level 1 connections. 

Is Security Mode 4 equal to Security Level High?

alexeichhorn, 2015-01-31:
@sandeepmistry I found out, that the problem occurs only in iOS 8. Unfortunately I couldn't find the reason, but I found an ANCS-Library for Arduino. In an update you could see the changes for iOS 8. It says: 
> I think I fixed the things that were messing up iOS 8. The forced
disconnect right after connecting was a problem. I think there were
some things with services.h. I extended the delay before security was
required.

Can you change the delay in noble? I searched for that, but couldn't find anything.

Here's the update description and more:
https://github.com/robotastic/ANCS-Library/commit/dff14e98cf0374b38902777bb0d0d7412c7eba45?diff=split

sandeepmistry, 2015-02-08:
@alexeichhorn 

Adding a delay before retrying does seem to help, it allows you to hit the "Pair" button on the iOS dialog.

Replace the following in ```src/linux/l2cap.js```:

```javascript
   } else if ((found = line.match(/^security = (.*)$/))) {
      this._security = found[1];           
                                           
      if (this._security === 'medium') {   
        // re-issue the current command    
        debug(this._address + ': write: ' + this._currentCommand.buffer.toString('hex'));
        this._l2capBle.stdin.write(this._currentCommand.buffer.toString('hex') + '\n');
      } else {                             
        this.upgradeSecurity();            
      }          
```

with:
```javascript
   } else if ((found = line.match(/^security = (.*)$/))) {
      this._security = found[1];           
                                           
      if (this._security === 'medium') {   
        // re-issue the current command    
        setTimeout(function() {            
          debug(this._address + ': write: ' + this._currentCommand.buffer.toString('hex'));
          this._l2capBle.stdin.write(this._currentCommand.buffer.toString('hex') + '\n');
        }.bind(this), 10000);               
      } else {                             
        this.upgradeSecurity();            
      }          
```

My only issue with putting this is in master is the 10s retry time period, it doesn't guarantee the iOS user has hit the pair button. Also, if you are already paired with the peripheral (iOS or other), there's an ```n``` second delay, even though there doesn't need to be.

A better approach is to query the connection to see when security has actually been upgraded, which is what I thought the following sequence in ```src/l2cap-ble.c``` was accomplishing already:

```c 
        struct bt_security btSecurity;
        socklen_t btSecurityLen;

        memset(&btSecurity, 0, sizeof(btSecurity));
        btSecurity.level = BT_SECURITY_MEDIUM;

        setsockopt(l2capSock, SOL_BLUETOOTH, BT_SECURITY, &btSecurity, sizeof(btSecurity));

        getsockopt(l2capSock, SOL_BLUETOOTH, BT_SECURITY, &btSecurity, &btSecurityLen);

        printf("security = %s\n", (BT_SECURITY_MEDIUM == btSecurity.level) ? "medium" : "low");
```

sandeepmistry, 2015-02-08:
@alexeichhorn 20e9af815cc5f55c340352489666b31ced540f92 should help with this.

Could you install noble from master and test it out - ```npm install sandeepmistry/noble``` ... I had to click the "pair" button on the iOS dialog every time. Also, make sure ```bluetoothd``` is running on your RPi/Edison.

closed 2015-02-14 by sandeepmistry