noble/noble #128: Problem with the writeHandle

vitorleal, 2015-02-02:
Hi,
first off all congratulation with the great job with this project!

Im having problems when doing a writeHandle in a specific handle (0x11). Im using Arduino with the [HM-13](http://www.fasttech.com/product/1827700-hm-13-dual-mode-bluetooth-4-0-ble-spp-le-serial) BLE module. When i send some data it not reaching the Serial Monitor in the Arduino, if i use the Bluetooth 2 in this module i can read and write to the Arduino without a problem. Do you know what could it be? Inspecting with the PacketLogger app i can see and **Read By Type Response** that contains an **Malformed Response**.

[image]

Here my code code running with the DEBUG, im using a Mac with Yosemite by the way.

The piece of code that im trying to do is this:
```javascript
peripheral.writehandle(0x11, new Buffer('#PM1234\n'));
peripheral.readhandle(0x11, function (err, data) { 
  if (data) {
    var buf = new Buffer(data);
   console.log(buf.toString('utf8'));
  }
});
```

Thanks in advance!

```
 DEBUG=yosemite-bindings node noble.js

  yosemite-bindings sendCBMsg: 1, {
  "kCBMsgArgName": "node-1422877922487",
  "kCBMsgArgOptions": {
    "kCBInitOptionShowPowerAlert": 0
  },
  "kCBMsgArgType": 0
} +0ms
  yosemite-bindings xpcEvent: {
  "kCBMsgId": 6,
  "kCBMsgArgs": {
    "kCBMsgArgState": 5
  }
} +5ms
  yosemite-bindings state change poweredOn +0ms
  yosemite-bindings sendCBMsg: 29, {
  "kCBMsgArgOptions": {},
  "kCBMsgArgUUIDs": []
} +0ms
  yosemite-bindings xpcEvent: {
  "kCBMsgId": 37,
  "kCBMsgArgs": {
    "kCBMsgArgAdvertisementData": {
      "kCBAdvDataManufacturerData": [
        72,
        77,
        0,
        14,
        11,
        0,
        57,
        77
      ],
      "kCBAdvDataServiceData": [
        [
          176,
          0
        ],
        [
          0,
          0,
          0,
          0
        ]
      ],
      "kCBAdvDataServiceUUIDs": [
[... 567 lines cut ...]
  "kCBMsgId": 79,
  "kCBMsgArgs": {
    "kCBMsgArgDeviceUUID": [
      34,
      241,
      72,
      218,
      71,
      72,
      70,
      48,
      142,
      162,
      241,
      184,
      206,
      64,
      105,
      2
    ],
    "kCBMsgArgResult": 21,
    "kCBMsgArgDescriptorHandle": 17
  }
} +0ms
```

sandeepmistry, 2015-02-03:
@vitorleal even though noble provides ```writeHandle``` API's on OS X, you still need to go through service and characteristic discovery - then it still might have some hiccups.

I suggest you take a look at @xorbit's [node-hm-ble](https://www.npmjs.com/package/node-hm-ble)

vitorleal, 2015-02-03:
Thank's @sandeepmistry i will take a look at the hm-ble module.

Cheers!

closed 2015-02-03 by vitorleal