noble/noble #129: Noble stops listening

doganyazar, 2015-02-05:
Noble seems to stop listening after some non-deterministic time and the only solution I came up with is to call startScanning again (or pkill -SIGUSR2 hci-ble).
Do you have any ideas or suggestions?

sandeepmistry, 2015-02-06:
@doganyazar what kernel / Linux distro / BlueZ version are you using?

I haven't experienced this behaviour. I would suggest you:

1. add print outs in [hci-ble.c](https://github.com/sandeepmistry/noble/blob/master/src/hci-ble.c)
2. run ```node-gyp rebuild```
3. run with debug on: ```sudo DEBUG=hci-ble <file>.js```

I can help out more once you've done the above ... and provide output.

sandeepmistry, 2015-02-06:
@doganyazar also wanted to confirm, you are calling startScanning with duplicates set to true, right?
```javascript
noble.startScaning([], true);
```

doganyazar, 2015-02-09:
Yes, duplicates is set to true. 
I have experienced this behaviour with several BlueZ versions > 5.20 and Debian with kernel > 3.14.

I will dig into this as soon as possible and will try to narrow it down. 
Thanks

sandeepmistry, 2015-02-10:
@doganyazar sounds good ... it would be good to compare noble behaviour with:

```sh
hcitool lescan --duplicates
```

As they should be making the same calls to the Linux kernel.

sandeepmistry, 2015-03-07:
@doganyazar any updates on this?

doganyazar, 2015-03-07:
Sorry apparently I had completely forgotten this :) Just setup an environment for this so will start investigating now!

doganyazar, 2015-03-17:
It takes couple of days to replicate this and that is why I could not really find anything interesting up to now. But I actively continue checking the running machine. Most probably I need to dig into bluez more to figure out the reason.

doganyazar, 2015-03-22:
After all these trials, I could only repeat this problem with the Plugable Bluetooth adapter
(http://plugable.com/products/usb-bt4le). I think the dongle gets into some weird state after some time which causes this behaviour. 

I have also seen noble printing
console.log('noble warning: adapter does not support Bluetooth Low Energy (BLE, Bluetooth Smart).');
when this adapter was used in some occasions.

sandeepmistry, 2015-03-28:
@doganyazar do you have a set of reproducible steps?

doganyazar, 2015-04-02:
Nope I have no deterministic way of producing it. But I think it has nothing to do with noble since when it happens lescan does not work either giving the following error.

```
$ sudo hcitool lescan
Set scan parameters failed: Input/output error
```

sandeepmistry, 2015-04-03:
@doganyazar good to know. Can we close this issue?

closed 2015-04-03 by doganyazar

jmaxxz, 2016-10-12:
@doganyazar is there any way to detect this condition so scanning can automatically be restarted?

doganyazar, 2016-10-12:
@jmaxxz  I don't think that callback was ever triggered but I cannot try it now since I do not have the setup anymore. 
Our workaround was to call startScaning every 10 minutes or so. And it worked fine.

sandeepmistry, 2016-10-12:
https://github.com/sandeepmistry/noble/pull/301 will emit the scan stop if it's triggered by an external program.