noble/noble #133: The main Readme lists OSX prerequisite, the "Getting started" page doesn't

raykamp, 2015-02-18:
The main [Readme](https://github.com/sandeepmistry/noble/blob/master/README.md) lists Xcode as a requirement/prerequisite for OS X. The [getting started page](https://github.com/sandeepmistry/noble/wiki/Getting-started) doesn't. 

Is Xcode necessary? If so, it seems like it should be listed as an OS X specific prerequisite on the getting started page.

sandeepmistry, 2015-02-19:
@raykamp good catch, I haven't had a chance to try recently on a fresh OS X install without Xcode. Probably safer to assume it is needed.

Feel free to update the wiki page accordingly, I haven't really been keeping the getting started page in sync with the readme.

duffo64, 2015-02-25:
I can confirm that xcode command line tools are not enough. Trying to install noble calls for complete xcode package at some point (10.9 and 10.0).

sandeepmistry, 2015-03-01:
@duffo64 thanks for confirming.

@raykamp I've updated the Getting started guide according: https://github.com/sandeepmistry/noble/wiki/Getting-started/_compare/ca132b88e6bf25dbd215630c7ab0fbc622375b2d...fe2dba13ca7891d46cf7891689c76037b466fceb

closed 2015-03-01 by sandeepmistry

raykamp, 2015-03-01:
Thanks @sandeepmistry. What's with the change from requiring bluez 4.101 to requiring 4.99?

sandeepmistry, 2015-03-02:
@raykamp oops, just changed it to "4.99 and above".

4.101 was required at one point, because 4.99 did not have the address type (public, random) in the headers. I decided to just declare the correct struct in noble. This saves Raspberry Pi/Raspbian users from having to build/install BlueZ from source.

raykamp, 2015-03-02:
ok, thanks for explaining @sandeepmistry !

raykamp, 2015-03-02:
Oh, one last question regarding this. Does "4.99 and above" include only versions with "4" as the major version? Or is version 5.x of bluez also supported?

sandeepmistry, 2015-03-03:
@raykamp 5.x is also supported.

I mostly use 4.99 (Raspbian) and 4.101 (Ubuntu), however do have 5.x on my Intel Edison. In general, I would stick with whatever comes with the Linux distort you are using.