noble/noble #134: Not discovering on Ubuntu 14.10

whitingm, 2015-02-19:
Hi,

I am looking to use noble for a project I am running, I have had it running fine on OSX running on my MacBookPro, I really want it running on Linux, so I dual booted my Mac into Ubuntu 14.10 and installed Noble as per the instructions, however when I run either my project or the included test project, it never discovers any peripherals even though Ubuntu can see them in bluetooth manager. Any ideas? I have also installed on a Lenovo G50 and have exactly the same issue. On debugging I can see that Noble gets the adapter state as 'poweredon' and starts scanning but nothing is ever discovered. Help would be appreciated.

schnippe, 2015-02-19:
I have noble up and running with Ubuntu 14.04. 
try running with  `sudo DEBUG=* NOBLE_HCI_DEVICE_ID=1 node test.js`. Maybe your System is using the wrong adapter or your adapter is not compatible. Please paste result.

whitingm, 2015-02-19:
matt@matt-Lenovo-G50-30:~/node_modules/noble$ sudo DEBUG=* NOBLE_HCI_DEVICE_ID=0 node test.js
  hci-ble hciBle = /home/matt/node_modules/noble/lib/linux/../../build/Release/hci-ble +0ms
noble
  hci-ble buffer = "adapterState poweredOn\n" +26ms
  hci-ble line = adapterState poweredOn +1ms
  hci-ble adapterState = poweredOn +0ms
  noble stateChange poweredOn +0ms
on -> stateChange: poweredOn
  noble scanStart +1ms
on -> scanStart
  hci-ble buffer = "event 7C:D1:C3:2D:7B:26,public,02011a0bff4c000906001fc0a80011,-48\n" +65ms
  hci-ble line = event 7C:D1:C3:2D:7B:26,public,02011a0bff4c000906001fc0a80011,-48 +1ms
  hci-ble address = 7C:D1:C3:2D:7B:26 +0ms
  hci-ble addressType = public +1ms
  hci-ble eir = 02011a0bff4c000906001fc0a80011 +0ms
  hci-ble rssi = -48 +0ms
  hci-ble advertisement = {"manufacturerData":[76,0,9,6,0,31,192,168,0,17],"serviceData":[],"serviceUuids":[]} +2ms
  hci-ble buffer = "event 7C:D1:C3:33:53:99,public,02011a0bff4c0009060004c0a8002a,-10\n" +71ms
  hci-ble line = event 7C:D1:C3:33:53:99,public,02011a0bff4c0009060004c0a8002a,-10 +0ms
  hci-ble address = 7C:D1:C3:33:53:99 +1ms
  hci-ble addressType = public +0ms
  hci-ble eir = 02011a0bff4c0009060004c0a8002a +0ms
  hci-ble rssi = -10 +0ms
  hci-ble advertisement = {"manufacturerData":[76,0,9,6,0,4,192,168,0,42],"serviceData":[],"serviceUuids":[]} +0ms
  hci-ble buffer = "event E7:FF:DC:CF:2E:C3,random,0409746b7203194002020106020a0403033e0f09ff0000c32ecfdcffe7,-26\n" +2s
  hci-ble line = event E7:FF:DC:CF:2E:C3,random,0409746b7203194002020106020a0403033e0f09ff0000c32ecfdcffe7,-26 +0ms
  hci-ble address = E7:FF:DC:CF:2E:C3 +1ms
  hci-ble addressType = random +0ms
  hci-ble eir = 0409746b7203194002020106020a0403033e0f09ff0000c32ecfdcffe7 +1ms
  hci-ble rssi = -26 +0ms
  hci-ble advertisement = {"localName":"tkr","txPowerLevel":4,"manufacturerData":[0,0,195,46,207,220,255,231],"serviceData":[],"serviceUuids":["f3e"]} +6ms
  hci-ble buffer = "event DB:7B:EF:B8:FE:B8,random,0409746b7203194002020106020a0403033e0f09ff0000b8feb8ef7bdb,-88\n" +1s
  hci-ble line = event DB:7B:EF:B8:FE:B8,random,0409746b7203194002020106020a0403033e0f09ff0000b8feb8ef7bdb,-88 +0ms
  hci-ble address = DB:7B:EF:B8:FE:B8 +1ms
  hci-ble addressType = random +0ms
  hci-ble eir = 0409746b7203194002020106020a0403033e0f09ff0000b8feb8ef7bdb +0ms
  hci-ble rssi = -88 +1ms
  hci-ble advertisement = {"localName":"tkr","txPowerLevel":4,"manufacturerData":[0,0,184,254,184,239,123,219],"serviceData":[],"serviceUuids":["f3e"]} +2ms

It certainly looks like it is seeing the devices, however when I run test.js not in debug, i only get the following 

matt@matt-Lenovo-G50-30:~/node_modules/noble$ sudo node test.js
noble
on -> stateChange: poweredOn
on -> scanStart

and no other output

whitingm, 2015-02-19:
The adapter is the built in RealTek rtl8723be

whitingm, 2015-02-19:
I've just noticed if I force an adapter statechange by turning off bluetooth in Ubuntu and then back on, it then discovers a device???? But only one at a time ????

matt@matt-Lenovo-G50-30:~/node_modules/noble$ sudo node test.js
noble
on -> stateChange: poweredOn
on -> scanStart
on -> stateChange: poweredOff
on -> scanStop
on -> stateChange: poweredOn
on -> scanStart
on -> discover: {"uuid":"7cd1c32d7b26","advertisement":{"manufacturerData":[76,0,9,6,0,31,192,168,0,17],"serviceData":[],"serviceUuids":[]},"rssi":-50,"state":"disconnected"}
on -> scanStop
on -> connect
on -> RSSI update -96
on -> disconnect
on -> stateChange: poweredOff
on -> scanStop
on -> stateChange: poweredOn
on -> scanStart
on -> discover: {"uuid":"7cd1c3335399","advertisement":{"manufacturerData":[76,0,9,6,0,4,192,168,0,42],"serviceData":[],"serviceUuids":[]},"rssi":-18,"state":"disconnected"}
on -> scanStop
on -> connect
on -> RSSI update -96
on -> disconnect

schnippe, 2015-02-19:
Please try to change line 167 : 
` if (this._discoveries[address].count % 2 === 0) {` 
=>
 `if (this._discoveries[address].count % 1 === 0) {`

Does it help? Maybe a recent change is not ok for your device.

whitingm, 2015-02-19:
Fantastic, worked straight away. Now picking up all the devices in both my app and the test app as well as debugging using webstorm. Thanks for your help.

schnippe, 2015-02-19:
@sandeepmistry @whitingm 
Yeah, that was a smooth one. The power cycle gave it away.

closed 2015-02-19 by whitingm

sandeepmistry, 2015-02-20:
@schnippe thanks for helping out!

@whitingm noble expects the device to emit both an advertisement and scan response packet (hence the % 2). It looks like your device is only doing an advertisement.

You can start scanning with duplicates to run the stock noble as an alternative to patching it:

```javascript
noble.startScanning([], true);
```

Just curious what peripheral are you using?

whitingm, 2015-02-20:
I have 4 peripherals, two apple TV's (which I'm not interested in) and twoTRACKR tags. It could well be the Apple TV's that are not communicating as expected. I tried running with duplicates and it still wasn't showing when i was testing before. I have had it running all night and it works as expected, i will try a different peripheral and let you know if it continues to work.

sandeepmistry, 2015-02-22:
@whitingm thanks I was wondering what the device was using the Apple Company Identifier (0x004c).

```
02 01 1a - flags
0b ff 4c00 09 06  00 04 c0 a8 00 2a - manufacturer data
```

I've pushed a change to bypass the ```count % 2```, see ee3a571aa26e39e9c39bcbebe7459710b8c11c0b

```sh
sudo NOBLE_REPORT_ALL_HCI_EVENTS=1 node <your file>.js
``` 

It's intended to be used with devices that use a new random BT address everything advertisement and have scan response disabled.