noble/noble #136: callback not called after setting notify for characteristic

freehaha, 2015-02-26:
Hi,
After the patch 8fd052c one of my application stop working and I found that it stopped after sending `Characteristic.notify` and seems to be waiting for the ack from the device that never came (thus no callback is called and all the following commands halt in the queue). 
I'm not sure if that's because the device(Myo) didn't follow spec or is it a bug in noble but I don't have other devices to test so can't really tell...

sandeepmistry, 2015-02-27:
@freehaha interesting.

Can you please provide debug logs with and without the change?

```sh
sudo DEBUG=l2cap-ble node <file>.js
```

freehaha, 2015-03-05:
sorry for the late reply. the logs are here: 
https://gist.github.com/freehaha/1491e7912d365f2d44f4

my script basically connect to a device(Myo), do a service and characteristic discovery and read its firmware version by doing a `Char.read`. After that I try to enable notification(`Char.notify(true)`) on 4 characteristics and write some value to one of their descriptor(after a `discverDescriptors`), respectively.

So the problem happens after the log `version: 1.1.755 hardware: 2`, before `initialized`, which I believe hangs after calling `Char.notify`. If I remove the `Char.notify` in my script and with the post-patch version, it actually works fine but this is not the standard procedure of using the device..

Hope this help.

sandeepmistry, 2015-03-05:
@freehaha would you be able to share the code that outputted the debug logs?

freehaha, 2015-03-06:
The code is here, in coffee-script:
https://github.com/freehaha/myo4l/blob/master/src/lib/myo.coffee

`notify` is in method `setNotification` at line 111.

sandeepmistry, 2015-03-06:
Here's the difference after service + characteristic discovery:

Before:
```
Thu, 05 Mar 2015 11:20:41 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 050124000229
Thu, 05 Mar 2015 11:20:41 GMT l2cap-ble D3:89:EF:01:16:D8: write: 1224000200
Thu, 05 Mar 2015 11:20:41 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 13
Thu, 05 Mar 2015 11:20:41 GMT l2cap-ble D3:89:EF:01:16:D8: write: 08220024000229
Thu, 05 Mar 2015 11:20:42 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 090424000200
Thu, 05 Mar 2015 11:20:42 GMT l2cap-ble D3:89:EF:01:16:D8: write: 1224000200
Thu, 05 Mar 2015 11:20:42 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 13
Thu, 05 Mar 2015 11:20:42 GMT l2cap-ble D3:89:EF:01:16:D8: write: 0428002800
Thu, 05 Mar 2015 11:20:42 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 050128000229
Thu, 05 Mar 2015 11:20:42 GMT l2cap-ble D3:89:EF:01:16:D8: write: 1228000100
Thu, 05 Mar 2015 11:20:42 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 13
Thu, 05 Mar 2015 11:20:42 GMT l2cap-ble D3:89:EF:01:16:D8: write: 08260028000229
Thu, 05 Mar 2015 11:20:42 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 090428000100
Thu, 05 Mar 2015 11:20:42 GMT l2cap-ble D3:89:EF:01:16:D8: write: 1228000100
Thu, 05 Mar 2015 11:20:42 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 13
Thu, 05 Mar 2015 11:20:42 GMT l2cap-ble D3:89:EF:01:16:D8: write: 041d001d00
Thu, 05 Mar 2015 11:20:42 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 05011d000229
Thu, 05 Mar 2015 11:20:42 GMT l2cap-ble D3:89:EF:01:16:D8: write: 121d000100
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 13
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: write: 081b001d000229
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 09041d000100
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: write: 121d000100
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 13
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: write: 042c002c00
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 05012c000229
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: write: 122c000100
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 13
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: write: 1219000103000101
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 13
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: write: 121900030101
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 1d2300010101
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 1b1c00b23eb8fd6f0429f403ff47ffe007fafffeff0300
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 1b1c00b23eb8fd6f0429f410ff3fffe407fbffffff0100
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 1b1c00b23eb8fd6f0429f403ff3dffe207fffffefffeff
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 1b1c00b23eb8fd6f0429f40aff3effe507fefffdffffff
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 1b1c00b23eb8fd6e0429f40bff4affeb070300f9ff0100
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 1b1c00b23eb9fd6e0429f413ff4affe107fffffdff0200
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 1b1c00b23eb8fd6e0429f410ff38ffd6070c00fafffcff
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 13
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: write: 1e

Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: write: 082a002c000229
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 1b1c00b23eb9fd6e0429f4fffe46ffef070c00f4fffcff
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 1b1c00b03ec0fd880425f418fe29ff89080f002700e2ff
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 1b1c00a43ebffd3a0531f42cf827fec905eaff0b001400
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 1b1c00a23ebcfd5b0533f417fa16fe4703f2fff9ff0100
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 1b1c00aa3ec5fdfb0431f48c02c300f208b5ff06000400
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 1b1c00a83ebdfd06052ef42dfe61fe480a1b000800f1ff
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 1b1c00a53ebafd330533f49ffd3aff3f03f9ffefff0d00
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 09042c000100
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: write: 122c000100
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 1d2300030000
Thu, 05 Mar 2015 11:20:43 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 1b1c00a73ebdfd160531f4b1022400910df8ff0b000500
Thu, 05 Mar 2015 11:20:44 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 1b1c009c3eabfd990535f443fac3fdcc021900f1fff9ff
Thu, 05 Mar 2015 11:20:44 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 1b1c00a83eb9fd130533f49903e900120bbdffffff0a00
Thu, 05 Mar 2015 11:20:44 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 1b1c00a13eadfd5d0534f4fefbcffd8f073200ffffffff
Thu, 05 Mar 2015 11:20:44 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 1b1c009a3ea8fdbd053af41efb76fe3d08daff26000200
Thu, 05 Mar 2015 11:20:44 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 1b1c009a3eb0fdc3053bf4f1fd4fff3f071a00b0fffcff
Thu, 05 Mar 2015 11:20:44 GMT l2cap-ble D3:89:EF:01:16:D8: line = data 13
Thu, 05 Mar 2015 11:20:44 GMT l2cap-ble D3:89:EF:01:16:D8: write: 1e
[... cut ...]
It looks like noble's previous "incorrect" stack queueing was affecting order of operations.

```discoverDescriptors``` is getting sent first after the change.

Could you try with the latest master version? This commit might help things: https://github.com/sandeepmistry/noble/commit/b8f7c5686c68ca434578dc05d57b056faaca4025

If it doesn't, for the next step I suggest you add some logs to each step of your code, to see how to order of operations differs for actions and callbacks/events before/after.

freehaha, 2015-03-06:
Unfortunately the commit doesn't solve the problem. I'll get some more detailed debug information back when I have time.

sandeepmistry, 2015-03-07:
@freehaha sounds good, the output of ```hcidump``` would also be useful, to compare with what noble thinks it's sending/receiving.

ManuelOverdijk, 2015-04-20:
@freehaha any update on this?

freehaha, 2015-05-16:
sorry for delaying this.. been busy lately.

Here are some more logs with the testing source code:
https://gist.github.com/freehaha/34f9fb2849f2bc9c0296

the flow is:
1. discover peripheral
2. connect
3. get a particular characteristic and `Char.notify true` (setNotiifcation)
4. without waiting for the above to complete, discover the descriptors in this Char. and write some value to it.

Basically, what seems to happen is in the current version after `Char.notify true`, subsequent commands weren't sent. Up to the `got characteristic` line, things seem to work properly, having only slightly different order due to that particular patch.

Again, i'm not sure if this is caused by the device itself not implemented properly or noble since I don't have other devices to test.

sandeepmistry, 2015-08-20:
@freehaha would you mind trying again with the latest version of noble? There have been some changes to the Linux bindings, that might have resolved the issue. Thanks.

sandeepmistry, 2015-09-28:
Closing due to lack of activity for now, please re-open once you've had a chance to try again based on my previous comment.

closed 2015-09-28 by sandeepmistry