noble/noble #137: pizza example not working properly

roshink123, 2015-02-26:
I am running both bleno and noble in different linux devices.While running these i am able to discover the services.But i am not getting any notification.Instead i am getting an error like this 

buffer.js:788
    throw RangeError('Trying to write outside buffer length');
          ^
RangeError: Trying to write outside buffer length
    at RangeError (<anonymous>)
    at checkInt (buffer.js:788:11)
    at Buffer.writeUInt8 (buffer.js:794:5)
    at bakePizza (/home/devel/node_modules/noble/examples/pizza/central.js:101:9)
    at /home/devel/node_modules/noble/examples/pizza/central.js:85:13
    at Service.<anonymous> (/home/devel/node_modules/noble/lib/service.js:53:7)
    at Service.g (events.js:180:16)
    at Service.EventEmitter.emit (events.js:95:17)
    at Noble.onCharacteristicsDiscover (/home/devel/node_modules/noble/lib/noble.js:249:13)
    at EventEmitter.emit (events.js:106:17)

what is wrong with this ? is it because i miss anything ? or becuase of issues related with bleno or noble ?

programmarchy, 2015-02-27:
Hi @roshink123 

This is an error in the example code. Sorry, I think I accidentally checked in some bugs on this part of the example.

I've made an attempt to fix them here:

https://github.com/programmarchy/noble/tree/pizza-example-fix

But I don't have a way to fully test it out because I don't have 2 BLE radios to run on at the moment (not until tomorrow.) Would you mind testing out this branch on your linux device and seeing if it resolves the issue for you?

sandeepmistry, 2015-03-01:
@programmarchy I've merged your fix into master.

Tested with bleno on Linux and noble on OS X.

closed 2015-03-01 by sandeepmistry

roshink123, 2015-03-02:
I am ruuning both noble and bleno in linux devices.But even after applying
the fix i am not getting the pizza example correctly.But the previous error
is not showing this time.

sandeepmistry, 2015-03-07:
@roshink123 I've pushed a fix for the example: 13c4c4d

programmarchy, 2015-03-09:
Thanks @sandeepmistry!

roshink123, 2015-03-10:
Thanks Sandeep.Its working properly.