noble/noble #138: iBeacon scanning frequently drops advertisements on Yosemite

snorp, 2015-02-27:
When I try to track an iBeacon via scanning with duplicates, I see periods where no advertisements are received. With the beacon advertising every 100ms, at best I get an advertisement every 2-300ms, but there are very frequent gaps of over 2s. I assume this is some kind of problem with the BLE daemon in OS X. I do not see the same behavior with an Android phone.

snorp, 2015-02-27:
If I change the beacon to broadcast every 900ms, I do get some advertisement at the 900ms interval, but frequently in the 3-4s range as well. It looks like maybe 30% of advertisements are just dropped on the floor for some reason.

sandeepmistry, 2015-02-27:
@snorp I'm also guessing it's a blued issue, if you provide the output in debug mode, that should give us a better idea:

```sh
DEBUG=yosemite-bindings node <file>.js
```

BTW, which beacons are you testing with?

snorp, 2015-02-28:
Here's some output[0] using DEBUG=yosemite-bindings. The beacon is set to advertise every 900ms, and you can see we pretty much always miss several advertisements.

I'm using a RedBear BLE Nano with more or less the example Nordic beacon app loaded. Using the iBeacon Scanner[0] app on my Android phone, I see updates every second as expected.

[0] https://gist.github.com/5fae505a30e9ddd99d06
[1] https://play.google.com/store/apps/details?id=kr.co.ixsoft.ibeacon.scanner

snorp, 2015-02-28:
Using PacketLogger from Hardware IO Tools was enlightening. It looks like it enters some sort of idle mode, and doesn't wake up unless there is a timeout.

https://gist.github.com/snorp/c73eaecea43c2225d491

snorp, 2015-02-28:
It looks like it *might* be possible to force our own power management policy by using IOKit. https://developer.apple.com/library/mac/documentation/IOKit/Reference/IOPMLib_header_reference/index.html#//apple_ref/c/func/IOAllowPowerChange

snorp, 2015-02-28:
I tried using IOCancelPowerChange() on the BroadcomBluetoothHostControllerUSBTransport service, but it does not succeed (returns kIOReturnError). Not sure if I'm doing something wrong or it just doesn't work.

sandeepmistry, 2015-03-01:
@snorp excellent research/investigating! Is there anything in particular you'd like me to try?

What's your goal/need for getting more precise discovery events?

snorp, 2015-03-01:
@sandeepmistry eventually I want to use it for presence detection in home automation. I wouldn't be using a Mac, this bug is just something irritating I found while playing around.

I am not sure there is anything you can do to work around it in noble. It seems like it's just a driver bug. I wouldn't think the hardware would be allowed to sleep while scanning is ongoing -- except maybe during the interval between scan windows.

sandeepmistry, 2015-03-02:
@snorp ok, just curious which Mac model were you testing on?

Can we close this issue?

snorp, 2015-03-02:
@sandeepmistry I tried on a late 2013 MBP with the built-in broadcom device, as well as an IOGEAR GBU521. We can close it if you want, since there doesn't appear to be anything actionable for noble.

snorp, 2015-03-02:
One last data point -- I used QuickBeacon on a Nexus 9 to advertise as an iBeacon, and I have the same behavior. Consistent sightings with a separate Android device, but the same dropping behavior on the Mac. I think this rules out the BLE Nano as being at fault.

sandeepmistry, 2015-03-07:
@snorp just tried with out with my 2013 MacBook Air and an iPhone using the [Location Beacon app](https://itunes.apple.com/ca/app/locate-beacon/id738709014?mt=8) to create an iBeacon.

I get multiple discover event per second.

Closing the issue, as you said there is nothing actionable for noble.

closed 2015-03-07 by sandeepmistry