noble/noble #140: noble on edison disconnects

venkvall7, 2015-03-02:
I am getting a continuous stream of data from a BLE peripheral using nrf51xxxx(Heart rate monitor program) and edison acts as central device.. it works well for sometime and gets disconnected after few mins..  After the initial disconnect it finds the service, connects to it and disconnects immediately..

is this a known issue?

sandeepmistry, 2015-03-03:
@venkvall7 I haven't used the BLE on the Intel Edison that much, so can't comment on it being a known issue.

Suggestions:
 * try using ```gatttool``` instead of noble to see if you still get disconnected.
 * provide ```hcidump``` capture, and I can take a look

venkvall7, 2015-03-05:
Thanks for the reply.. i see the similar issue with gatttool. But after rest of the peripheral, gatttool seems to work but not noble..

This is not consistent - may be edison stack issue?

Please see the attached hcidump when it disconnects..

device: hci0 snap_len: 1500 filter: 0xffffffff
< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7
    type 0x01 (active)
    interval 10.000ms window 10.000ms
    own address: 0x00 (Public) policy: All
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x01 (scanning enabled)
    filter duplicates 0x01 (enabled)
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
> HCI Event: LE Meta Event (0x3e) plen 37
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr C8:AA:87:A8:E2:07 (Random)
      Flags: 0x06
      Complete service classes: 0x180d 0x180a
      Unknown type 0x19 with 2 bytes data
      Complete local name: 'Aries HRM.'
      RSSI: -31
> HCI Event: LE Meta Event (0x3e) plen 12
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
      bdaddr C8:AA:87:A8:E2:07 (Random)
      RSSI: -30
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x00 (disabled)
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
< HCI Command: Inquiry (0x01|0x0001) plen 5
    lap 0x9e8b33 len 4 num 0
> HCI Event: Command Status (0x0f) plen 4
    Inquiry (0x01|0x0001) status 0x00 ncmd 1
< HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr C8:AA:87:A8:E2:07 type 1
    interval 96 window 48 initiator_filter 0
    own_bdaddr_type 0 min_interval 40 max_interval 56
    latency 0 supervision_to 42 min_ce 0 max_ce 0
> HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
> HCI Event: Inquiry Complete (0x01) plen 1
    status 0x00
> HCI Event: LE Meta Event (0x3e) plen 19
    LE Connection Complete
      status 0x00 handle 64, role master
      bdaddr C8:AA:87:A8:E2:07 (Random)
< ACL data: handle 64 flags 0x00 dlen 7
    ATT: MTU req (0x02)
      client rx mtu 256
> ACL data: handle 64 flags 0x02 dlen 7
    ATT: MTU resp (0x03)
      server rx mtu 23
< ACL data: handle 64 flags 0x00 dlen 11
    ATT: Read By Group req (0x10)
      start 0x0001, end 0xffff
      type-uuid 0x2800
> HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 2
> ACL data: handle 64 flags 0x02 dlen 24
    ATT: Read By Group resp (0x11)
      attr handle 0x0001, end group handle 0x0007
      value 0x00 0x18
      attr handle 0x0008, end group handle 0x000b
      value 0x01 0x18
      attr handle 0x000c, end group handle 0x0013
      value 0x0d 0x18
< ACL data: handle 64 flags 0x00 dlen 11
    ATT: Read By Group req (0x10)
      start 0x0014, end 0xffff
      type-uuid 0x2800
> HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 1
> HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x08
    Reason: Connection Timeout
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x00 (disabled)

sandeepmistry, 2015-03-06:
@venkvall7 good to know gatttool results in the same disconnect behaviour. I'm not sure why the connection is timing out:

```
HCI Event: Disconn Complete (0x05) plen 4
status 0x00 handle 64 reason 0x08
Reason: Connection Timeout
```

Do you also get Connection Timeout with noble again once the device is reset?

sandeepmistry, 2015-04-07:
Closing due to inactivity, please re-open with response if you are still interested.

closed 2015-04-07 by sandeepmistry