noble/noble #141: Not all services are saved as string

marcel12bell, 2015-03-03:
I have problems connecting to my characteristic... Is it related to the key in the noble object?
locking at ffe0 ->
services: 
   { '1': { uuid: '1800', startHandle: 1, endHandle: 11 },
     '12': { uuid: '1801', startHandle: 12, endHandle: 15 },
     '16': { uuid: 'ffe0', startHandle: 16, endHandle: 65535 },
     '1800': { uuid: '1800', startHandle: 1, endHandle: 11 },
     '1801': { uuid: '1801', startHandle: 12, endHandle: 15 },
     ffe0: { uuid: 'ffe0', startHandle: 16, endHandle: 65535 } } }

Error occured on /mavericks.js:299
-> serviceUuid = this._peripherals[deviceUuid].services[serviceStartHandle].uuid;
    TypeError: Cannot read property 'uuid' of undefined

sandeepmistry, 2015-03-05:
@marcel12bell could you please share the code you used to produce that output, along with debug output:

```sh
DEBUG=* node <file>.js
```

Which version of OS X, node.js and noble are you using?

marcel12bell, 2015-03-14:
@sandeepmistry  -> file and output: https://gist.github.com/marcel12bell/de5877482ccfe6859dbf 
OSX: 10.9.5, Node: v0.10.18, and current noble
Thanks for your tool btw...

sandeepmistry, 2015-03-14:
@marcel12bell why are you calling ```connect``` and ```discoverServices``` twice on the same peripheral? [1](https://gist.github.com/marcel12bell/de5877482ccfe6859dbf#file-gistfile1-txt-L16) and [2](https://gist.github.com/marcel12bell/de5877482ccfe6859dbf#file-gistfile1-txt-L35). This might be messing up noble's internal book keeping.

You should definetly remove this code:
```javascript
  peripheral.connect(function(error) {
    console.log('connected to peripheral: ' + peripheral.uuid);
    peripheral.discoverServices([peripheral.uuid], function(error, services) {
 
      console.log('discovered device information service');
      console.log('discovered service: ' + services);
 
    });
  });
```
Not sure why you are passing in the ```peripheral.uuid``` into ```discoverServices``` - that's just a bad idea! Also, I wouldn't recommend digging into noble's internals like your are doing on [this line](https://gist.github.com/marcel12bell/de5877482ccfe6859dbf#file-gistfile1-txt-L30).

What are you trying to do? Access all of the characteristics of the ```ffe0``` service?

marcel12bell, 2015-03-14:
The code was only to show how I got the Lines in my first post.
I want to send a string like I do with the LightBlue App. I had problems sending to the ffe0 service and digget into noble's internals to understand what happens behind...

marcel12bell, 2015-03-14:
The connection to my BLE seems a bit buggy. Thanks for your hints!

closed 2015-03-14 by marcel12bell

sandeepmistry, 2015-03-14:
@marcel12bell no problem, did you successfully get your app to write to characteristic FFE1?

marcel12bell, 2015-03-14:
@sandeepmistry yes, now it works.