noble/noble #147: characteristic.read() & characteristic.on('read',successhandler) troubles / non-GATT

trcharlie, 2015-03-10:
So I've been trying to communicate with a device with only one characteristic. To write a command, something like `set temp 55` is written to the characteristic. To read a value, some command like `read temp` is written to the same characteristic and the device itself writes the response on the same characteristic.

As it's not possible to change this behavior, I have to work around this issue, which i do at the moment by writing the command and adding an eventhandler to `characteristic.on('read',...)` which works fine for some time. As with every command sent to the device one eventhandler is added, also if it's the same command, I get an error like this 
```
(node) warning: possible EventEmitter memory leak detected. 11 read listeners added. Use emitter.setMaxListeners() to increase limit. 
```

I tried using `characteristic.read()`but it returns `Buffer <00>`. Maybe the device doesn't write the response quick enough.

Is there a way to remove the eventhandler after reading the value? Or any other possible solution for this behaviour?

regards,
charlie

sandeepmistry, 2015-03-10:
@trcharlie interesting device behaviour.

noble characteristic inherits from the built-in node.js [EventEmitter](https://nodejs.org/api/events.html#events_class_events_eventemitter).

I think you want to use ```.once``` instead of ```.on```, however read the EventEmitter docs linked above.

```javascript
characteristic.once('read', ...)
```

Which device are you using? Did you have to enable notifications to receive the read event?

trcharlie, 2015-03-11:
@sandeepmistry thanks for the quick response, cahnging to `characteristic.once('read', ...)` did the trick!
I'm using the Anova precision cooker, taking the events from here: https://github.com/neilpa/circulate/
Regarding notifications:
If I do nothing regarding notifications, I receive the response.
adding
```javascript
  anovaCharacteristic.notify(false, function(error,res) {
  });
```
prevents me from receiving the response.

sandeepmistry, 2015-03-12:
@trcharlie cool, no problem.

Devices aren't suppose to "auto-enable notify" on connect (only enable if it was enabled on the previous connection).

closed 2015-03-14 by sandeepmistry

dsully, 2016-12-05:
@trcharlie: I'm also trying to talk to an Anova - did you get any further? Have any code you can share?