noble/noble #148: Mac install error (Mavericks 10.9.5, Node 0.10.35)

vielmetti, 2015-03-11:
Here's a gist of my npm-debug.log:

https://gist.github.com/ecc4b51226bba6b8a847

I'm using the correct gcc:

```
Edwards-MacBook-Air:~ emv$ node -v
v0.10.35
Edwards-MacBook-Air:~ emv$ gcc -v
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 6.0 (clang-600.0.57) (based on LLVM 3.5svn)
Target: x86_64-apple-darwin13.4.0
Thread model: posix
```

and the error lines that get emitted are

```
148 error Darwin 13.4.0
149 error argv "node" "/usr/local/bin/npm" "install" "-g" "noble"
150 error node v0.10.35
151 error npm  v2.7.0
152 error code ELIFECYCLE
153 error noble@0.3.10 install: `node install.js`
```

Thanks for any help (trying to get the new TI sensortag running)

sandeepmistry, 2015-03-12:
@vielmetti I would expect some output from the "node install.js" command, strange that there is none.

Two things to try out:
1) Run: ```npm install xpc-connection``` (the OS X only dependency, ```node install.js``` installs)
2) Clone noble run "npm install" and maybe 1) is also needed in noble folder.

P.S.: thank you for the thorough bug report!

vielmetti, 2015-03-12:
npm install xpc-connection worked just fine.

I did the "git clone" and put the package in my "node-red" modules directory, which is the system I'm planning to use noble with. That worked, and I was able to connect to BLE devices and load the "node-red-node-sensortag" package that this depends on.

I'm puzzled by what happened, but it's working now, so I'll close this for now (and reopen if I run into any new problems).

closed 2015-03-12 by vielmetti

sandeepmistry, 2015-03-12:
@vielmetti strange, just to confirm, now a regular ```npm install noble``` works?

vielmetti, 2015-03-12:
No, npm install noble doesn't work, but I did a git clone of the codebase and things worked just fine.

```
Edwards-MacBook-Air:noble emv$ pwd
/usr/local/lib/node_modules/node-red/node_modules/noble
Edwards-MacBook-Air:noble emv$ git show
commit a61e0fd6830105cc1f23770e5f0d739003c18372
Author: Sandeep Mistry <sandeep.mistry@gmail.com>
Date:   Sat Mar 7 11:41:16 2015 -0500

    - uprev version

diff --git a/package.json b/package.json
index 716f755..e75538c 100644
--- a/package.json
+++ b/package.json
@@ -2,7 +2,7 @@
   "author": "Sandeep Mistry <sandeep.mistry@gmail.com>",
   "name": "noble",
   "description": "A node.js BLE (Bluetooth low energy) central library.",
-  "version": "0.3.9",
+  "version": "0.3.10",
   "repository": {
     "type": "git",
     "url": "https://github.com/sandeepmistry/noble.git"
```

sandeepmistry, 2015-03-13:
That's strange, how did you install node? Was there any output to stdout/stderr as to why "node install.js" returns 3?

https://gist.github.com/vielmetti/ecc4b51226bba6b8a847#file-npm-debug-log-L287

vielmetti, 2015-03-24:
Fixed by the fix in #149 (for 0.3.11).