noble/noble #149: noble install error - unlock target is not locked

finson, 2015-03-20:
OSX 10.10.2, node 0.10.25, npm 2.7.1, noble 0.3.10

When I try to install noble, I get an error related to unlocking another package that is not locked.  Sometimes it is debug, sometimes it is xpc-connection.  For example:

> npm ERR! Attempt to unlock /Users/Finson/node_modules/noble/node_modules/xpc-connection, which hasn't been locked

A screen cap and npm-debug.log are online: [https://gist.github.com/finson/8e22caf6dc8763ed5973](https://gist.github.com/finson/8e22caf6dc8763ed5973).

Any thoughts?  Thanks.

Doug Johnson

sandeepmistry, 2015-03-24:
@finson please try the latest (v0.3.11), I've removed the ```install.js``` script and made xpc-connection an optional dependency: 9650f722f5dca9ea49d6e95c21d863974e4c88a4

finson, 2015-03-24:
Success!  Thank you Sandeep. FYI, here's the console output.  Doug
```
holly:~/Documents/ollie$ sudo npm install noble
/
> xpc-connection@0.1.2 install /Users/Finson/Documents/ollie/node_modules/noble/node_modules/xpc-connection
> node-gyp rebuild

  CXX(target) Release/obj.target/binding/src/XpcConnection.o
  SOLINK_MODULE(target) Release/binding.node
  SOLINK_MODULE(target) Release/binding.node: Finished

> noble@0.3.11 install /Users/Finson/Documents/ollie/node_modules/noble
> node-gyp rebuild

  LINK(target) Release/hci-ble
  LINK(target) Release/hci-ble: Finished
  LINK(target) Release/l2cap-ble
  LINK(target) Release/l2cap-ble: Finished
noble@0.3.11 node_modules/noble
├── debug@0.7.4
└── xpc-connection@0.1.2 (nan@1.7.0)
holly:~/Documents/ollie$
```

closed 2015-03-24 by finson