noble/noble #150: `Uncaught TypeError: Cannot read property 'uuid' of undefined` error

taktran, 2015-03-20:
I randomly get this error on OSX 10.8.5, noble v0.3.10:

```
Uncaught TypeError: Cannot read property 'uuid' of undefined
    at eval (/Users/.../node_modules/noble/lib/mac/legacy.js:392:59)
    at EventEmitter.emit (events.js:95:17)
    at eval (/Users/.../node_modules/noble/lib/mac/legacy.js:44:8)
    at EventEmitter.emit (events.js:95:17)
    at eval (/Users/.../node_modules/noble/lib/mac/legacy.js:20:10)
    at XpcConnection.EventEmitter.emit (events.js:95:17)
```

Looking at the code where the error is generated:

```
nobleBindings.on(isLessThan10_8_5 ? 'kCBMsgId35' : 'kCBMsgId56', function(args) {
  var peripheralHandle = args.kCBMsgArgPeripheralHandle;
  var peripheralUuid = this._peripherals[peripheralHandle].uuid;
  var characteristicHandle = args.kCBMsgArgCharacteristicHandle;
  var isNotification = args.kCBMsgArgIsNotification ? true : false;
  var data = args.kCBMsgArgData;

  for(var i in this._peripherals[peripheralHandle].services) {
    if (this._peripherals[peripheralHandle].services[i].characteristics &&
        this._peripherals[peripheralHandle].services[i].characteristics[characteristicHandle]) {

      this.emit('read', peripheralUuid, this._peripherals[peripheralHandle].services[i].uuid,
        this._peripherals[peripheralHandle].services[i].characteristics[characteristicHandle].uuid, data, isNotification);
      break;
    }
  }
});
```

https://github.com/sandeepmistry/noble/blob/v0.3.10/lib%2Fmac%2Flegacy.js#L392

It fails on `this._peripherals[peripheralHandle]` being undefined.

Here are some values I got when I was console logging:

```
args = {"kCBMsgArgData":{"0":86,"length":1,"parent":{"0":86,"length":1},"offset":0},"kCBMsgArgIsNotification":1,"kCBMsgArgCharacteristicHandle":25,"kCBMsgArgPeripheralHandle":13}
peripheralHandle = 13
this._peripherals[peripheralHandle] = {}", 
```

```
args = {"kCBMsgArgData":{"0":86,"length":1,"parent":{"0":86,"length":1},"offset":0},"kCBMsgArgIsNotification":1,"kCBMsgArgCharacteristicHandle":25,"kCBMsgArgPeripheralHandle":13}
peripheralHandle = 13 
this._peripherals[peripheralHandle] = {"2":{"uuid":"7da6250984464963b413d3864fda8b6a","handle":2,"advertisement":{"localName":null,"txPowerLevel":null,"manufacturerData":null,"serviceData":[],"serviceUuids":[]},"rssi":-85},"3":{"uuid":"ff5964dec61145a8a65ab57dc005e234","handle":3,"advertisement":{"localName":null,"txPowerLevel":null,"manufacturerData":{"0":76,"1":0,"2":12,"3":14,"4":0,"5":144,"6":6,"7":142,"8":136,"9":58,"10":228,"11":26,"12":218,"13":188,"14":249,"15":28,"16":36,"17":234,"length":18,"parent":{"0":76,"1":0,"2":12,"3":14,"4":0,"5":144,"6":6,"7":142,"8":136,"9":58,"10":228,"11":26,"12":218,"13":188,"14":249,"15":28,"16":36,"17":234,"length":18},"offset":0},"serviceData":[],"serviceUuids":[]},"rssi":-63},"ff5964dec61145a8a65ab57dc005e234":{"uuid":"ff5964dec61145a8a65ab57dc005e234","handle":3,"advertisement":{"localName":null,"txPowerLevel":null,"manufacturerData":{"0":76,"1":0,"2":12,"3":14,"4":0,"5":144,"6":6,"7":142,"8":136,"9":58,"10":228,"11":26,"12":218,"13":188,"14":249,"15":28,"16":36,"17":234,"length":18,"parent":{"0":76,"1":0,"2":12,"3":14,"4":0,"5":144,"6":6,"7":142,"8":136,"9":58,"10":228,"11":26,"12":218,"13":188,"14":249,"15":28,"16":36,"17":234,"length":18},"offset":0},"serviceData":[],"serviceUuids":[]},"rssi":-63},"7da6250984464963b413d3864fda8b6a":{"uuid":"7da6250984464963b413d3864fda8b6a","handle":2,"advertisement":{"localName":null,"txPowerLevel":null,"manufacturerData":null,"serviceData":[],"serviceUuids":[]},"rssi":-85}}
```

Any ideas why this would happen? Is it as simple as adding a check to ignore the `kCBMsgId35`/`kCBMsgId56` event if `this._peripherals[peripheralHandle]` is undefined?

taktran, 2015-03-20:
I wonder if this is related: https://github.com/sandeepmistry/noble/issues/141 ?

sandeepmistry, 2015-03-24:
@taktran a simple check makes sense, with a warning as well.

I believe this occurs when another app is using BLE and a notification/indication is received, blued sends it out to all XPC clients.

kevster, 2015-03-30:
I'm getting a similar error (OSX 10.10.2, noble 0.3.11). It only occurs when I do the following:

1. Launch my application
2. Device is discovered and connected
3. Device is disconnected (I pull the power from the device)
4. Device is reconnected
5. Crash with the following error:

```
W20150330-10:18:18.728(2)? (STDERR) /Users/kevinandersen/Documents/Meteor/ArduinoToMeteor_V05/.meteor/local/isopacks/npm-container/npm/node_modules/noble/lib/mac/yosemite.js:528
W20150330-10:18:18.728(2)? (STDERR)  this._peripherals[deviceUuid].services[serviceUuid].characteristics[character
W20150330-10:18:18.729(2)? (STDERR)                                                                     ^
W20150330-10:18:18.733(2)? (STDERR) TypeError: Cannot read property '2222' of undefined
W20150330-10:18:18.733(2)? (STDERR)     at [object Object].nobleBindings.write (/Users/kevinandersen/Documents/Meteor/ArduinoToMeteor_V05/.meteor/local/isopacks/npm-container/npm/node_modules/noble/lib/mac/yosemite.js:528:103)
W20150330-10:18:18.733(2)? (STDERR)     at Noble.write (/Users/kevinandersen/Documents/Meteor/ArduinoToMeteor_V05/.meteor/local/isopacks/npm-container/npm/node_modules/noble/lib/noble.js:272:19)
W20150330-10:18:18.733(2)? (STDERR)     at Characteristic.write (/Users/kevinandersen/Documents/Meteor/ArduinoToMeteor_V05/.meteor/local/isopacks/npm-container/npm/node_modules/noble/lib/characteristic.js:64:15)
W20150330-10:18:18.733(2)? (STDERR)     at [object Object].<anonymous> (app/ArduinoToMeteor.js:275:38)
W20150330-10:18:18.733(2)? (STDERR)     at [object Object].wrapper [as _onTimeout] (timers.js:261:14)
W20150330-10:18:18.733(2)? (STDERR)     at Timer.listOnTimeout [as ontimeout] (timers.js:112:15)
```

sandeepmistry, 2015-04-05:
@taktran any updates? Did you try out the "simple check"?

sandeepmistry, 2015-04-05:
@kevster, please open a separate issue for the behaviour you are seeing, which includes a code sample to reproduce. It looks unrelated, because the error is being thrown from the ```write method```.

taktran, 2015-04-05:
It did fix the problem. I'll look into creating a PR.

closed 2015-04-06 by sandeepmistry

taktran, 2015-04-08:
Thanks! When do you think you'll publish a new version on npm?

sandeepmistry, 2015-04-09:
@taktran probably sometime this weekend ...

taktran, 2015-04-09:
Great thanks! Good luck =)

sandeepmistry, 2015-04-12:
@taktran v0.3.12 published to npm. cc/ @mhle

taktran, 2015-04-13:
:fireworks: thanks! =)