noble/noble #151: noble install on osx 10.10.2

mapdock, 2015-03-22:
I'm unable to install noble and see the following when I run 'npm install noble'

> noble@0.3.10 install /Users/paddock/.node-red/node_modules/noble
> node install.js
>
> noble install: platform is "darwin'
> noble install: installing xpc-connection ...
> npm ERR! Darwin 14.1.0
> npm ERR! argv "node" "/Users/paddock/.node/bin/npm" "install" "xpc-connection@~0.1.0"
> npm ERR! node v0.10.36
> npm ERR! npm  v2.7.3
> npm ERR! path /var/root/.npm/xpc-connection
> npm ERR! code EACCES
> npm ERR! errno 3
> 
> npm ERR! Error: EACCES, mkdir '/var/root/.npm/xpc-connection'
> npm ERR!  { [Error: EACCES, mkdir '/var/root/.npm/xpc-connection']
> npm ERR!   errno: 3,
> npm ERR!   code: 'EACCES',
> npm ERR!   path: '/var/root/.npm/xpc-connection',
> npm ERR!   parent: 'noble' }
> npm ERR! 
> npm ERR! Please try running this command again as root/Administrator.
> 
> npm ERR! Please include the following file with any support request:
> npm ERR!     /Users/paddock/.node-red/node_modules/noble/npm-debug.log
> noble install: done
> npm ERR! Darwin 14.1.0
> npm ERR! argv "node" "/Users/paddock/.node/bin/npm" "install" "noble"
> npm ERR! node v0.10.36
> npm ERR! npm  v2.7.3
> npm ERR! code ELIFECYCLE
> 
> npm ERR! noble@0.3.10 install: `node install.js`
> npm ERR! Exit status 3
> npm ERR! 
> npm ERR! Failed at the noble@0.3.10 install script 'node install.js'.
> npm ERR! This is most likely a problem with the noble package,
> npm ERR! not with npm itself.
> npm ERR! Tell the author that this fails on your system:
> npm ERR!     node install.js
> npm ERR! You can get their info via:
> npm ERR!     npm owner ls noble
> npm ERR! There is likely additional logging output above.
> 
> npm ERR! Please include the following file with any support request:
> npm ERR!     /Users/paddock/.node-red/npm-debug.log

finson, 2015-03-23:
This error appears to result from the same issue that is causing me problems (issue #149).

sandeepmistry, 2015-03-24:
@mapdock please try the latest (v0.3.11), I've removed the ```install.js``` script and made xpc-connection an optional dependency: 9650f722f5dca9ea49d6e95c21d863974e4c88a4

Were you running npm install with sudo?

closed 2015-04-07 by sandeepmistry