noble/noble #152: Not supported....

MeMyselfI, 2015-03-24:
Hi. 

Thank you very much for your software. I set up Noble on my Raspberry PI 2 with new Bluetooth dongles. From time to time I get a "Not supported" state although restarting (on the same device ID) the program results in a "poweredOn" state. I don't really why and where the problem is.

Besides, can I "reset" Noble and try again? If so, how?

sandeepmistry, 2015-03-28:
@MeMyselfI do you have a set of reproducible steps for this? I've seen this occasionally, but haven't tracked it down.

You can try a ```sudo hciconfig hci0 down && sudo hciconfig hci0 up``` to reset the adapter.

I think this is same behaviour as #129.

MeMyselfI, 2015-03-28:
Thanks for your answer. It helps when starting the program again so I guess I don't need to reset the adapter. The problem isn't related to scanning I start scanning only after a poweredOn which I don't get from time to time. 

And sorry, no, I cannot reproduce it - it happens from time to time after starting the program and it works the next time (without any reset or reboot). I will use a workaround with a cron job, though.

BTW, I used two dongles at the same time.

sandeepmistry, 2015-04-05:
@MeMyselfI how should we proceed with this issue? Without a set of steps to reproduce, it's hard to create a fix.

MeMyselfI, 2015-04-05:
No problem just close it. If I get more information I will open a new issue. Only thing that may help me is a method to restart noble, like noble.restart.

closed 2015-04-05 by MeMyselfI