noble/noble #155: Cannot read property 'stdin' of undefined

wezside, 2015-03-26:
I get this strange issue running Ubuntu 14.04. Noble scans and detects my Polar Cadence Sensor. I use `noble.startScanning(serviceUUIDs, false);` I can connect to the peripheral and read values. If I stop the server or the peripheral disconnects and I restart then I get this error. If I wait a few seconds and start the server again or run `sudo service bluetooth restart` then all is good. Any help will be much appreciated.

    TypeError: Cannot read property 'stdin' of undefined
        at L2capBle._queueCommand (nodejs/node_modules/noble/lib/linux/l2cap-ble.js:232:21)
        at L2capBle.exchangeMtu (/nodejs/node_modules/noble/lib/linux/l2cap-ble.js:327:8)
        at nobleBindings.onConnect (/nodejs/node_modules/noble/lib/linux/bindings.js:112:29)
        at emit (events.js:98:17)
        at L2capBle.onStdoutData (/nodejs/node_modules/noble/lib/linux/l2cap-ble.js:98:12)
        at Socket.emit (events.js:95:17)
        at Socket.<anonymous> (_stream_readable.js:765:14)
        at Socket.emit (events.js:92:17)
        at emitReadable_ (_stream_readable.js:427:10)
        at emitReadable (_stream_readable.js:423:5)

My disconnect code:

    process.on('SIGINT', function() 
    {
        peripheral.disconnect(function(err) {
            if (err) {console.log(err);} 
            console.log("Caught interrupt signal");
            process.exit();
        });     
    });

sandeepmistry, 2015-03-28:
@wezside are there any lingering ```l2cap-ble``` processes after you kill your app?

Could you also provide debug logs and a link to your code if available? ```DEBUG=* node <file>.js```

wezside, 2015-03-30:
Thanks for responding. I've spend more time looking at my code and I was indeed having two connections. So signalling the server for exit was causing one connection to stay open.

closed 2015-03-30 by wezside