noble/noble #156: noble version dependency on very old version of 'debug'

vielmetti, 2015-03-28:
`noble` 0.3.11 has a dependency on a very old version of `debug`, 0.7.2. This results in a conflict when I pull in a new version of `debug'.

```
edwards-air:~ emv$ sudo npm install -g debug
npm WARN unmet dependency /usr/local/lib/node_modules/node-red/node_modules/noble requires debug@'~0.7.2' but will load
npm WARN unmet dependency /usr/local/lib/node_modules/debug,
npm WARN unmet dependency which is version 2.1.3
debug@2.1.3 /usr/local/lib/node_modules/debug
└── ms@0.7.0
```

sandeepmistry, 2015-03-28:
@vielmetti I have no objections to upgrading the dependency.

I'm surprised it's complaining, I thought npm modules had independently installed dependencies ... see output of ```npm ls``` ...

sandeepmistry, 2015-05-16:
@vielmetti updated to 2.2.0 in https://github.com/sandeepmistry/noble/commit/a9065eabe364f5d8bc2de3128e7714580ba40444

closed 2015-05-16 by sandeepmistry