noble/noble #158: reading data from bluetooth blood pressure monitor U80IH

xmlking, 2015-04-05:
I got a bluetooth blood pressure monitor [http://www.ebay.com/itm/231430845385] and trying to understand how I can read data from this this device using `noble`. I am new to BT and noble, any help will be appreciated.  

I run scan and info commands using `cylon-ble` and got this results.... 
are they using GATT? if so, appreciate  if u can share example code to read data...

```bash
sudo node_modules/.bin/cylon-ble-info 8cde5263626e
```
```
  peripheral with UUID 8cde5263626e found
  Local Name        = Bluetooth BP
  Service Data      = 
  Service UUIDs     = 

services and characteristics:
1800 (Generic Access)
  2a00 (Device Name)
    properties  read
    value       426c7565746f6f7468204250 | 'Bluetooth BP'
  2a01 (Appearance)
    properties  read
    value       8000 | ''
  2a04 (Peripheral Preferred Connection Parameters)
    properties  read
    value       060080000000800c | '
                                    '
180a (Device Information)
  2a29 (Manufacturer Name String)
    properties  read
    value       49535343 | 'ISSC'
  2a24 (Model Number String)
    properties  read
    value       425435303530 | 'BT5050'
  2a25 (Serial Number String)
    properties  read
    value       384344453532363336323645 | '8CDE5263626E'
  2a27 (Hardware Revision String)
    properties  read
    value       353035305f5350500000000000 | '5050_SPP'
  2a26 (Firmware Revision String)
    properties  read
    value       32303235303231 | '2025021'
  2a28 (Software Revision String)
    properties  read
    value       30303030 | '0000'
  2a23 (System ID)
    properties  read
    value       0000000000000000 | ''
  2a2a (IEEE 11073-20601 Regulatory Certification Data List)
    properties  read
    value       0001000400000000 | ''
49535343fe7d4ae58fa99fafd205e455
  495353436daa4d02abf619569aca69fe
    properties  read, write
    value       003600360000002a00 | '66*'
  49535343aca3481c91ecd85e28a60318
    properties  write, notify
fff0
  fff1
    properties  notify
  fff2
    properties  write
```

xmlking, 2015-04-05:
Tried to read data with [LightBlue](https://itunes.apple.com/us/app/lightblue/id639944780) on Mac and noticed non-decimal  value changing on `fff1` while it is measuring the  blood pressure. 
Their  Android and iOS apps were able to initiate/start the  measuring process. so I thing there will be a way to activate measuring process by writing hex value into one of those  characteristics!

sandeepmistry, 2015-04-05:
@xmlking it's probably best to contact the seller or manufacturer for the GATT specification of the device. You could also ask them if they have an iOS or Android SDK. Another alternative is to decompile the Android APK.

Closing the issue, as it is unfortunately outside the scope of noble.

closed 2015-04-05 by sandeepmistry