noble/noble #160: GATT Peripheral and Observer (central) on same machine

AlexanderLea, 2015-04-07:
Hi,

I'm trying to use your library to create some software that would run on a server, and listen to incoming information from a number of peripherals, do something with this information and then transmit it (as a peripheral) to a number of observing devices.

I have both the central and peripheral code (peripheral using bleno) running separately, but when I try and combine them, it doesn't work properly: if I set up as the GATT central first, I cannot set up bleno to advertise; and if bleno is set to advertise first, I cannot connect to the peripheral as a GATT central.

I'm pretty sure that it's possible to do both of these things simultaneously using Bluez on the command line, but is is possible using bleno and noble?

Any pointers/opinions would be greatly appreciated!

Thanks, Alex

sandeepmistry, 2015-04-07:
> I'm pretty sure that it's possible to do both of these things simultaneously using Bluez on the command line, but is is possible using bleno and noble?

@AlexanderLea can you confirm it's possible with BlueZ and the command line? Which kernel are you using? What command sequence?

If you get that successfully working, bleno and noble can be adapted accordingly - but I haven't had much success with it in the past ...

AlexanderLea, 2015-04-08:
@sandeepmistry, it appears as though you're correct. I'm certain I've simultaneously acted as a peripheral and central before, but I am unable to get it working today.

Have ordered another Bluetooth Adapter, which will hopefully allow me to do what I want to.

Many thanks!

sandeepmistry, 2015-04-09:
@AlexanderLea no problem, you should be good to go with two adapters, you can use env. var's to specify which adapter noble/bleno should use:

https://github.com/sandeepmistry/noble#multiple-adapters

https://github.com/sandeepmistry/bleno#multiple-adapters

closed 2015-04-09 by sandeepmistry

AlexanderLea, 2015-04-09:
@sandeepmistry, sorry to re-open, but I have another question: 

In the code, is it possible to set which adapter bleno/noble should use? - that way I could just run with a single command "node ble_central.js" for example, and then be able to deal with both the observer and peripheral as a single program?

sandeepmistry, 2015-04-12:
@AlexanderLea you should be able to set them using something like this:

```javascript
process.env. NOBLE_HCI_DEVICE_ID="0"
process.env. BLENO_HCI_DEVICE_ID="1"

var noble = require('noble');
var bleno = require('bleno');

// ...
```

AlexanderLea, 2015-04-13:
You're correct. Thank you!