noble/noble #161: Error: dlopen( ***/noble/node_modules/xpc-connection/build/Release/binding.node, 1): Symbol not found: ___CFConstantStringClassReference

scaret, 2015-04-07:
Hi,

When I clone the repo and run `npm install`, it looks fine.
When I try to run any example including one in readme, it will always trigger this error:

```
module.js:349
  Module._extensions[extension](this, filename);
                               ^
Error: dlopen(/Users/scaret/temp/noble/node_modules/xpc-connection/build/Release/binding.node, 1): Symbol not found: ___CFConstantStringClassReference
  Referenced from: /Users/scaret/temp/noble/node_modules/xpc-connection/build/Release/binding.node
  Expected in: dynamic lookup

    at Error (native)
    at Module.load (module.js:349:32)
    at Function.Module._load (module.js:305:12)
    at Module.require (module.js:357:17)
    at require (module.js:373:17)
    at Object.<anonymous> (/Users/scaret/temp/noble/node_modules/xpc-connection/index.js:3:15)
    at Module._compile (module.js:449:26)
    at Object.Module._extensions..js (module.js:467:10)
    at Module.load (module.js:349:32)
    at Function.Module._load (module.js:305:12)
```

OS: OSX 10.10.2 (14C109)
node: v0.11.13

scaret, 2015-04-07:
When I checked nodejs version I tried to update node version to v0.12, and it works fine for me!

sandeepmistry, 2015-04-07:
@scaret do you still care about this?

Did you have any weird compilers installed, what's the output of ``` gcc -v```?

scaret, 2015-04-08:
@sandeepmistry 

I've never changed compiler manually on this Macbook, and most npm modules compiles & works fine. What I did to make noble work is just switching to node v0.12.2

Anyway I don't care about this any longer, so you can close this issue.

FYI my gcc version information

```
denny:~ scaret$ gcc -v
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 6.0 (clang-600.0.57) (based on LLVM 3.5svn)
Target: x86_64-apple-darwin14.1.0
Thread model: posix
```

sandeepmistry, 2015-04-09:
Closing for now.

closed 2015-04-09 by sandeepmistry