noble/noble #162: Connect to devices based on a white list of MAC addresses

AlexanderLea, 2015-04-09:
Hi,

I am looking to implement a program to act as a BLE central, and as such initiate connections to a number of BLE peripherals.

Currently I'm connecting to all peripherals that expose my GATT service and characteristic, which works, but I would be much happier if I could implement a whitelist of MAC addresses of my peripherals and only connect to those.

I can't find anywhere where the MAC address is accessible within noble - am I correct?

I'm very new to node.js, and am lacking the confidence/ability to fork this repo and attempt to make these changes. Please would you be able to give me any advice on how to go about doing this?

Many thanks
Alex

sandeepmistry, 2015-04-12:
@AlexanderLea would you be able to share your code? It sounds like you are connecting to every peripheral discovered.

You should inspect the peripheral the discover event: https://github.com/sandeepmistry/noble#peripheral-discovered

On Linux, the uuid is the BT address, and the address will be accessible shortly (the change is already on master).

AlexanderLea, 2015-04-14:
@sandeepmistry thank you. I've got it - I'm now using my white list to only connect to devices that I want to.

Thanks for your help!

closed 2015-04-14 by AlexanderLea