noble/noble #163: `ReferenceError: deviceUuid is not defined` on 0.3.12

taktran, 2015-04-13:
On the new version (0.3.12), it breaks on legacy macs. Noble doesn't start. I get this error:

```
/Users/.../node_modules/noble/lib/mac/legacy.js:177
  }.bind(this))(deviceUuid, peripheralHandle, advertisement, rssi);
                ^
ReferenceError: deviceUuid is not defined
    at nobleBindings.connect.sendCBMsg.kCBMsgArgOptions.kCBConnectOptionNotifyOnDisconnection (/Users/.../node_modules/noble/lib/mac/legacy.js:177:17)
    at emit (events.js:95:17)
    at nobleBindings.sendCBMsg.sendXpcMessage.kCBMsgId (/Users/.../node_modules/noble/lib/mac/legacy.js:46:8)
    at emit (events.js:95:17)
    at null.<anonymous> (/Users/.../node_modules/noble/lib/mac/legacy.js:22:10)
    at XpcConnection.emit (events.js:95:17)
```

https://github.com/sandeepmistry/noble/blob/master/lib/mac/legacy.js#L177

It seems like `deviceUuid` isn't used anywhere. It seems like supposed to be `peripheralUuid`.

Will dig in further, and submit a PR if I fix it.

taktran, 2015-04-13:
Oh, it seems to be another change: https://github.com/sandeepmistry/noble/commit/a2511407dbd460885b1511ca8ef9de5873430a26#diff-5c90076a70226ec417f622c0a8a32d8fR169

taktran, 2015-04-13:
I renamed the variable here: https://github.com/taktran/noble/commit/dd4b2d4b23a0bc4b85706778d7f4483e746fe342, and it seemed to run ok, but it was blocking my [nodewebkit](https://github.com/nwjs/nw.js/) from displaying anything.

It displays for v0.3.11 though. Strange.....

sandeepmistry, 2015-04-13:
@taktran I've merged in https://github.com/taktran/noble/commit/dd4b2d4b23a0bc4b85706778d7f4483e746fe342

Strange that it still says v0.3.11 ... do you think it's stuck trying to parse: /Library/Preferences/com.apple.Bluetooth.plist ? It does require a new dependency: ``"bplist-parser": "0.0.6"``

sandeepmistry, 2015-04-13:
@taktran I'm closing this, as your commit fixes it, v0.3.13 published with your fic.

closed 2015-04-13 by sandeepmistry

taktran, 2015-04-14:
Thanks for the update @sandeepmistry - unfortunately, node webkit not displaying doesn't seem to be a freak occurrence.

It's because of the `uuidToAddress` method from this commit: https://github.com/sandeepmistry/noble/commit/a2511407dbd460885b1511ca8ef9de5873430a26#diff-5c90076a70226ec417f622c0a8a32d8fR169

It completely spams my app, so that it can't start. Reverting to just