noble/noble #165: Connection Rejected due to Limited Resources (central.js)

kclim2, 2015-04-13:
I'm testing Noble' pizza central (Intel Galileo (Gen2) with Bleno' pizza peripheral. The following is the hcidump (Noble' pizza central)

root@galileo:~# hcidump 
HCI sniffer - Bluetooth packet analyzer ver 5.18
device: hci0 snap_len: 1500 filter: 0xffffffff
< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7
    type 0x01 (active)
    interval 10.000ms window 10.000ms
    own address: 0x00 (Public) policy: All
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x01 (scanning enabled)
    filter duplicates 0x01 (enabled)
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 2
    status 0x00
> HCI Event: LE Meta Event (0x3e) plen 33
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr F8:16:54:CD:28:B3 (Public)
      Flags: 0x05
      Unknown type 0x06 with 16 bytes data
      RSSI: -65
> HCI Event: LE Meta Event (0x3e) plen 24
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
      bdaddr F8:16:54:CD:28:B3 (Public)
      Shortened local name: 'PizzaSquat'
      RSSI: -65
< HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr F8:16:54:CD:28:B3 type 0
    interval 96 window 48 initiator_filter 0
    own_bdaddr_type 0 min_interval 40 max_interval 56
    latency 0 supervision_to 42 min_ce 0 max_ce 0
> HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x0d ncmd 1
    Error: Connection Rejected due to Limited Resources

Using the same Intel Galileo I have no problem with peripheral-explorer.js:

root@galileo:~# hcidump 
HCI sniffer - Bluetooth packet analyzer ver 5.18
device: hci0 snap_len: 1500 filter: 0xffffffff
< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7
    type 0x01 (active)
    interval 10.000ms window 10.000ms
    own address: 0x00 (Public) policy: All
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x01 (scanning enabled)
    filter duplicates 0x01 (enabled)
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 2
    status 0x00
> HCI Event: LE Meta Event (0x3e) plen 33
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr F8:16:54:CD:28:B3 (Public)
      Flags: 0x05
      Unknown type 0x06 with 16 bytes data
      RSSI: -65
> HCI Event: LE Meta Event (0x3e) plen 24
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
      bdaddr F8:16:54:CD:28:B3 (Public)
      Shortened local name: 'PizzaSquat'
      RSSI: -65
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x00 (disabled)
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 2
    status 0x00
< HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr F8:16:54:CD:28:B3 type 0
    interval 96 window 48 initiator_filter 0
    own_bdaddr_type 0 min_interval 40 max_interval 56
    latency 0 supervision_to 42 min_ce 0 max_ce 0
> HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 2
> HCI Event: LE Meta Event (0x3e) plen 19
    LE Connection Complete
      status 0x00 handle 3585, role master
      bdaddr F8:16:54:CD:28:B3 (Public)
< ACL data: handle 3585 flags 0x00 dlen 7
    ATT: MTU req (0x02)
      client rx mtu 256
> HCI Event: Number of Completed Packets (0x13) plen 5
    handle 3585 packets 1
> ACL data: handle 3585 flags 0x02 dlen 7
    ATT: MTU resp (0x03)
      server rx mtu 256
< ACL data: handle 3585 flags 0x00 dlen 11
    ATT: Read By Group req (0x10)
      start 0x0001, end 0xffff
      type-uuid 0x2800
> HCI Event: Number of Completed Packets (0x13) plen 5
    handle 3585 packets 1
....

The gatttool works well with the same Intel Galileo too:
root@galileo:~/node_modules/noble/examples# gatttool -b F8:16:54:CD:28:B3 -I
[F8:16:54:CD:28:B3][LE]> connect
Attempting to connect to F8:16:54:CD:28:B3
Connection successful
[F8:16:54:CD:28:B3][LE]> characteristics 
handle: 0x0002, char properties: 0x02, char value handle: 0x0003, uuid: 00002a00-0000-1000-8000-00805f9b34fb
handle: 0x0004, char properties: 0x02, char value handle: 0x0005, uuid: 00002a01-0000-1000-8000-00805f9b34fb
handle: 0x0007, char properties: 0x0a, char value handle: 0x0008, uuid: 13333333-3333-3333-3333-333333330001
handle: 0x000a, char properties: 0x0a, char value handle: 0x000b, uuid: 13333333-3333-3333-3333-333333330002
handle: 0x000d, char properties: 0x18, char value handle: 0x000e, uuid: 13333333-3333-3333-3333-333333330003

May I know what can we do to fix the problem (Connection Rejected due to Limited Resources)?

//root@galileo:~/node_modules/noble/examples# hciconfig -a
hci0:	Type: BR/EDR  Bus: USB
	BD Address: F8:16:54:CD:5B:BC  ACL MTU: 1021:5  SCO MTU: 96:6
	UP RUNNING PSCAN 
	RX bytes:3469 acl:69 sco:0 events:180 errors:0
	TX bytes:3088 acl:60 sco:0 commands:89 errors:0
	Features: 0xff 0xfe 0x0f 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH HOLD SNIFF 
	Link mode: SLAVE ACCEPT 
	Name: 'galileo'
	Class: 0x200104
	Service Classes: Audio
	Device Class: Computer, Desktop workstation
	HCI Version: 4.0 (0x6)  Revision: 0xe00
	LMP Version: 4.0 (0x6)  Subversion: 0xe00
	Manufacturer: Intel Corp. (2)

//uname -a
Linux galileo 3.8.7-yocto-standard #1 Fri Feb 6 08:29:08 GMT 2015 i586 GNU/Linux

//node --version
v0.10.35

//root@galileo:~/node_modules/noble/examples# cat ../package.json 
{
  "author": {
    "name": "Sandeep Mistry",
    "email": "sandeep.mistry@gmail.com"
  },
  "name": "noble",
  "description": "A node.js BLE (Bluetooth low energy) central library.",
  "version": "0.3.11",
  "repository": {
    "type": "git",
    "url": "https://github.com/sandeepmistry/noble.git"
  },
  "bugs": {
    "url": "https://github.com/sandeepmistry/noble/issues"
  },
  "keywords": [
    "bluetooth",
    "BLE",
    "bluetooth low energy",
    "bluetooth smart",
    "central"
  ],
  "main": "./index.js",
  "engines": {
    "node": ">=0.8"
  },
  "os": [
    "darwin",
    "linux"
  ],
  "dependencies": {
    "debug": "~0.7.2",
    "xpc-connection": "~0.1.0"
  },
  "optionalDependencies": {
    "xpc-connection": "~0.1.0"
  },
  "devDependencies": {
    "jshint": "latest",
    "mocha": "~1.8.2",
    "should": "~1.2.2",
    "sinon": "~1.6.0",
    "async": "~0.2.9",
    "ws": "~0.4.31"
  },
  "scripts": {
    "pretest": "jshint *.js lib/. test/.",
    "test": "mocha -R spec test/*.js",
    "install": "node-gyp rebuild"
  },
  "browser": {
    "./distributed/bindings": false,
    "./mac/bindings": false,
    "./linux/bindings": false
  },
  "gypfile": true,
  "gitHead": "52eb60bbad39e63b1d9c218f109e21d6f4df7538",
  "homepage": "https://github.com/sandeepmistry/noble",
  "_id": "noble@0.3.11",
  "_shasum": "deb7401ecf9261979a47b9b443f43faed6f1178a",
  "_from": "noble@",
  "_npmVersion": "1.4.21",
  "_npmUser": {
    "name": "sandeepmistry",
    "email": "sandeep.mistry@gmail.com"
  },
  "maintainers": [
    {
      "name": "sandeepmistry",
      "email": "sandeep.mistry@gmail.com"
    }
  ],
  "dist": {
    "shasum": "deb7401ecf9261979a47b9b443f43faed6f1178a",
    "tarball": "http://registry.npmjs.org/noble/-/noble-0.3.11.tgz"
  },
  "directories": {},
  "_resolved": "https://registry.npmjs.org/noble/-/noble-0.3.11.tgz",
  "readme": "ERROR: No README data found!"
}

sandeepmistry, 2015-04-13:
@kclim2 are you running bleno and noble on the same Intel Galileo (Gen2)?

kclim2, 2015-04-14:
different node. 

Intel Galileo has no problem to run as the peripheral node (bleno) but not able to run as the central (noble).

sandeepmistry, 2015-04-14:
@kclim2 can you please try scanning and connecting with gatttool at the same time (two different terminal windows).

1) ```sudo hcitool lescan```
2) ```gatttool -b F8:16:54:CD:28:B3 -I```

kclim2, 2015-04-15:
using two different terminal windows:
terminal windows #1: hcitool lescan

terminal windows #2: 
root@galileo:~/node_modules/noble/examples# gatttool -b F8:16:54:CD:28:B3 -I
[F8:16:54:CD:28:B3][LE]> connect
Attempting to connect to F8:16:54:CD:28:B3
Error: connect error: Connection refused (111)

gattool fail to connect!!

I then kill the hcitool lescan at terminal windows #1 (ctrl+c to end the hcitool lescan)

trying to connect with gattool again..

terminal windows #2:
[F8:16:54:CD:28:B3][LE]> connect
Attempting to connect to F8:16:54:CD:28:B3
Connection successful
[F8:16:54:CD:28:B3][LE]> characteristics 
handle: 0x0002, char properties: 0x02, char value handle: 0x0003, uuid: 00002a00-0000-1000-8000-00805f9b34fb
handle: 0x0004, char properties: 0x02, char value handle: 0x0005, uuid: 00002a01-0000-1000-8000-00805f9b34fb
handle: 0x0007, char properties: 0x0a, char value handle: 0x0008, uuid: 13333333-3333-3333-3333-333333330001
handle: 0x000a, char properties: 0x0a, char value handle: 0x000b, uuid: 13333333-3333-3333-3333-333333330002
handle: 0x000d, char properties: 0x18, char value handle: 0x000e, uuid: 13333333-3333-3333-3333-333333330003

it doesn't give the connection reject error anymore! :)

in the central.js file i manually inserted noble.stopScanning() into the noble.on('discover', function(peripheral) {

noble.on('discover', function(peripheral) {                                                                                    
  console.log('found peripheral:', peripheral.advertisement);                                                                  
  //manually inserted noble.stopScanning() here before peripheral.connect()...
  noble.stopScanning()                                                                                                        
  peripheral.connect(function(err) {

the pizza example works perfectly! Thanks. :)

kclim2, 2015-04-15:
root@galileo:~/node_modules/noble/examples/pizza# node central.js 
scanning...
found peripheral: { localName: 'PizzaSquat',
  txPowerLevel: undefined,
  manufacturerData: undefined,
  serviceData: [],
  serviceUuids: [ '13333333333333333333333333333337' ] }
found service: 13333333333333333333333333333337
found characteristic: 13333333333333333333333333330001
found characteristic: 13333333333333333333333333330002
found characteristic: 13333333333333333333333333330003
Our pizza is ready!
The result is crispy.

closed 2015-04-15 by kclim2

sandeepmistry, 2015-04-15:
@kclim2 good to know! I've updated the example to stop scanning after discovering a peripheral (826b0f9).