noble/noble #166: TypeError: Cannot read property 'uuid' of undefined

ManuelOverdijk, 2015-04-13:
Hey guys,

I get the following error: 
```javascript
/node_modules/noble/lib/mac/yosemite.js:411
serviceUuid = this._peripherals[deviceUuid].services[serviceStartHandle].uuid;
                                                                      ^
TypeError: Cannot read property 'uuid' of undefined

```

sandeepmistry, 2015-04-13:
@ManuelOverdijk could you please provide debug output:

```sh
DEBUG=* node <file>.js
```

Also, would you be able to share more of your code?

matthewnourse, 2015-04-17:
@sandeepmistry I'm getting a similar error but in a uuid-to-address.js.  I've added this line to uuid-to-address.js in my local copy to help with debugging:

```
console.log("obj=", require('util').inspect(obj, {showHidden: true, depth: null}));
```

My code:

```
var noble = require('noble');

noble.startScanning();

noble.on('discover', function(peripheral) {
  console.log("peripheral=", peripheral);
});
```

I ran it with this command line on Mac OS (Yosemite):
```
env NODE_PATH=.:/usr/local/lib/node_modules DEBUG=* node cadenceAlarm.js > debug.out 2>&1
```

The contents of debug.out:

```
Fri, 17 Apr 2015 11:09:20 GMT yosemite-bindings sendCBMsg: 1, {
  "kCBMsgArgName": "node-1429268960474",
  "kCBMsgArgOptions": {
    "kCBInitOptionShowPowerAlert": 0
  },
  "kCBMsgArgType": 0
}
Fri, 17 Apr 2015 11:09:20 GMT yosemite-bindings sendCBMsg: 29, {
  "kCBMsgArgOptions": {},
  "kCBMsgArgUUIDs": []
}
Fri, 17 Apr 2015 11:09:20 GMT noble scanStart
Fri, 17 Apr 2015 11:09:20 GMT yosemite-bindings xpcEvent: {
  "kCBMsgId": 6,
  "kCBMsgArgs": {
    "kCBMsgArgState": 5
  }
}
Fri, 17 Apr 2015 11:09:20 GMT yosemite-bindings state change poweredOn
Fri, 17 Apr 2015 11:09:20 GMT noble stateChange poweredOn
Fri, 17 Apr 2015 11:09:29 GMT yosemite-bindings xpcEvent: {
  "kCBMsgId": 37,
  "kCBMsgArgs": {
    "kCBMsgArgAdvertisementData": {
      "kCBAdvDataIsConnectable": 1,
      "kCBAdvDataServiceUUIDs": [
        {
          "type": "Buffer",
          "data": [
            24,
            22
          ]
        }
      ],
      "kCBAdvDataLocalName": "RPM"
    },
    "kCBMsgArgRssi": -59,
    "kCBMsgArgDeviceUUID": {
      "type": "Buffer",
      "data": [
        0,
        230,
        10,
        139,
[... 171 lines cut ...]
               DataElementSize: 2 },
             { DataElementType: 1, DataElementValue: 106, DataElementSize: 2 },
             { DataElementType: 1, DataElementValue: 256, DataElementSize: 2 },
             [length]: 3 ],
          '0000 - ServiceRecordHandle*': 65540,
          '0001 - ServiceClassIDList': [ <Buffer 11 01>, [length]: 1 ],
          '0009 - BluetoothProfileDescriptorList': 
           [ [ <Buffer 11 01>,
               { DataElementType: 1, DataElementValue: 256, DataElementSize: 2 },
               [length]: 2 ],
             [length]: 1 ] } },
    BluetoothVersionNumber: 3 },
  [length]: 1 ]
/usr/local/lib/node_modules/noble/lib/mac/uuid-to-address.js:19
    var coreBluetoothCacheEntry = obj[0].CoreBluetoothCache[formattedUuid];
                                                           ^
TypeError: Cannot read property '00E60A8B-6951-42A4-B199-B2F8E4818BFF' of undefined
    at /usr/local/lib/node_modules/noble/lib/mac/uuid-to-address.js:19:60
    at tryParseBuffer (/usr/local/lib/node_modules/noble/node_modules/bplist-parser/bplistParser.js:25:5)
    at /usr/local/lib/node_modules/noble/node_modules/bplist-parser/bplistParser.js:33:7
    at fs.js:334:14
    at FSReqWrap.oncomplete (fs.js:95:15)
```

Thanks for noble, it looks very cool.

sandeepmistry, 2015-04-17:
@matthewnourse it's a separate issue: https://github.com/sandeepmistry/noble/issues/168 - should be addressed in the v0.3.14 release ...

sandeepmistry, 2015-04-18:
@ManuelOverdijk any updates (debug logs, code) on this?

matthewnourse, 2015-04-18:
@sandeepmistry perfect, I upgraded to v0.3.14 and it works great now, thanks, and sorry that I missed #168.

sandeepmistry, 2015-04-26:
Closing due to lack of activity ...

closed 2015-04-26 by sandeepmistry