noble/noble #167: Cannot npm install

pequnio3, 2015-04-15:
Hi,

I'm trying to install noble but am running into an error when I do npm install:

I pasted below the error I'm seeing.  What am I doing wrong?

home-automation pequnio3$ sudo npm install noble
npm WARN package.json node_sandbox@1.0.0 No description
npm WARN package.json node_sandbox@1.0.0 No README data
\
> xpc-connection@0.1.3 install /Users/pequnio3/Documents/home-automation/node_modules/noble/node_modules/xpc-connection
> node-gyp rebuild

gyp WARN install got an error, rolling back install
gyp ERR! configure error 
gyp ERR! stack Error: EPERM, utime '/Users/pequnio3/.node-gyp/0.12.2'
gyp ERR! stack     at Error (native)
gyp ERR! System Darwin 13.3.0
gyp ERR! command "node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/pequnio3/Documents/home-automation/node_modules/noble/node_modules/xpc-connection
gyp ERR! node -v v0.12.2
gyp ERR! node-gyp -v v1.0.3
gyp ERR! not ok 
npm WARN optional dep failed, continuing xpc-connection@0.1.3

> noble@0.3.13 install /Users/pequnio3/Documents/home-automation/node_modules/noble
> node-gyp rebuild

gyp WARN install got an error, rolling back install
gyp ERR! configure error 
gyp ERR! stack Error: EPERM, utime '/Users/pequnio3/.node-gyp/0.12.2'
gyp ERR! stack     at Error (native)
gyp ERR! System Darwin 13.3.0
gyp ERR! command "node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/pequnio3/Documents/home-automation/node_modules/noble
gyp ERR! node -v v0.12.2
gyp ERR! node-gyp -v v1.0.3
gyp ERR! not ok 
npm ERR! Darwin 13.3.0
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "noble"
npm ERR! node v0.12.2
npm ERR! npm  v2.7.4
npm ERR! code ELIFECYCLE

npm ERR! noble@0.3.13 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the noble@0.3.13 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the noble package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls noble
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/pequnio3/Documents/home-automation/npm-debug.log

sandeepmistry, 2015-04-15:
@pequnio3 looks like permissions for your ``~/.node-gyp`` folder are incorrect. Maybe you npm installed something using sudo before ...

Try running ```sudo chown pequnio3:staff ~/.node-gyp```, then try the npm install again ...

sandeepmistry, 2015-04-18:
Closing for now, please re-open if install doesn't work after permissions have been setup correctly.

closed 2015-04-18 by sandeepmistry