noble/noble #168: TypeError: Cannot read property 'BD8B4C1B-4F8D-4430-A309-AD981645677D' of undefined

aensidhe, 2015-04-16:
I tried to play with my bracelet, but noble crashed. Debug output below.
Source is [discovery example](https://github.com/sandeepmistry/noble/blob/master/examples/advertisement-discovery.js)

````
Forest:dami aensidhe$ DEBUG=* node discovery.js
  yosemite-bindings sendCBMsg: 1, {
  "kCBMsgArgName": "node-1429211348937",
  "kCBMsgArgOptions": {
    "kCBInitOptionShowPowerAlert": 0
  },
  "kCBMsgArgType": 0
} +0ms
  yosemite-bindings xpcEvent: {
  "kCBMsgId": 6,
  "kCBMsgArgs": {
    "kCBMsgArgState": 5
  }
} +8ms
  yosemite-bindings state change poweredOn +1ms
  noble stateChange poweredOn +0ms
  yosemite-bindings sendCBMsg: 29, {
  "kCBMsgArgOptions": {},
  "kCBMsgArgUUIDs": []
} +1ms
  noble scanStart +1ms
  yosemite-bindings xpcEvent: {
  "kCBMsgId": 37,
  "kCBMsgArgs": {
    "kCBMsgArgAdvertisementData": {
      "kCBAdvDataIsConnectable": 1,
      "kCBAdvDataServiceUUIDs": [
        {
          "type": "Buffer",
          "data": [
            255,
            240
          ]
        }
      ]
    },
    "kCBMsgArgRssi": -85,
    "kCBMsgArgDeviceUUID": {
      "type": "Buffer",
      "data": [
        189,
        139,
        76,
        27,
        79,
        141,
        68,
        48,
        163,
        9,
        173,
        152,
        22,
        69,
        103,
        125
      ]
    },
    "kCBMsgArgName": "BTL01001"
  }
} +32s
  yosemite-bindings peripheral bd8b4c1b4f8d4430a309ad981645677d discovered +0ms
/Users/aensidhe/Sources/2015/dami/node_modules/noble/lib/mac/uuid-to-address.js:17
    var coreBluetoothCacheEntry = obj[0].CoreBluetoothCache[formattedUuid];
                                                           ^
TypeError: Cannot read property 'BD8B4C1B-4F8D-4430-A309-AD981645677D' of undefined
    at /Users/aensidhe/Sources/2015/dami/node_modules/noble/lib/mac/uuid-to-address.js:17:60
    at tryParseBuffer (/Users/aensidhe/Sources/2015/dami/node_modules/noble/node_modules/bplist-parser/bplistParser.js:25:5)
    at /Users/aensidhe/Sources/2015/dami/node_modules/noble/node_modules/bplist-parser/bplistParser.js:33:7
    at fs.js:334:14
    at FSReqWrap.oncomplete (fs.js:95:15)
````

aensidhe, 2015-04-16:
It seems that it connected to bluetooth cache.

[SO](http://stackoverflow.com/questions/23549859/clearing-corebluetooth-gatt-cache-without-removing-bond) helped me.

closed 2015-04-16 by aensidhe

sandeepmistry, 2015-04-16:
@aensidhe which version of OS X were you using?

I've pushed b6ff96b to cover the case where the ``CoreBluetoothCache`` section of ``/Library/Preferences/com.apple.Bluetooth.plist`` is empty.

aensidhe, 2015-04-17:
I'm using Yosemite 10.10.3.

 Thank you.