noble/noble #169: Reading Characteristics sequentially

ManuelOverdijk, 2015-04-17:
Today I encountered the following problem. When reading two characteristics of the same service in parallel with the code below, both services but no characteristics will be found. Running the same code sequentially with Promises will fix this. Before continuing to add a queue for communication operations, is this expected behavior, or is my code wrong?

``` javascript
this.peripheral.discoverServices([UUID], function(error, services) {
            var service = services[0];
            service.discoverCharacteristics([charUUID1], function(error, characteristics) {
                var characteristic = characteristics[0];
               characteristic.read(function(error, data) {});
                characteristic.once('read', function(data, isNotification) {
                      ...
                });
            });
        });
this.peripheral.discoverServices([UUID], function(error, services) {
            var service = services[0];
            service.discoverCharacteristics([charUUID2], function(error, characteristics) {
                var characteristic = characteristics[0];
               characteristic.read(function(error, data) {});
                characteristic.once('read', function(data, isNotification) {
                      ...
                });
            });
        });
```

sandeepmistry, 2015-04-18:
@ManuelOverdijk I'll need to take a closer look at this, it should work. Are you using OS X or Linux? Which version?

In the mean time, I suggest you only call discover services and characteristics once:

```javascript
this.peripheral.discoverServices([UUID], function(error, services) {
            var service = services[0];
            service.discoverCharacteristics([charUUID1, charUUID2 ], function(error, characteristics) {
                     // ....
            });
});
```

ManuelOverdijk, 2015-04-20:
I'm currently using Yosemite

zasoniak, 2015-04-27:
I suppose it may not be a bug. Program code (in general, not only this case) is processed much faster than bluetooth communication. The same issue exists in Android or C code. 
Therefore you need to manually introduce queue and runs communication operations sequentially, waiting for answer from previous one or just introducing some interval between 2 operations.
 
So addition question is: will this feature (queuing operations) be build into noble on the lower level?

ManuelOverdijk, 2015-04-30:
@sandeepmistry Any update on this?

sandeepmistry, 2015-05-02:
@ManuelOverdijk no updates since my original comment.

rzr, 2020-09-06:
Relate-to: https://github.com/noble/noble/issues/923#issuecomment-687851954