noble/noble #172: Problem installing on OSX - Cannot find module 'nan'

thegecko, 2015-04-23:
I see errors when installing on the following platform:

OSX: 10.9.5
NPM: 2.7.4
Node: v0.12.2

```
Error: Cannot find module 'nan'
    at Function.Module._resolveFilename (module.js:336:15)
    at Function.Module._load (module.js:278:25)
    at Module.require (module.js:365:17)
    at require (module.js:384:17)
    at [eval]:1:1
    at Object.exports.runInThisContext (vm.js:74:17)
    at Object.<anonymous> ([eval]-wrapper:6:22)
    at Module._compile (module.js:460:26)
    at evalScript (node.js:431:25)
    at startup (node.js:90:7)
gyp: Call to 'node -e "require('nan')"' returned exit status 1. while trying to load binding.gyp
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:343:16)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
gyp ERR! System Darwin 13.4.0
gyp ERR! command "node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/robmor01/Projects/noble_test/node_modules/noble/node_modules/xpc-connection
gyp ERR! node -v v0.12.2
gyp ERR! node-gyp -v v1.0.3
gyp ERR! not ok 
npm WARN optional dep failed, continuing xpc-connection@0.1.3
```
Although xpc-connection is marked as optional, it seems to be required when using noble:
```
module.js:338
    throw err;
          ^
Error: Cannot find module 'xpc-connection'
    at Function.Module._resolveFilename (module.js:336:15)
    at Function.Module._load (module.js:278:25)
    at Module.require (module.js:365:17)
    at require (module.js:384:17)
    at Object.<anonymous> (/Users/robmor01/Projects/noble_test/node_modules/noble/lib/mac/mavericks.js:7:21)
    at Module._compile (module.js:460:26)
    at Object.Module._extensions..js (module.js:478:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Module.require (module.js:365:17)
```

Any pointers?

sandeepmistry, 2015-04-23:
@thegecko I made it optional because it is not needed on Linux, as node-xpc-connection is OS X only. However, it is required on OS X.

How did you install noble, by running ```npm install noble```? Do you have a ```npm-debug.log``` to share?

thegecko, 2015-04-24:
I used `npm install node`.

Embarrassingly, the error had gone since a reboot .grrr.

closed 2015-04-24 by thegecko