noble/noble #173: Major and Minor Values?

kevinits, 2015-04-27:
While scanning, can you read the major and minor values of the peripheral beacons like you can with node-bleacons?

sandeepmistry, 2015-04-27:
@kevinits I'm trying to keep noble a generic BLE lib. Beacon specifics should go in node-bleacon.

Is there a reason for not using node-bleacon?

kevinits, 2015-04-28:
No, not really.  I'm just a newbie with bluetooth, trying to figure it all out.  That makes sense.  For my use case, I like the ability in noble to not see all of the duplicates.  In node-bleacon, is that a possibility or do I need to figure out how to filter them out myself?  Thanks for the quick reply.

sandeepmistry, 2015-04-28:
@kevinits please open an issue for this in the node-bleacon repo.

closed 2015-04-28 by sandeepmistry