noble/noble #174: Disconnect reason

rosek86, 2015-04-29:
Hi,

Is it possible to pass disconnect reason in the 'disconnect' event, please? This would be useful to determinate if connection was established or not. For instance due to noisy environment we receive 'Connection Failed to be Established' reason which should lead to quick re-connection:

> HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x3e
    Reason: Connection Failed to be Established

Thanks,
Krzysztof

sandeepmistry, 2015-04-29:
@rosek86 any suggestions (code snippets) on how to do this?

I don't think it will be possible on OS X, so might be a Linux only feature ....

rosek86, 2015-05-05:
@sandeepmistry haven't been able to check noble code yet, I'll try to look on this in a spare time.

It's a shame that OS X doesn't support this feature, I guess it's better to keep noble consistent on all systems.

sandeepmistry, 2015-05-16:
@rosek86 some notes on the OS X side:

XPC event on API initiated disconnect
```
  yosemite-bindings xpcEvent: {
  "kCBMsgId": 40,
  "kCBMsgArgs": {
    "kCBMsgArgDeviceUUID": [
      59,
      227,
      11,
      209,
      28,
      224,
      73,
      144,
      172,
      209,
      161,
      121,
      126,
      186,
      85,
      185
    ],
    "kCBMsgArgResult": 0
  }
```

Timeout or remote device connect:
```
  yosemite-bindings xpcEvent: {
  "kCBMsgId": 40,
  "kCBMsgArgs": {
    "kCBMsgArgDeviceUUID": [
      59,
      227,
      11,
      209,
      28,
      224,
      73,
      144,
      172,
      209,
      161,
      121,
      126,
      186,
      85,
      185
    ],
    "kCBMsgArgResult": 23
  }
```

So on OS X we might not be able to determine the exact reason (code), but when can tell if it's API initiated or not. Would this be "good enough"?

Have you made any progress on the Linux front?

rzr, 2021-01-26:
It looks the maintenance of this project is in pause, but feel free to contribute to @abandonware fork until upstream is active again

Feel free to comment:
https://github.com/abandonware/noble/issues/2