noble/noble #175: peripherals with duplicate services or characteristics are not handled properly

corecode, 2015-04-29:
All services and characteristics are accessed via hashes.  If a peripheral contains multiple services/characteristics of the same uuid, only the last discovered service/characteristic is accessible.

A possible solution would be to carry the handle to the objects, and perform accesses only using handles.

OJFord, 2015-07-10:
Same UUID characteristics in different services work fine for me. (Maybe this has been fixed but not closed?)

Are same-UUID services, or same-UUID characteristics within a service even legal?

sandeepmistry, 2015-07-19:
@OJFord it's still an issue.

Yes, a peripheral can have one or many services with the same UUID, and one or many characteristics with the same UUID in a sever.

VinceGa, 2015-11-25:
Yes, this is a very legit request. Currently we could read/write attributes via handles. But there's no way(not AFAIK) to get the handle of a specific attribute. We should add this property in the discover/read/write APIs.

sandeepmistry, 2015-11-28:
@VinceGa yes, it's on my TODO list, plan is to add an ```id``` field to all attribute objects (services, characteristics, descriptors). For most platforms noble supports it will be mapped to the attribute handle, however if that info is not available another unique number could be used.

P.S. reading attributes by handle works well on Linux, however on OS X CoreBluetooth/blued blocks this operation sometimes.

aiham, 2016-02-15:
I've run into this problem too. I cannot listen to notify events by a characteristic with the same UUID as another characteristic.

LarsBuur, 2016-02-18:
@sandeepmistry you wrote above: "P.S. reading attributes by handle works well on Linux, however on OS X CoreBluetooth/blued blocks this operation sometimes." 

How can I detect that the operation is blocked?

rzr, 2021-03-02:
Please can you update status of this issue on this latest release: 

https://www.npmjs.com/package/@abandonware/noble/v/1.9.2-12

OJFord, 2021-03-02:
@rzr I have no idea because I haven't used this project in years, but it looks like that package is being maintained (and is actually being maintained, not just hosted) at abandonware/noble, so you'll probably have better luck checking for a similar (perhaps closed) issue there, or asking via Discussions or something.