noble/noble #176: noble warns BLE state is unsupported state sporadically

LookingSharp, 2015-04-30:
When I register for state change I occasionally get `unsupported` for the state.  I'm not sure why this happens, but to get the expected `poweredOn` state I need to kill and restart the app.  Is there a way to recover from this from within the program?  A way to tell noble to reset?

```
noble warning: adapter does not support Bluetooth Low Energy (BLE, Bluetooth Smart).
               Try to run with environment variable:
               [sudo] NOBLE_HCI_DEVICE_ID=x node ...
ble: stateChange: state = unsupported
```

I'm running on a Raspberry Pi.

Thanks.

taktran, 2015-04-30:
I'm also getting this. I was thinking of running the app on a background thread and restarting it for now. But a restart method of some sort would be amazing!

haughank, 2015-04-30:
I have seen the same error on a Fujitsu Lifebook A Series running ubuntu 14.04 LTS. Doesn't seem to be a Raspberry Pi only issue. Again, restarting the application solves the problem

haughank, 2015-04-30:
Additional note: the frequency of this occurring is about 20% of the time that the app is launched

sandeepmistry, 2015-05-02:
@LookingSharp @taktran @haughank if you could provide some steps to reproduce that would be great. As well as the output of ``hcidump``.

Which BLE USB adapters are you using?

LookingSharp, 2015-05-02:
It is intermittent, so there aren't good repro steps.  Just getting the state change, which is the first thing the program does, causes the problem.

I'm trying to capture it in hcidump, but thus far I haven't had luck.  It isn't reproing for me at the moment.

I'm using the IOGear adapter
http://www.amazon.com/IOGEAR-Bluetooth-Micro-Adapter-GBU521/dp/B007GFX0PY/

LookingSharp, 2015-05-12:
Is there any way to work around this node?  Is there a way to shut down nobel and restart it from within Node?

LookingSharp, 2015-05-12:
@sandeepmistry This is the output for hcidump -

```
pi@raspberrypi ~/node $ hcidump
HCI sniffer - Bluetooth packet analyzer ver 5.30
device: hci0 snap_len: 1500 filter: 0xffffffff
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x0c
    Error: Command Disallowed
```

However it turns out if I ignore this and start scanning anyway, then I can discover peripherals, get their services, read characteristics.  Here's a much longer hcidump that includes instances of that.

```
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x0c                                  // note: I'm not sure if this first one
    Error: Command Disallowed                    // was from previous session
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x0c
    Error: Command Disallowed
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
> HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
> HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x3e
    Reason: Connection Failed to be Established
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
> HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
> HCI Event: Read Remote Ver Info Complete (0x0c) plen 8
    status 0x00 handle 64
    LMP Version: 4.1 (0x7) LMP Subversion: 0x2203
    Manufacturer: Broadcom Corporation (15)
> HCI Event: Command Status (0x0f) plen 4
    Read Remote Version Information (0x01|0x001d) status 0x00 ncmd 1
> HCI Event: Read Remote Ver Info Complete (0x0c) plen 8
    status 0x00 handle 64
    LMP Version: 4.1 (0x7) LMP Subversion: 0x2203
    Manufacturer: Broadcom Corporation (15)
> HCI Event: Command Status (0x0f) plen 4
    Read Remote Version Information (0x01|0x001d) status 0x00 ncmd 1
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
> HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
[... 207 lines cut ...]
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
> HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
> HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 65 reason 0x16
    Reason: Connection Terminated by Local Host
> HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
> HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 66 reason 0x16
    Reason: Connection Terminated by Local Host
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
> HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
> HCI Event: Command Complete (0x0e) plen 4
    LE Create Connection Cancel (0x08|0x000e) ncmd 1
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
```

sandeepmistry, 2015-05-14:
@LookingSharp the above it great! Code you also share the JS code that was run?

It looks like the kernel is not in a state that allows for scanning:
```
> HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1

> HCI Event: Command Complete (0x0e) plen 4
    LE Create Connection Cancel (0x08|0x000e) ncmd 1

> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
```

1. Create connection
2. Cancel create connection
3. Start scanning

LookingSharp, 2015-05-14:
@sandeepmistry Ok, but for this captured iteration I called start scanning anyway (on accident).  And each of those error `0x0c` are followed by an `LE Set Scan Enable` that succeeded `0x00`.  I included a lot of the dump there, but I'd start from the top (instead of the tail, which I cut off the output).  The very first disallowed is only initializing noble and handling the status event, after which I called `process.exit`.  The next one I let run - I got the status `unsupported`, then a few seconds later I started scanning.  And the scans worked as expected.  

I have a lot of BLE devices around, so the dump can be confusing since we don't know what device it is connecting to.  But they fall into a few categories - 

* Fitbits - These will connect, and I won't find my service.
* iBeacons - these will just fail to connect. They're not supposed to connect.
* iPhones - these will connect, sometimes disconnect but eventually succeed.  Some of the iPhones will respond to a service discovery of a specified UUID, which I can then read a characteristic from.  _This succeeded even though `state` was `unsupported`._

Looking at a section of the dump above, here we can see a 'failed scan enable' followed immediately by a successful enable, then a successful connect and read?
``` 
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
> HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
> HCI Event: Read Remote Ver Info Complete (0x0c) plen 8
    status 0x00 handle 64
    LMP Version: 4.1 (0x7) LMP Subversion: 0x2203
    Manufacturer: Broadcom Corporation (15)
> HCI Event: Command Status (0x0f) plen 4
    Read Remote Version Information (0x01|0x001d) status 0x00 ncmd 1
> HCI Event: Read Remote Ver Info Complete (0x0c) plen 8
    status 0x00 handle 64
    LMP Version: 4.1 (0x7) LMP Subversion: 0x2203
    Manufacturer: Broadcom Corporation (15)
> HCI Event: Command Status (0x0f) plen 4
    Read Remote Version Information (0x01|0x001d) status 0x00 ncmd 1
```
I don't really understand how to decode the dump, but I do know that I was successful in reading a characteristic while in the `unsupported` state.

Here's my current `stateChange` handler.
```javascript
noble.on('stateChange', function(state) {
    console.log(0, "             ble: stateChange: state = " + state);
	if (state === 'poweredOn') {
		myStartScanning();
	} else {
		// don't call stop
		//myStopScanning();                  
								// I had process.exit() here for the first run.
		// try to idle start later
		startScanningAtIdle = 1;
	}
});
```
And my idle scan starter will start scanning about 5 seconds later.
```javascript
	var allowDuplicates = true;
	var serviceUuids = [];
	var now = new Date();
	console.log(1, "             ble: startScanning  " + now);
	noble.startScanning(serviceUuids, allowDuplicates);
```

sandeepmistry, 2015-05-15:
@LookingSharp good information, but let's take a step back.

What would really help is the exact code you are running and the steps you go through that causes you to occasionally run into the issue. It would be even better is you get the code and steps down to a minimum. In conjunction with this, the ``hcidump`` with markers as to what was going on at that time code/step wise.

I'd really like to be able to reproduce this issue on my Raspberry Pi, then track down the root cause.

LookingSharp, 2015-05-15:
@sandeepmistry I'd love to get it down to the minimal code and steps, but it is an inconsistent repro which is why I was hoping the dump above is enough.  It repro'd twice in a row yesterday, but no luck at all today.

I've got a slimmed down app that I'll keep trying to get a repro on.  Wish me luck.

sandeepmistry, 2015-05-15:
@LookingSharp care to share the code you have so far?

LookingSharp, 2015-05-15:
@sandeepmistry Sure, here is a slimmed down version so far - though I haven't repro'd it with this slimmed version yet.  This is pretty simple - not doing any discovery of services or characteristics.
```javascript
var noble = require('noble');
var fs = require('fs');
var util = require('util');
var now = new Date().toISOString();
var logfile = now + '.log';
var log_file = fs.createWriteStream(__dirname + '/logs/' + logfile, {flags : 'w'});
var log_stdout = process.stdout;

console.log = function(d) {
	log_file.write(util.format(d) + '\n');
	log_stdout.write(util.format(d) + '\n');
}

function myStartScanning() {
	var allowDuplicates = true;
	// Scan for all Uuids.
	var serviceUuids = [];
	console.log("ble: startScanning")
	noble.startScanning(serviceUuids, allowDuplicates);
}

function myStopScanning() {
	console.log("ble: stopScanning");
	noble.stopScanning();
}

noble.on('stateChange', function(state) {
	console.log("ble: stateChange: state = " + state);
//	if (state === 'poweredOn') {
//		myStartScanning();
//		process.exit(0);
//	} else {
		myStartScanning();
//	}
});

noble.on('discover', function(peripheral) {
	var space = (peripheral.rssi > -100 ) ? " " : "";
	console.log("ble:" 
		+ " rssi=" + space + peripheral.rssi
		+ " uuid=" + peripheral.uuid
		+ " addr=" + peripheral.address
		+ " pow="  + peripheral.txPowerLevel
		+ " name=" + peripheral.advertisement.localName);

	if (!peripheral.connectAttached) {
		peripheral.connectAttached = 1;
		peripheral.on('connect', function(error) {
			console.log("connect " + peripheral.uuid + " (" + error + ")");
			peripheral.disconnect(function(error) {
				console.log("peripheral.disconnect " + peripheral.uuid);
				if (error) console.log("error: " + error);
				myStartScanning();
			});	
		});
	}

	myStopScanning();
	peripheral.connect();
});
```

sandeepmistry, 2015-05-15:
@LookingSharp thanks! So are you just leaving it running or do you start it and ctrl-c it + restart it?

LookingSharp, 2015-05-15:
Start, ctrl-c, restart.

General dev pattern is to run `node ble.js`, observe output, let it run while I churn on the code or do other things, sync new `ble.js`  to the pi, ctrl-c, `node ble.js`.

LookingSharp, 2015-05-15:
@sandeepmistry I updated the code above to add a `peripheral.on('disconnect'` handler, which is necessary if the connect times out during connection and no `peripheral.on('connect'` is sent.

LookingSharp, 2015-05-15:
For some reason sometimes I am getting this error now, which I haven't seen before.
```
pi@raspberrypi ~/node $ sudo node ble2.js
             ble: stateChange: state = poweredOn
             ble: startScanning
6947e19a2123 rssi= -59 uuid=6947e19a2123 addr=69:47:e1:9a:21:23 pow=undefined name=undefined
             ble: stopScanning
dee915e62123 rssi= -80 uuid=dee915e62123 addr=de:e9:15:e6:21:23 pow=undefined name=estimote
             ble: stopScanning
6947e19a2123 connect (null)
6947e19a2123 disconnected.
             ble: startScanning
6947e19a2123 rssi= -59 uuid=6947e19a2123 addr=69:47:e1:9a:21:23 pow=undefined name=undefined
             ble: stopScanning
dee915e62123 connect (null)
dee915e62123 disconnected.
             ble: startScanning
6947e19a2123 connect (null)
6947e19a2123 disconnected.
             ble: startScanning
41eb69bdf123 rssi= -59 uuid=41eb69bdf123 addr=41:eb:69:bd:f1:23 pow=undefined name=undefined
             ble: stopScanning
41eb69bdf123 connect (null)
41eb69bdf123 disconnected.
             ble: startScanning
6947e19a2123 rssi= -59 uuid=6947e19a2123 addr=69:47:e1:9a:21:23 pow=undefined name=undefined
             ble: stopScanning
6947e19a2123 connect (null)
6947e19a2123 disconnected.
             ble: startScanning
6947e19a2123 rssi= -59 uuid=6947e19a2123 addr=69:47:e1:9a:21:23 pow=undefined name=undefined
             ble: stopScanning
dee915e62123 rssi= -78 uuid=dee915e62123 addr=de:e9:15:e6:21:23 pow=undefined name=estimote
             ble: stopScanning
6947e19a2123 connect (null)
6947e19a2123 disconnected.
             ble: startScanning
dee915e62123 rssi= -81 uuid=dee915e62123 addr=de:e9:15:e6:21:23 pow=undefined name=estimote
             ble: stopScanning
/home/pi/node/node_modules/noble/lib/linux/l2cap-ble.js:232
      this._l2capBle.stdin.write(this._currentCommand.buffer.toString('hex') +
                    ^
TypeError: Cannot read property 'stdin' of undefined
    at L2capBle._queueCommand (/home/pi/node/node_modules/noble/lib/linux/l2cap-ble.js:232:21)
    at L2capBle.exchangeMtu (/home/pi/node/node_modules/noble/lib/linux/l2cap-ble.js:327:8)
    at nobleBindings.onConnect (/home/pi/node/node_modules/noble/lib/linux/bindings.js:114:29)
    at emit (events.js:110:17)
    at L2capBle.onStdoutData (/home/pi/node/node_modules/noble/lib/linux/l2cap-ble.js:98:12)
    at Socket.emit (events.js:107:17)
    at readableAddChunk (_stream_readable.js:163:16)
    at Socket.Readable.push (_stream_readable.js:126:10)
    at Pipe.onread (net.js:529:20)
```

ara2006, 2015-05-15:
Hello Sandeep

As we talked via e-mail I need to take the battery level and temperature from the sensoro brand beacons and when I run the code below 

var noble = require('./node_modules/noble');

noble.on('discover', function (peripheral) {
    peripheral.connect(function (error) {
        console.log('connected to peripheral: ' + peripheral.uuid);
        peripheral.discoverServices(['180f'], function (error, services) {
            var batteryService = services[0];
            console.log('discoveredBatter service', batteryService);
        });
    });
});

if (noble.state === 'poweredOn') {
    noble.startScanning([], true);
} else {
    noble.on('stateChange', function () {
        noble.startScanning([], true);
    });
}

I take the result

noble warning: adapter does not support Bluetooth Low Energy (BLE, Bluetooth Sma                                                                                                                                                             rt).
               Try to run with environment variable:
               [sudo] NOBLE_HCI_DEVICE_ID=x node ...
connected to peripheral: 0117c55cbd72
discoveredBatter service undefined

As you can see the battery service prints undefined. But when I change peripheral.discoverServices(['180f'] to peripheral.discoverServices(['1800'] the result seems okay. Below is the result of peripheral.discoverServices(['1800']

connected to peripheral: 0117c55fe57a
discoveredBatter service { _noble:
   { state: 'poweredOn',
     _bindings:
      { _addresses: [Object],
        _addresseTypes: [Object],
        _pendingConnection: false,
        _connectionQueue: [],
        _hciBle: [Object],
        _l2capBle: [Object],
        init: [Function],
        onStateChange: [Function],
        startScanning: [Function],
        onScanStart: [Function],
        stopScanning: [Function],
        onScanStop: [Function],
        onDiscover: [Function],
        connect: [Function],
        onConnect: [Function],
        disconnect: [Function],
        onDisconnect: [Function],
        exchangeMtu: [Function],
        onMtu: [Function],
        updateRssi: [Function],
        onRssi: [Function],
        discoverServices: [Function],
        onServicesDiscovered: [Function],
        discoverIncludedServices: [Function],
        onIncludedServicesDiscovered: [Function],
        discoverCharacteristics: [Function],
        onCharacteristicsDiscovered: [Function],
        read: [Function],
        onRead: [Function],
        write: [Function],
        onWrite: [Function],
        broadcast: [Function],
        onBroadcast: [Function],
        notify: [Function],
        onNotify: [Function],
        onNotification: [Function],
        discoverDescriptors: [Function],
        onDescriptorsDiscovered: [Function],
        readValue: [Function],
        onValueRead: [Function],
        writeValue: [Function],
        onValueWrite: [Function],
        readHandle: [Function],
        onHandleRead: [Function],
        writeHandle: [Function],
        onHandleWrite: [Function],
        onHandleNotify: [Function],
        _events: [Object],
        _scanServiceUuids: [] },
     _peripherals: { '0117c55fe57a': [Object] },
     _services: { '0117c55fe57a': [Object] },
     _characteristics: { '0117c55fe57a': [Object] },
     _descriptors: { '0117c55fe57a': [Object] },
     _events: { discover: [Function], stateChange: [Function] },
     _discoveredPeripheralUUids: [ '0117c55fe57a' ],
     _allowDuplicates: true },
  _peripheralUuid: '0117c55fe57a',
  uuid: '1800',
  name: 'Generic Access',
  type: 'org.bluetooth.service.generic_access',
  includedServiceUuids: null,
  characteristics: null }

LookingSharp, 2015-05-15:
Likely it didn't find your service at all.  When you get called back, I believe services could be empty.  Try something like this -

```javascript
	    peripheral.on('servicesDiscover', function (services) {
	        console.log(peripheral.uuid + " peripheral.discoverServices " + "(" + services.length + ") ");
	        if (services.length == 0) {
	            console.log(peripheral.uuid + " No services found.");
	            peripheral.disconnect();
	            return;
	        }

	        // find your service
	        var theService = null;
	        var serviceUuid = '180f';
	        for (var i = 0; i < services.length; i++) {
	            if (services[i].uuid === serviceUuid) {
	                if (theService != null) {
	                    console.log(peripheral.uuid + " found multiple serviceUuid (" + i + ")");
	                    continue;
	                }
	                theService = services[i];
	            }
	        }
	    // now do something with theService
	    }
```

sandeepmistry, 2015-05-16:
@ara2006 Let's keep issues focused on a single topic, to avoid noise.

It would be best to ask your question on gitter: https://gitter.im/sandeepmistry/noble

sandeepmistry, 2015-05-16:
@LookingSharp I've pushed a change (https://github.com/sandeepmistry/noble/commit/be17d4f25b18081398d860350c50e32c11e7527a) for you to try out. You'll need to pull the changes and then run ```node-gyp rebuild```

I was able to reproduce the by running ```sudo hcitool lescan``` in a separate window then start noble start scanning. It might not exactly what you we're seeing, hard to say.

sandeepmistry, 2015-08-02:
@LookingSharp any updates on this? Can we close?

sandeepmistry, 2015-08-10:
Closing due to inactivity ...

closed 2015-08-10 by sandeepmistry