noble/noble #177: Error: connect(hci0): Interrupted system call

LookingSharp, 2015-05-01:
When trying to do a connect, I received this error in the on 'connect'
```
	peripheral.on('connect'...
    error: Error: connect(hci0): Interrupted system call
	peripheral.on('disconnect'...
```

After hitting this I did get the on 'disconnect'. But calling startScanning returned no discovers (there were about 8 devices around). Has anyone seen this before?

On a raspberry Pi.

LookingSharp, 2015-05-01:
I think this is a bug on my side, but in case someone else sees it this may help.

Here's my general model -

Scan for all peripherals, duplicates allowed.
  On discover of a peripheral, stop scanning.
  Connect to the peripheral.
  Discover Services
  Discover characteristics
  Disconnect.
  startScanning

However if I find that I've been trying to get data from a peripheral for over 10 seconds, then I just assume failure and call disconnect (with some code to ensure I do this only once).

When I received the error above I had done a peripheral.connect, decided it had taken too long, called peripheral.disconnect.  I then received my on('connect'), got the error above, ignored it (my bug), and called discoverServices)().  I then received my on('disconnect').

So I'm definitely doing something wrong - when I receive an error on connect I shouldn't proceed with discoverServices.

Thanks.

sandeepmistry, 2015-05-02:
@LookingSharp seems like you've solved the problem, can we close this issue?

LookingSharp, 2015-05-02:
I had thought that calling DiscoverServices after receiving an error on connect caused this problem.  However I fixed that and still see it.  So my current fix is to never call disconnect on a peripheral while it is in progress on an async operation.  I wanted to disconnect because I wanted to impose a timeout on connect time.

If disconnect should be supported while performing another operation, then the issue can remain open.  If you don't plan to support that then it can be closed.

sandeepmistry, 2015-08-12:
noble now uses bluetooth-hci-socket instead of spawned (hci-ble and l2cap-ble) processes.

This issue should be resolved now.

closed 2015-08-12 by sandeepmistry