noble/noble #178: will it be possible to connect two bluetooth LE peripherals simultaneously?

xmlking, 2015-05-04:
Env:  Mac OS X 10.9.5 
Recently I tried to add two bluetooth LE peripherals ( LightBlue Bean & SensorTag)  to node-RED and  log data to debug. this attempt failed but I can able to connect one peripheral at time. 
wonder if I can connect two peripherals and consume data simultaneously? if yes could you point me to an example?

BorisKourt, 2015-05-04:
I have been able to connect to many BLE peripherals simultaneously using just noble itself. 

Simply keep scanning for further devices and when you find what you want to connect to run `peripheral.connect([callback(error)]);` with the discovered peripheral. *(via `noble.on('discover', function(peripheral) { ... });`)*

It is up to you how you wish to store/interact with peripheral information for multiple devices.
***
Your issue may be with a library that uses noble, and not noble itself. (If you are using one)

vielmetti, 2015-05-04:
I've seen the same issue, and there needs to be a rethink of how Node-RED interacts with BLE. It probably makes sense to have a single BLE or "node-red-noble" package that knows how to deal with a lot of individual sensors, rather than to have completely broken out support for each sensor.

I suspect this issue belongs in some other repo, though, since noble itself does have the support that @BorisKourt describes.

sandeepmistry, 2015-05-04:
@BorisKourt @vielmetti thanks for helping out!

@xmlking thethingsystem's steward had something like @vielmetti mentioned, it would have central scanner and then see if any of the drivers supported it: https://github.com/TheThingSystem/steward

This fits in with [noble-device](https://github.com/sandeepmistry/noble-device), module that are built on top of it will have a ```.is``` method, which takes a peripheral object as a argument, and returns if the module supports it. Then you can pass the peripheral in when you create a new object.

One other note, node-sensortag v1.0.0 is built-on noble-device, and I believe the LightBlue module is as well, so the discover method's on both should be compatible when used in the same app.

closed 2015-05-04 by sandeepmistry