noble/noble #179: Notify for 2 characteristics on different services

Lock-up, 2015-05-05:
Hi,
I have a Sensortag here and trying to get notified for the temperature and light sensor.

If I use following code I only get notified for the light sensor:

    var noble = require('noble');

    function initLightService(peripheral){
			peripheral.discoverServices(['f000aa7004514000b000000000000000'], function(error, services) {
        console.log('light service discovered');
		    var service = services[0];
		    service.discoverCharacteristics(['f000aa7204514000b000000000000000'], function(error, characteristics) {
			    console.log('characteristic enable discovered');
        	var characteristic = characteristics[0];
			    characteristic.write(new Buffer([0x01]), false, function(error) {
			    	console.log('set enable to (1)');
				
				    service.discoverCharacteristics(['f000aa7104514000b000000000000000'], function(error, characteristics) {
					    console.log('characteristic value discovered');
        			var characteristic = characteristics[0];
					    characteristic.on('read', function(data, isNotification) {
					      console.log('light value is: ', data.readUInt8(0));
        			});
					    characteristic.notify(true, function(error) {
					      console.log('light value notification on');
        			});
      			});
        	});
      	});
			});
		}

    noble.on('discover', function(peripheral) {
	    peripheral.connect(function(error) {
		    console.log('connected to peripheral: ' + peripheral.uuid);
		    peripheral.discoverServices(['f000aa0004514000b000000000000000'], function(error, services) {
	      	console.log('temp service discovered');
			    var service = services[0];
			    service.discoverCharacteristics(['f000aa0204514000b000000000000000'], function(error, characteristics) {
				  console.log('characteristic enable discovered');
        		var characteristic = characteristics[0];
				    characteristic.write(new Buffer([0x01]), false, function(error) {
				      console.log('set enable to (1)');
					
					    service.discoverCharacteristics(['f000aa0104514000b000000000000000'], function(error, characteristics) {
						    console.log('characteristic value discovered');
        				var characteristic = characteristics[0];
						    characteristic.on('read', function(data, isNotification) {
						      console.log('temp value is: ', data.readUInt8(0));
        				});
						    characteristic.notify(true, function(error) {
						      console.log('temp value notification on');
							    initLightService(peripheral);
        				});
      				});
        		});
      		});
		    });
  	  });
    });
    noble.startScanning();

Using this code I get notified for both sensors:

    var noble = require('noble');

    noble.on('discover', function(peripheral) {
	    peripheral.connect(function(error) {
		    console.log('connected to peripheral: ' + peripheral.uuid);
		    peripheral.discoverServices(['f000aa0004514000b000000000000000', 'f000aa7004514000b000000000000000'], function(error, services) {
	      	console.log('temp service discovered');
			    var service = services[0];
			    service.discoverCharacteristics(['f000aa0204514000b000000000000000'], function(error, characteristics) {
				    console.log('characteristic enable discovered');
        		var characteristic = characteristics[0];
				    characteristic.write(new Buffer([0x01]), false, function(error) {
				      console.log('set enable to (1)');
					
					    service.discoverCharacteristics(['f000aa0104514000b000000000000000'], function(error, characteristics) {
						    console.log('characteristic value discovered');
        				var characteristic = characteristics[0];
						    characteristic.on('read', function(data, isNotification) {
						      console.log('temp value is: ', data.readUInt8(0));
        				});
						    characteristic.notify(true, function(error) {
						      console.log('temp value notification on');

							    console.log('setup light sensor');
							    var service = services[1];
							    service.discoverCharacteristics(['f000aa7204514000b000000000000000'], function(error, characteristics) {
								    console.log('characteristic enable discovered');
					        	var characteristic = characteristics[0];
								    characteristic.write(new Buffer([0x01]), false, function(error) {
								      console.log('set enable to (1)');
				
									    service.discoverCharacteristics(['f000aa7104514000b000000000000000'], function(error, characteristics) {
										    console.log('characteristic value discovered');
					        			var characteristic = characteristics[0];
										    characteristic.on('read', function(data, isNotification) {
										      console.log('light value is: ', data.readUInt8(0));
					        			});
										    characteristic.notify(true, function(error) {
										      console.log('light value notification on');
					        			});
					      			});	
					        	});
					      	});
        				});
      				});
        		});
      		});
		    });
  	  });
    });
    noble.startScanning();

In the first snippet the services are discovered one after another, in the second they are discovered with one call to `peripheral.discoverServices`.
I recognized that when I setup the notify for the temperature sensor all works until I discover a new service.

Maybe a bug and if not I hope it helps others with this issue.

Greetings

sandeepmistry, 2015-05-05:
@Lock-up are you using OS X or Linux? Which version/distro?

It might be related to #169 ...

Lock-up, 2015-05-06:
Im using OS X Yosemite (10.10.3)
Maybe its related, but my code is discovering the services and is reading/writing the characteristics sequentially.
I think the problem is calling `peripheral.discoverServices` multiple times.

ManuelOverdijk, 2015-05-06:
I'm experiencing the same behavior on Yosemite, I can only read the latest notify. 

Work-around by @Lock-up works though.

navesta, 2016-02-17:
Hi
Having a similar issue on Intel Edison, edison 3.10.17-poky-edison+,  running node v0.10.38, and Nobel version 0.1.9. If I set 2 notify handlers on 2 different characteristics, it seems the 2nd one always points to the 1st. I've tried 1) using   discoverSomeServicesAndCharacteristics, with both services  and both characteristics as arguments, 2) calling discoverSomeServicesAndCharacteristics twice, with single service and characteristic, and 3) calling discoverServices along discoverCharacteristics for each pair of service and characteristic:
1) var serviceUUID=["481d178c10dd11e4","68b527384a0440e18f"]; var charUUID=["4e92f4abc01b4b5ab","7a5433056b9e4878"];
	    peripheral.discoverSomeServicesAndCharacteristics(serviceUUID,charUUID, sensNotify);
2) var serviceUUID=["481d178c10dd11e4"]; var charUUID=["4e92f4abc01b4b5ab"];
	    peripheral.discoverSomeServicesAndCharacteristics(serviceUUID,charUUID, sensNotify);...
3) peripheral.discoverServices(["481d178c10dd11e4"], function(error, services) {
	    	services[0].discoverCharacteristics(["4e92f4abc01b4b5ab"], sensNotify01);
	    }); ...
Any thoughts?
Cheers,

sandeepmistry, 2016-02-21:
@navesta you need to use 16-bit or 128-bit UUID's.

navesta, 2016-02-22:
Thanks Sandeep. Indeed, I was in fact using 128 bit UUIDs - just did not copied them over - my bad. Problem was that I had not taken into account the async nature of Noble characteristics callback, i.e., I was just looping through the list characteristics. The solution was to use  async.whilst and loop through each characteristics handler serially.

TorstenRobitzki, 2016-08-16:
Maybe me issue is related to this. I have _one_ service with an 128 bit UUID, that contains 3 characteristics from which I expect to see notifications for two of them:

    control_point_char.on 'data', (data, is_notification)->
        control_point_callback( data, is_notification )

    progress_char.on 'data,', (data, is_notification)->
        progress_callback( data, is_notification )

    control_point_char.notify true, (error)->
        if error
            cb(null, error)
        else
            progress_char.notify true, (error)->
                if error
                    cb(null, error)
                else
                    cb(peripheral, null)

In a sniffer, I see that my client subscribes to both characteristics and I see notifications send by the GATT server. I see that the first characteristics callback is called, while the second is not called.