noble/noble #180: Cache discovery data

rosek86, 2015-05-08:
Hi,

Is it possible in noble to cache discovery data and use this during next connection, please? Discovery takes time and possibly it's not necessary to this each time.

Thanks for any hint,
Krzysztof

sandeepmistry, 2015-05-09:
@rosek86 are you referring to peripheral discovery or service/characteristic/descriptor discovery?

rosek86, 2015-05-11:
@sandeepmistry I'm thinking of service/characteristic/descriptor discovery, I think it should be possible to only cache characterisic handlers, so this part is not needed when central connects to the pheripleral multiple times.

sandeepmistry, 2015-05-11:
@rosek86 any thoughts on how to handle service/characteristic/descriptor changing on peripheral between connects?

Were you thinking the cache would be in-memory only or also on disk?

rosek86, 2015-05-12:
@sandeepmistry I'm not sure how to do this for now. I believe it should work at low level, at least from my experience, it seems that service/characteristic handlers doesn't change. We should be able to write/read using the same handlers each time. 

I was thinking of in-memory cache for now, run discovery only once for each device when the application starts and store this information in array.

I'll try to look into this.

sandeepmistry, 2015-05-12:
@rosek86 the service/characteristic handler can change (especially for people creating custom peripherals), however probably not on production devices. I'm worried this will cause confusion.

iOS caches and it caused some confusion for people create peripherals with [bleno](https://github.com/sandeepmistry/bleno).

rosek86, 2015-05-12:
@sandeepmistry ah, I wasn't thinking of adding cache into noble itself but I'm looking for possibility to save handlers in e.g. service.discoverCharacteristics() callback, and then recall these handlers on next connection. The user could decide if cache is needed or not.

However I've just noticed that this wouldn't be easy in noble, handlers are not passed to noble user and are only available in nobleBindings. Correct me if I'm wrong, such solution would require modification of all binding?

Thanks

sandeepmistry, 2015-05-13:
@rosek86 got it, that makes sense, I'll have to check the behaviour on OS X again, ``blued`` didn't like it if you didn't go through the discovery process.

This can potentially be incorporated with the work need for #179

rosek86, 2015-05-13:
Many thanks, this feature would decrease connection time by around 2 seconds in our application. We're looking for any possibility to decrease connection time,- modification of l2cap-ble.c sets connection interval to 7.5 ms. We'd like to send also 6 packets per connection interval but haven't got luck with this yet.  

Let me know if I can help with this, I don't know noble library but I could help with parts of implementation if it's useful.

sandeepmistry, 2015-08-18:
This enhancement will not work with OS X, closing for now ....

Btw, the initial connection interval is more aggressive now on Linux.

closed 2015-08-18 by sandeepmistry