noble/noble #183: Set peripheral.connect timeout?

duyhotan2000, 2015-05-13:
Hello,
I called peripheral.connect function but the peripheral is turned off and the function hang period of time (I don't know exactly) before it call the callback with error. So can you have a way to set this timeout?
Thanks!

sandeepmistry, 2015-05-13:
@duyhotan2000 no way to have a timeout currently.

duyhotan2000, 2015-05-13:
OK. But you have a way to cancel connecting or clear all peripheral cache and re-scanning?

LookingSharp, 2015-05-13:
On Raspian I have experimented with setting a timer and manually calling disconnect peripheral before a connect returns.  The behavior was inconsistent and caused some problems with scanning, so I disabled it.

My general pattern is to start scanning, then when I discover a peripheral I stop scanning, then I connect, then when my connect is completed I start scanning again.  It's a bit of a pain, but that's the only way I was able to continue scanning.

duyhotan2000, 2015-05-14:
@LookingSharp I have 2 devices and I push two of them to a queue and try it again when it fails to connect. But if a device is turn off, it get stuck in connect function. Do you have a solution for this?

alfian777, 2015-05-20:
hi @duyhotan2000 , what device you are working on? i tried to simulate your condition with my TI sensortag, and implement @LookingSharp Solution by manually call 

```
peripheral.disconnect(function(err){
});
```

And everything works fine.

rzr, 2021-03-02:
Please can you update status of this issue on this latest release: 

https://www.npmjs.com/package/@abandonware/noble/v/1.9.2-12