noble/noble #184: No callback after characteristic.write

buissonth, 2015-05-15:
Hello,

I am try communicating between my iPhone iOS8.3 and my Edison.
But I have no callback when I write on characteristic.
However, it works when I replace my iPhone by my Macbook air.

You can find the debug :
root@edison:~/node_modules/noble/examples/pizza# DEBUG=l2cap-ble node imu.js                                                                                                                      
Scanning BLE device...                                                                                                                                                                            
found peripheral: { localName: undefined,                                                                                                                                                         
  txPowerLevel: undefined,                                                                                                                                                                        
  manufacturerData: undefined,                                                                                                                                                                    
  serviceData: [],                                                                                                                                                                                
  serviceUuids: [ '13333333333333333333333333333337' ] }                                                                                                                                          
  l2cap-ble 4d:96:c9:55:9f:84: l2capBle = /home/root/node_modules/noble/lib/linux/../../build/Release/l2cap-ble +0ms                                                                              
  l2cap-ble 4d:96:c9:55:9f:84: buffer = "info using 98:4F:EE:03:26:26@hci0\n" +16ms                                                                                                               
  l2cap-ble 4d:96:c9:55:9f:84: line = info using 98:4F:EE:03:26:26@hci0 +1ms                                                                                                                      
  l2cap-ble 4d:96:c9:55:9f:84: buffer = "connect success\n" +5s                                                                                                                                   
  l2cap-ble 4d:96:c9:55:9f:84: line = connect success +2ms                                                                                                                                        
  l2cap-ble 4d:96:c9:55:9f:84: write: 020001 +9ms                                                                                                                                                 
  l2cap-ble 4d:96:c9:55:9f:84: buffer = "write = success\n" +8ms                                                                                                                                  
  l2cap-ble 4d:96:c9:55:9f:84: line = write = success +1ms                                                                                                                                        
  l2cap-ble 4d:96:c9:55:9f:84: buffer = "data 08100010000328" +218ms                                                                                                                              
  l2cap-ble 4d:96:c9:55:9f:84: buffer = "data 08100010000328\ndata 039e00\n" +1ms                                                                                                                 
  l2cap-ble 4d:96:c9:55:9f:84: line = data 08100010000328 +0ms                                                                                                                                    
  l2cap-ble 4d:96:c9:55:9f:84: ignoring request/command ... +1ms                                                                                                                                  
  l2cap-ble 4d:96:c9:55:9f:84: line = data 039e00 +0ms                                                                                                                                            
  l2cap-ble 4d:96:c9:55:9f:84: new MTU is 158 +1ms                                                                                                                                                
  l2cap-ble 4d:96:c9:55:9f:84: write: 100100ffff0028 +0ms                                                                                                                                         
  l2cap-ble 4d:96:c9:55:9f:84: buffer = "write = success\n" +2ms                                                                                                                                  
  l2cap-ble 4d:96:c9:55:9f:84: line = write = success +0ms                                                                                                                                        
  l2cap-ble 4d:96:c9:55:9f:84: buffer = "data 11060100050000180600" +175ms                                                                                                                        
  l2cap-ble 4d:96:c9:55:9f:84: buffer = "data 1106010005000018060009000118\n" +2ms                                                                                                                
  l2cap-ble 4d:96:c9:55:9f:84: line = data 1106010005000018060009000118 +0ms                                                                                                                      
  l2cap-ble 4d:96:c9:55:9f:84: write: 100a00ffff0028 +2ms                                                                                                                                         
  l2cap-ble 4d:96:c9:55:9f:84: buffer = "write = success\n" +2ms                                                                                                                                  
  l2cap-ble 4d:96:c9:55:9f:84: line = write = success +0ms                                                                                                                                        
  l2cap-ble 4d:96:c9:55:9f:84: buffer = "data 11140a000e006643ae10" +114ms                                                                                                                        
  l2cap-ble 4d:96:c9:55:9f:84: buffer = "data 11140a000e006643ae107948f8a59145b4bb781e61d0\n" +1ms
    l2cap-ble 4d:96:c9:55:9f:84: line = data 11140a000e006643ae107948f8a59145b4bb781e61d0 +0ms                                                                                                      
  l2cap-ble 4d:96:c9:55:9f:84: write: 100f00ffff0028 +1ms                                                                                                                                         
  l2cap-ble 4d:96:c9:55:9f:84: buffer = "write = success\n" +2ms                                                                                                                                  
  l2cap-ble 4d:96:c9:55:9f:84: line = write = success +0ms                                                                                                                                        
  l2cap-ble 4d:96:c9:55:9f:84: buffer = "data 11060f0012" +116ms                                                                                                                                  
  l2cap-ble 4d:96:c9:55:9f:84: buffer = "data 11060f0012000f18130018000518\n" +1ms                                                                                                                
  l2cap-ble 4d:96:c9:55:9f:84: line = data 11060f0012000f18130018000518 +1ms                                                                                                                      
  l2cap-ble 4d:96:c9:55:9f:84: write: 101900ffff0028 +0ms                                                                                                                                         
  l2cap-ble 4d:96:c9:55:9f:84: buffer = "write = success\n" +2ms                                                                                                                                  
  l2cap-ble 4d:96:c9:55:9f:84: line = write = success +0ms                                                                                                                                        
[... cut ...]
              else {
                console.log('missing characteristics');
              }
            })
          })
        })
      })
    })
}

function sendData() {
    console.log('sendData ...');
    // ACCEL
    var accel = new Buffer(1);
    accel.writeUInt8(9, 0);
    console.log('Write Accel ...');
    imuAccelCharacteristic.write(accel, false, function(err) {
        if (!err) {
        	console.log('ACCEL OK');
        }
        else {
        	console.log('ACCEL error');
        }
    });
}

sandeepmistry, 2015-05-16:
@buissonth a write error is received:
```
01 12 3100 0e 
```

0x0e maps to the error code "unlikely". Write errors are currently not handling ...

rzr, 2021-02-15:
It looks the maintenance of this project is in pause, but feel free to contribute to @abandonware fork until upstream is active again

Feel free to comment:
abandonware#2