noble/noble #186: Duplicate discover not working

duyhotan2000, 2015-05-23:
Hello,

I use function noble.startScanning([],true) to discover peripherals with duplicate option, but it call the discover callback only first time, also disconnect the connected peripheral

This function works on my computer but doesnt work on my friend computer, both have Ubuntu 14.04 installed. My Raspberry Pi with Raspbian also doesnt work.

sandeepmistry, 2015-05-23:
@duyhotan2000 does ```sudo hcitool lescan --duplicates``` work?

Are you both using the same USB Bluetooth 4.0 adapter? What about kernel version?

duyhotan2000, 2015-05-24:
On my computer and my friend's both installed Ubuntu 14.04 with Linux 3.12 kernel.
My computer works fine on built-in Atheros bluetooth adapter, and my friend's adapter is built-in Broadcom chipset

On the Raspberry Pi I use iBuffalo adapter with Cambridge Silicon Radio chipset with latest 3.18 kernel. When I run 'sudo hcitool lescan --duplicates' and without '--duplicates', it has the same output (definitely print out the device, my SensorTag). But on my computer with Atheros and Ubuntu, it prints only 1 time (without '--duplicates')
[image]

Please help me!

sandeepmistry, 2015-05-31:
@duyhotan2000 have you tried an alternative Bluetooth 4.0 USB adapter? Have you tried the CSR on your Ubuntu computer?

This looks more like a kernel/driver/hardware issue than a noble issue, as ```hcitool``` is producing the same behaviour.

duyhotan2000, 2015-06-01:
WIth build-in adapter on my computer, when I use kernel 3.13 it's work well, but not with newer kernel
On my RPi, it looks like a issue with my CSR adapter, which dont work with my Ubuntu computer

Thanks for your answer!

sandeepmistry, 2015-06-01:
@duyhotan2000 no problem. I'm closing as it's not an issue with noble ...

closed 2015-06-01 by sandeepmistry

duyhotan2000, 2015-06-02:
Is noble tested with Raspbian and can you tell me the kernel version and bluez you used?

sandeepmistry, 2015-06-02:
@duyhotan2000 Rasbian, BlueZ 4.99.

```sh
pi@raspberrypi ~ $ uname -a
Linux raspberrypi 3.18.11+ #781 PREEMPT Tue Apr 21 18:02:18 BST 2015 armv6l GNU/Linux
```

```sh
pi@raspberrypi ~ $ lsusb 
Bus 001 Device 002: ID 0424:9512 Standard Microsystems Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. 
Bus 001 Device 004: ID 1a40:0101 Terminus Technology Inc. 4-Port HUB
Bus 001 Device 005: ID 0a5c:21e8 Broadcom Corp. 
```

duyhotan2000, 2015-06-03:
Thank you so much! I gotta buy a Broadcom chipset adapter (Pluggable BT4LE)
Noble is a great module. I used it for my graduate thesis.

sandeepmistry, 2015-06-03:
@duyhotan2000 sounds good! You can also try out these adapters: https://github.com/sandeepmistry/noble/wiki/Compatible-devices#linux

Is your graduate thesis online? I'd love to take a look at it if it is ...

duyhotan2000, 2015-06-10:
I have just uploaded my project to github https://github.com/duyhotan2000/BLEGateway
You can run it by command "node main-server.js" and browse "localhost:8081" with user: admin, pass: admin

duyhotan2000, 2015-06-10:
@sandeepmistry I have bought a Plugable dongle and it works well on Raspberry Pi with 3.13 kernel, but not with latest 3.18 kernel. It can only connect to one peripheral at a time. Please help me!

sandeepmistry, 2015-06-24:
@duyhotan2000 I'm away from home currently, but I believe no issues with the 3.18 kernel and multiple connections.

Please open a new issue for this, as it is unrelated to the original topic. Also, the results for scanning via ```hcitool``` while attempting to connect to multiple devices manually using ```gatttool``` just to rule out noble vs the kernel.