noble/noble #191: l2cap-ble unhandled exceptions

rosek86, 2015-06-02:
Hi, occasionally I receve these two unhandled exceptions from noble:

fatal: uncaughtException: TypeError: Cannot read property 'stdin' of undefined
at L2capBle._queueCommand (/usr/jrd/app/node_modules/noble/lib/linux/l2cap-ble.js:232:21)
at L2capBle.exchangeMtu (/usr/jrd/app/node_modules/noble/lib/linux/l2cap-ble.js:327:8)
at nobleBindings.onConnect (/usr/jrd/app/node_modules/noble/lib/linux/bindings.js:114:29)
at EventEmitter.emit (events.js:98:17)
at L2capBle.onStdoutData (/usr/jrd/app/node_modules/noble/lib/linux/l2cap-ble.js:98:12)
at Socket.EventEmitter.emit (events.js:95:17)
at Socket.<anonymous> (_stream_readable.js:745:14)
at Socket.EventEmitter.emit (events.js:92:17)
at emitReadable_ (_stream_readable.js:407:10)
at emitReadable (_stream_readable.js:403:5)

76:be:f2:d8:22:81: stderr: Could not create connection: Connection timed out
2015-06-02T08:55:42.887Z - fatal: uncaughtException: TypeError: Cannot call method 'kill' of undefined
at L2capBle.kill (/usr/jrd/app/node_modules/noble/lib/linux/l2cap-ble.js:73:18)
at nobleBindings.onDisconnect (/usr/jrd/app/node_modules/noble/lib/linux/bindings.js:142:27)
at EventEmitter.emit (events.js:95:17)
at L2capBle.onStdoutData (/usr/jrd/app/node_modules/noble/lib/linux/l2cap-ble.js:100:12)
at Socket.EventEmitter.emit (events.js:95:17)
at Socket.<anonymous> (_stream_readable.js:745:14)
at Socket.EventEmitter.emit (events.js:92:17)
at emitReadable_ (_stream_readable.js:407:10)
at emitReadable (_stream_readable.js:403:5)
at readableAddChunk (_stream_readable.js:165:9)
at Socket.Readable.push (_stream_readable.js:127:10)

sandeepmistry, 2015-06-02:
@rosek86 what devices do you have in range?

rosek86, 2015-06-03:
@sandeepmistry We have a few property devices all based on nRF51822, usually noble connects to these devices without any problem but very occasionally our application crashes due to these exceptions.

sandeepmistry, 2015-08-12:
noble now uses bluetooth-hci-socket instead of spawned (hci-ble and l2cap-ble) processes.

This issue should be resolved now.

closed 2015-08-12 by sandeepmistry