noble/noble #193: "once" internal events may never be removed (and be incorrectly triggered later)

bragma, 2015-06-04:
Hi, while working with noble we've found several problems related to devices being disconnected while in-between noble calls.The disconnection is not a problem by itself, but those "once" events set in the various methods are, since they are never removed and may trigger later.

The typical noble method implementation pattern is:

```
Characteristic.prototype.broadcast = function(broadcast, callback) {
  if (callback) {
    this.once('broadcast', function() {
      callback(null);
    });
  }

  this._noble.broadcast(this._peripheralUuid, this._serviceUuid, this.uuid, broadcast);
};
```

The problem is that if the device disconnects after the event listener is set, the event will never be triggered or even worse it will be triggered on the next call since peripherals/services/characteristics are cached. Depending on the methods involved, this can manifest in different misbehavior of noble.

In my case (connect -> discover services -> discover characteristics -> set notify), if a disconnection happens at the wrong time, this causes notifications not to work on the next connection to the same device.
At the moment, the only solution I've found is not to pass the callback function to the noble method and to bind to the event in my code. I store the event handler, so on device disconnection I can remove the listener manually. But this is ugly.
Second, this problem is also related to the (missing) error handling in the various calls. If something goes wrong, callbacks will never be called. The nodejs common pattern for callbacks function(err, res) is not fully implemented.

I was wondering how to "fix" the problem but all the solutions I've found involve major changes to noble, sometimes even reflecting to its public interface. I think that the best solution would be to:

1. Properly propagate errors in events, so that callbacks are called in the event of an error
2a. Either trigger all event types with an "Error" parameter on disconnect or
2b. Keep a list of all handlers so that they can be removed/signaled of the error or
2c. When the l2cap-ble (or any other binding) process dies/disconnects, to process the commands queue and trigger the events for each queued command signaling the error.

Step 1 is not really complex. The method above would become:

```
Characteristic.prototype.discoverDescriptors = function(callback) {
  if (callback) {
    this.once('descriptorsDiscover', function(error, descriptors) {
      callback(error, descriptors);
    });
  }
...
```

But then all emits need to be changed for the new parameter:

```characteristic.emit('descriptorsDiscover', null, descriptors_);```

And in case of error, something like:

```characteristic.emit('descriptorsDiscover', new Error('disconnected'));```

Step 2 involves more work, depending on which option is chosen.
2a. works but it's plain ugly.
2b. is a bit better: it needs to keep an array of objects storing EventEmitter and event name, so that events can triggered on a disconnect or also keep the listener if we want to just remove the listener.
2c. is probably the best but requires a major reworking of the bindings too.

Feel free to comment/elaborate. Thanks!

sandeepmistry, 2015-06-04:
@bragma correct, I've been delaying fixing this, but it needs to be addressed.

I agree with your assessment, step 1 isn't too bad and suggestion 2a should not be that ugly. I think it just needs to trickle down the disconnect event to services, characteristics, descriptors from peripheral. The tricky part is knowing what operation is pending, and if there are any non-internal listeners attached when the disconnected error event is dispatched.

Thoughts?

duyhotan2000, 2015-06-05:
I remembered that sandeepmistry has temporarily fixed this problem in a topic before

bragma, 2015-06-05:
The idea is to emit for all possible event types, so we don't need to know which operation is pending. If none is listening, the emit will go waste but not cause any problem. I don't see the need for discriminating between  internal and "non-internal" listeners: non internal will still be waiting for events that will never come, if the peripheral disconnects. If they receive an error, they should be able to handle or ignore it. Even better, I think that event non internal listeners should know what happened, else they'll have the exact same problem we have to the internal ones.

@duyhotan2000 I've found this discussion https://github.com/sandeepmistry/noble/issues/111 where @sandeepmistry solves the problem in another way, by adding another event listener for disconnect. It has never been implmented, though.

sandeepmistry, 2015-06-06:
@bragma I'd like to keep the same consistent for now. What do you think about something like this?

```javascript
Characteristic.prototype.broadcast = function(broadcast, callback) {
  this.once('_broadcast', function(error, state) {
    if (callback) {
      callback(error);
    }
      
    if (error) {
      this.emit('broadcastError', error);
    } else {
      this.emit('broadcast', state);
    }
  }.bind);

  this._noble.broadcast(this._peripheralUuid, this._serviceUuid, this.uuid, broadcast);
};
```

Shifting towards an noble internal ```_broadcast``` event, which then gets converted to an external event.

rosek86, 2015-06-06:
Hi guys,

Have you seen 'Production Practices' on error handling from Joyent? 
https://www.joyent.com/developers/node/design/errors

I think it's good to follow best practices suggested by language designers. One thing which I've noticed there is:

"If you want to deliver an error asynchronously, You generally want to use one or the other of these (callback or event emitter), but not both."

I wonder if emitted events are needed for the simplest functions? Possibly it would be easier for Noble users to use callbacks when possible and events emitter when it's really needed?

Otherwise we could end up with a bunch of events not used by anyone which needs to be maintained.

I belive that callbacks are just fine for functions like; write, read, notify or broadcast, but then there are events which must use events emitter; notify, stateChange, discover, disconnect, (connect for consistency). 

Characteristic.prototype.broadcast = function(broadcast, callback) {
  if (callback) {
    this.once('_broadcast', function(error, state) {
      callback(error, state);
    });
  }

  this._noble.broadcast(this._peripheralUuid, this._serviceUuid, this.uuid, broadcast);
};

Maybe even force user to pass a callback:

if (!callback) {
  throw new Error('Callback must be a function');
}

What do you think?

bragma, 2015-06-08:
@rosek86 as you point out, there is probably an overuse of events in noble. Maybe the idea was to keep consistent. Still I think going back to callbacks would require some mayor rework on the code, in particular the bindings.

sandeepmistry, 2015-06-09:
@rosek86 thanks for the link, it makes sense to drop the ```broadcastError``` event, however I would still like success events (mostly to maintain API compatibility).

Here's where I'm at:

```javascript
Characteristic.prototype.broadcast = function(broadcast, callback) {
  this.once('_broadcast', function(error, state) {
    if (callback) {
      callback(error);
    }

    if (!error) {
      this.emit('broadcast', state);
    }
  }.bind);

  this._noble.broadcast(this._peripheralUuid, this._serviceUuid, this.uuid, broadcast);
};
```

bragma, 2015-06-09:
@sandeepmistry there is something strange in the code you suggest. I don't understand the logic. If a callback is specified, you call it but pass only an error, which may be null. Still, you call the event if there is no error. Or, reading it in another way, if there is an error you call the callback, else you call both the callback and the event.
Seems very confusing to me.

rosek86, 2015-06-09:
I agree with @bragma that this code is a bit confusing. @sandeepmistry I think we need at least pass a result (state in the example) to a callback:

```javascript
callback(error, state)
```

I think that emitted events should be called on error too in this way:

```javascript
this.emit('broadcast', error, state);
```

This would be very useful for a 'data' event, e.g. when application waits for data using once function we need to make sure that callback is executed even though the error occurs. Otherwise callback can be called again in inappropriate time which may cause a lot of problems.

I believe that this is how user should handle data events:

```javascript
char.once('data', function (err, data) {
  if (err) {
    // do something
    return;
  }

  // do something else 
});
```

bragma, 2015-06-17:
Probably the best solution is to use callbacks (if provided by the user) and emit events just in case someone wants to subscribe, but in this case is user's responsibility to use once, on and remove the listeners when required. The user should be allowed to provide a callback, or null if he wants to use events.

It requires a major rework of current code base. In the meanwhile, while I am not a big fan of this solution, we could do what suggested by @sandeepmistry in https://github.com/sandeepmistry/noble/issues/111 i.e. something like this:

```
Peripheral.prototype.discoverServices = function(uuids, callback) {
  var servicesDiscoverCallback = function(services) {
    this.removeListener('disconnect', disconnectCallback);
    callback(null, services);
  };

  var disconnectCallback = function() {
    this.removeListener('servicesDiscover', servicesDiscoverCallback);
    callback(new Error('peripheral disconnected'));
  }.bind(this);

  if (callback) {
    this.once('servicesDiscover', servicesDiscoverCallback);
    this.once('disconnect', disconnectCallback);
  }

  this._noble.discoverServices(this.uuid, uuids);
};
```

sandeepmistry, 2015-06-24:
@bragma @rosek86 I still don't think there is anything strange with the following:

1. An internal ```_broadcast``` event listener is added.
2. If a callback is supplied, it is called with an error parameter - which is most likely null, to indicate no error. There is no need to pass ```state```, it not standard Node.js style from my understanding. In this case the value will be the same as the ```broadcast``` argument.
3. If there is no error, the ```broadcast``` event is emitted. I do agree that is a bit overkill, but removing it is a breaking API change, which I prefer to avoid at this point in time.

```javascript
Characteristic.prototype.broadcast = function(broadcast, callback) {
  this.once('_broadcast', function(error, state) {
    if (callback) {
      callback(error);
    }

    if (!error) {
      this.emit('broadcast', state);
    }
  }.bind);

  this._noble.broadcast(this._peripheralUuid, this._serviceUuid, this.uuid, broadcast);
};
```

bragma, 2015-06-25:
It seems to me that the callback(err, data) is fully Node.js style indeed (see https://docs.nodejitsu.com/articles/getting-started/control-flow/what-are-callbacks). I don't like it, because of http://callbackhell.com/ (and I hope soon promises will be available) but again if you give it a look and also look at the entire Node.js API, when callbacks are present the standard is callback(err, arguments).

But let's go back to the original discussion: internal "once" events may not be triggered and cause double triggering on next event. Let's discuss how to solve this, which imho it is not a matter of style, but a bug.

Reconsidering your suggestion https://github.com/sandeepmistry/noble/issues/111#issuecomment-67779701 while I am not a big fan of it (increases proliferation of events), it should solve the original problem and not break the API. I am reporting your code for discoverServices here, the same can be done for all methods using "once" for transforming an event to an externally supplied callback:

```
Peripheral.prototype.discoverServices = function(uuids, callback) {
  var servicesDiscoverCallback = function(services) {
    this.removeListener('disconnect', disconnectCallback);
    callback(null, services);
  };

  var disconnectCallback = function() {
    this.removeListener('servicesDiscover', servicesDiscoverCallback);
    callback(new Error('peripheral disconnected'));
  }.bind(this);

  if (callback) {
    this.once('servicesDiscover', servicesDiscoverCallback);
    this.once('disconnect', disconnectCallback);
  }

  this._noble.discoverServices(this.uuid, uuids);
};
```

rosek86, 2015-06-26:
+1, and when you use callback(err, data) convention you can use libraries like async, e.g. do all BT LE request in series and handle results at the end. 

```javascript
async.series([
  func_read1,
  func_read2
], function (err, results) {
  // check error && handle results 
});
```

which reduces callback hell a bit.

sandeepmistry, 2015-06-26:
Totally agree on the ```read``` scenario.

The ```broadcast``` API I was referring to was an action API (you are enabling/disabling broadcast mode on a characteristic). This is similar to a ```write``` operation, I don't think it's convention to return the data that was written to the callback.

bragma, 2015-06-26:
@sandeepmistry I agree, sorry if I contributed to the misunderstanding. I'm not saying that all methods should return the "input" data as "output" on the callback. But if the purpose of the method is to "read" or gather some data and the operation, even if async, gathers the result in a single step, it should pass it to the callback. So if the main purpose of the method is to "write" something and there is nothing else to get back aside from failure/success status, a callback(err) is enough. But if the method provides some other data, then the prototype should be something like callback(err, result1 [, result2, ... resultN]).
What I'm not totally in is the idea of having a callback only for errors, and an event only for results (unless it is a stream, which is a different scenario because a single operation can generate multiple results at unknown times).
Probably the best example of this are the exec vs spawn methods.

sandeepmistry, 2015-07-05:
@bragma we are on the same page, I'll add this to my TODO list.

oroce, 2015-07-11:
I've been thinking on this issue since it was created.

I wouldn't go with `(err, result)` signature in events, because that's unfamiliar for every node developers.
I think we could keep (for at least now) the events but and try out something small (huge refactor will cause a lot of bugs). 

Express.js team addressed team with a pretty slick way, using [`ee-first`](https://github.com/jonathanong/ee-first) (it's used in one of express' dependencies, in [`on-finished`](https://github.com/jshttp/on-finished/blob/master/index.js#L90-L130)).

```
var first = require('ee-first');
Characteristic.prototype.broadcast = function(broadcast, callback) {
  if (callback) {
     first(['error', 'disconnect', 'broadcast'], function(err, ee, event, args) {
        callback.apply(null, [err].concat(args));
     });
  }

  this._noble.broadcast(this._peripheralUuid, this._serviceUuid, this.uuid, broadcast);
};
```

This way we could:
* keep the events (less refactor)
* make sure event listeners get removed
* keep consistent callback style `(err, result)`

What do you think?

sandeepmistry, 2015-07-12:
@oroce thanks for suggesting ```ee-first```, it looks like a good fit.

I've pushed my WIP for "private events", which was discussed with @bragma and @rosek86 above: https://github.com/sandeepmistry/noble/tree/private-events

@bragma @rosek86 what are your thoughts on ```ee-first``` vs "private event"?

After the "disconnect" issue is resolved, the next step is allowing the bindings to propagate errors, see #184 and #207. Both mechanisms would work, ```ee-first``` looks like a cleaner approach.

benturner, 2015-11-20:
Hi, @sandeepmistry,  just curious about the status of your branch. Are you thinking it's mostly ready to merge? Or does it need more work?

sandeepmistry, 2015-11-28:
@benturner unfortunately, it needs more work, I'm not too happy with the change ATM.

rzr, 2021-03-02:
Please can you update status of this issue on this latest release: 

https://www.npmjs.com/package/@abandonware/noble/v/1.9.2-12