noble/noble #199: Characteristic write event not triggering on OSX Mavericks

alistairjcbrown, 2015-06-10:
I'm using `voodootikigod/node-rolling-spider` module, which uses `noble` as the underlying bluetooth connection library, and attempting to connect to a Parrot Rolling Spider drone.
The [sample code they provide](https://github.com/voodootikigod/node-rolling-spider#taking-off-moving-and-landing) connects, runs the setup, but hangs on handshake.

Digging into `node-rolling-spider`, it looks like it waits for a callback after writing to a characteristic, but that the callback is never called.
https://github.com/voodootikigod/node-rolling-spider/blob/584adcf062f5d71efbda705e0c695a94bfd44c21/lib/drone.js#L295-L305

Jumping into the characteristics code in `nobel` I can see that the callback would be triggered on a write event on the characteristic, which I can see is triggered on the characteristic by nobel when the binding triggers a write event.
https://github.com/sandeepmistry/noble/blob/ffa0f178ed43467558a0c66ef7c69ad1f25fe783/lib/characteristic.js#L58-L62

I believe this to be an issue with the binding for my OSX version as I am on Mavericks and a colleague, who was also on Mavericks and experiencing the same issue, updated to Yosemite after which he was able to connect successfully.

Environment data:

```
Node: 0.10.36
Operating System: OSX 10.9.5
Hardware: MacBook Pro (Retina, 15-inch, Mid 2014)
```

Please let me know if there's any additional data I can give you or any steps I can do to help debug.

sandeepmistry, 2015-06-10:
@alistairjcbrown Could you please try with the latest noble version in Github.

https://github.com/sandeepmistry/noble/commit/ffa0f178ed43467558a0c66ef7c69ad1f25fe783 should fix it. If that works, I can publish a new version to npm and we can update node-rolling-spider to use it.

I've tested a similar patch on OS X 10.8 with someone at JSConf, and it resolved the problem.

sandeepmistry, 2015-07-19:
@alistairjcbrown closing, this change has been published to npm, and node-rolling-spider is now using the latest version of noble: https://github.com/voodootikigod/node-rolling-spider/pull/57 thanks to @mcollina

closed 2015-07-19 by sandeepmistry