noble/noble #200: npm install problem - yosemite 10.10.3 - node 0.12.4 - npm 2.11.2

uhunkler, 2015-06-14:
I also get the »node-gyp rebuild« error when I try to install noble. I have updated node and npm to the latest versions. I tried to install it on 2 yosemite computers getting the same error.

What I wonder is why the »node-gyp rebuild« works for »xpc-connection« but not for »noble«. Any help is very welocme.

npm install noble
-
> xpc-connection@0.1.3 install /Users/urshunkler/workspace/rolling_spider/node_modules/noble/node_modules/xpc-connection
> node-gyp rebuild

  CXX(target) Release/obj.target/binding/src/XpcConnection.o
  SOLINK_MODULE(target) Release/binding.node

> noble@1.0.1 install /Users/urshunkler/workspace/rolling_spider/node_modules/noble
> node-gyp rebuild

  LINK(target) Release/hci-ble
Undefined symbols for architecture x86_64:
  "_main", referenced from:
      start in crt1.10.5.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [Release/hci-ble] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:269:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
gyp ERR! System Darwin 14.3.0
gyp ERR! command "node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/urshunkler/workspace/rolling_spider/node_modules/noble
gyp ERR! node -v v0.12.4
gyp ERR! node-gyp -v v2.0.1
gyp ERR! not ok 
npm ERR! Darwin 14.3.0
npm ERR! argv "node" "/usr/local/bin/npm" "install" "noble"
npm ERR! node v0.12.4
npm ERR! npm  v2.11.2
npm ERR! code ELIFECYCLE

npm ERR! noble@1.0.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the noble@1.0.1 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the noble package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls noble
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/urshunkler/workspace/rolling_spider/npm-debug.log

uhunkler, 2015-06-14:
g++ -v:
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/usr/include/c++/4.2.1
Apple LLVM version 6.1.0 (clang-602.0.53) (based on LLVM 3.6.0svn)
Target: x86_64-apple-darwin14.3.0
Thread model: posix

uhunkler, 2015-06-14:
I tried to change the $PATH to 
/usr/bin:/bin:/usr/sbin:/sbin:/usr/local/bin:/opt/X11/bin

sandeepmistry, 2015-06-24:
@uhunkler can you please try creating a ```src/dummy.c``` with an empty main method.

Then adding something like this to the ```binding.gyp```:

```
       ['OS=="mac"', {
          'sources': [
            'src/dummy.c'
          ]
        }]
```

DomiR, 2015-07-01:
The dummy works

sandeepmistry, 2015-07-05:
Fixed by @DomiR via #205.

closed 2015-07-05 by sandeepmistry